MichaIng/DietPi #1886: Can't connect to WiFi in Tinker board

Hax4us, 2018-07-02:
Yesterday I installed diepi then manually connect to my WiFi and everything was OK 
But today I can't connect to my WiFi through dietpi-config it is showing like this after entering __ssid__ and __key__ 
```
Current Details :
 - SSID = haxyy | 0 Mbit | strength : 0
- Key = mykey
- Mode = DHCP
- IP = 0.0.0.0
- Mask = 0.0.0.0
- DNS = 0.0.0.0
```

Hax4us, 2018-07-02:
OK so here is solution but this solution is temporary 
1.Disable Wireless Adapter Through dietpi-config
2. Then reboot 
3. Then reenable wireless adapter and scan for WiFi then enter credentials and connect 

I tried different solutions 
First I tried reboot ( not solved 
Second I tried reboot then disable wireless adapter and reenable wireless adapter then connect to WiFi :( not solved
Third I tried above temporary solution :) solved

Why it is temporary solution ?
Bcz after shutdown and reopen dietpi , problem appeared again so again apply above solution

Fourdee, 2018-07-02:
@Hax4us 

Hi, thanks for the report 👍 

We are aware of this issue, losing WiFi functionality after reboot: https://github.com/Fourdee/DietPi/issues/1760

We are unsure why this is occurring, pointing towards kernel. We will look at it again in v6.11, once v6.10 is released today.

I'll mark this as closed in preference of this ticket: https://github.com/Fourdee/DietPi/issues/1760 Please continue discussions in that ticket.

closed 2018-07-02 by Fourdee