MichaIng/DietPi #1892: Can't upgrade to latest kernel on Rock64

burgyman, 2018-07-03:
Hi,

Tried everything to upgrade kernel on Rock64, specially instructions from Ayufan according to
https://github.com/ayufan-rock64/linux-build/releases/tag/0.6.44 

CODE:
sudo apt-get update -y
sudo apt-get install linux-rock64 -y

But I'm stuck on original kernel (4.4.77-rockchip-ayufan-136)

Thread in forum https://dietpi.com/phpbb/viewtopic.php?f=11&t=3879
so reporting it here as well, hope you can help.

Thanks.

MichaIng, 2018-07-04:
@burgyman 
Thanks for opening the ticket here on GitHub. Better to handle here than in the forum 👍.

First of all let's check the repo(s):
`ls -l /etc/apt/sources.list.d`
Then the content(s):
`cat /etc/apt/sources.list.d/*`
Then let's check, which kernel packages are available:
`apt search linux-image`

DietPi by default installs/keeps `linux-rock64-package`, so let's check details about this, as well as the package you are upgrading:
`apt show linux-rock64-package linux-rock64`

burgyman, 2018-07-04:
Thanks for the quick response, please see below:

```
root@rock64:~# ls -l /etc/apt/sources.list.d
total 8
-rw-r--r-- 1 root root 183 Jul  4 00:13 ayufan-rock64.list
-rw-r--r-- 1 root root  62 May 17 22:49 mono-xamarin.list
```

```
root@rock64:~# cat /etc/apt/sources.list.d/*
deb http://deb.ayufan.eu/orgs/ayufan-rock64/releases /

# uncomment to use pre-release kernels and compatibility packages
# deb http://deb.ayufan.eu/orgs/ayufan-rock64/pre-releases /deb http://download.mono-project.com/repo/debian stretch main
```

```
root@rock64:~# apt search linux-image
Sorting... Done
Full Text Search... Done
linux-headers-4.16.0-0.bpo.2-arm64/stretch-backports 4.16.16-2~bpo9+1 arm64
  Header files for Linux 4.16.0-0.bpo.2-arm64               linux-headers-4.9.0-3-arm64/stable 4.9.30-2+deb9u5 arm64
  Header files for Linux 4.9.0-3-arm64

linux-headers-4.9.0-4-arm64/stable 4.9.65-3+deb9u1 arm64
  Header files for Linux 4.9.0-4-arm64

linux-headers-4.9.0-5-arm64/stable 4.9.65-3+deb9u2 arm64
  Header files for Linux 4.9.0-5-arm64

linux-headers-4.9.0-6-arm64/stable 4.9.88-1+deb9u1 arm64
  Header files for Linux 4.9.0-6-arm64

linux-image-4.13.0-rc1-rockchip-ayufan-102-g6104f84/unknown 0.5.8 arm64
  Linux kernel, version 4.13.0-rc1-rockchip-ayufan-102-g6104f84

linux-image-4.13.0-rc1-rockchip-ayufan-102-g6104f84-dbg/unknown 0.5.8 arm64
  Linux kernel debugging symbols for 4.13.0-rc1-rockchip-ayufan-102-g6104f84

linux-image-4.13.0-rc1-rockchip-ayufan-104-g6104f84/unknown 0.5.9 arm64
  Linux kernel, version 4.13.0-rc1-rockchip-ayufan-104-g6104f84

linux-image-4.13.0-rc1-rockchip-ayufan-104-g6104f84-dbg/unknown 0.5.9 arm64
  Linux kernel debugging symbols for 4.13.0-rc1-rockchip-ayufan-104-g6104f84

linux-image-4.13.0-rockchip-ayufan-118-gd3263fd/unknown 0.5.10 arm64
  Linux kernel, version 4.13.0-rockchip-ayufan-118-gd3263fd

linux-image-4.13.0-rockchip-ayufan-118-gd3263fd-dbg/unknown 0.5.10 arm64
  Linux kernel debugging symbols for 4.13.0-rockchip-ayufan-118-gd3263fd

linux-image-4.14.0-rockchip-ayufan-136-gee91453/unknown 0.5.15 arm64
  Linux kernel, version 4.14.0-rockchip-ayufan-136-gee91453

linux-image-4.14.0-rockchip-ayufan-136-gee91453-dbg/unknown 0.5.15 arm64
  Linux kernel debugging symbols for 4.14.0-rockchip-ayufan-136-gee91453

linux-image-4.16.0-0.bpo.2-arm64/stretch-backports 4.16.16-2~bpo9+1 arm64
  Linux 4.16 for 64-bit ARMv8 machines

linux-image-4.16.0-0.bpo.2-arm64-dbg/stretch-backports 4.16.16-2~bpo9+1 arm64
  Debug symbols for linux-image-4.16.0-0.bpo.2-arm64

linux-image-4.17.0-rc6-1014-ayufan-g5183d4fd6b6a/unknown 4.17.0-rc6-1014-ayufan arm64
  Linux kernel, version 4.17.0-rc6-1014-ayufan-g5183d4fd6b6a
linux-image-4.17.0-rc6-1014-ayufan-g5183d4fd6b6a-dbg/unknown 4.17.0-rc6-1014-ayufan arm64
  Linux kernel debugging symbols for 4.17.0-rc6-1014-ayufan-g5183d4fd6b6a

linux-image-4.4.126-rockchip-ayufan-239/unknown,now 0.6.44 arm64 [installed]
[... 525 lines cut ...]
Maintainer: Kamil Trzciński <ayufan@ayufan.eu>
Installed-Size: unknown
Depends: linux-rock64-package (= 0.3.3), linux-image-4.4.70-rockchip-ayufan-55 (= 0.3.3), linux-headers-4.4.70-rockchip-ayufan-55 (= 0.3.3)
Homepage: https://gitlab.com/ayufan-rock64/linux-build
License: MIT
Vendor: Kamil Trzciński
Download-Size: 1,102 B
APT-Sources: http://deb.ayufan.eu/orgs/ayufan-rock64/releases  Packages
Description: Rock64 Linux virtual package: depends on kernel and compatibility package

Package: linux-rock64
Version: 0.3.2
Priority: optional
Section: admin
Maintainer: Kamil Trzciński <ayufan@ayufan.eu>
Installed-Size: unknown
Depends: linux-rock64-package (= 0.3.2), linux-image-4.4.70-rockchip-ayufan-54 (= 0.3.2), linux-headers-4.4.70-rockchip-ayufan-54 (= 0.3.2)
Homepage: https://gitlab.com/ayufan-rock64/linux-build
License: MIT
Vendor: Kamil Trzciński
Download-Size: 1,106 B
APT-Sources: http://deb.ayufan.eu/orgs/ayufan-rock64/releases  Packages
Description: Rock64 Linux virtual package: depends on kernel and compatibility package

```

burgyman, 2018-07-04:
apt-get dist-upgrade didn't do anything (tried that before):
```
root@rock64:~# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Still no change:
```
root@rock64:~# G_AGP linux-image-4.4.77*
[  OK  ] Root access verified.
[ INFO ] Install verified: linux-image-4.4.77*
[  OK  ] APT removal for: linux-image-4.4.77* , please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Package 'linux-image-4.4.77-rockchip-ayufan-102' is not installed, so not removed
Package 'linux-image-4.4.77-rockchip-ayufan-102-dbg' is not installed, so not removed
Package 'linux-image-4.4.77-rockchip-ayufan-104' is not installed, so not removed
Package 'linux-image-4.4.77-rockchip-ayufan-104-dbg' is not installed, so not removed
Package 'linux-image-4.4.77-rockchip-ayufan-118' is not installed, so not removed
Package 'linux-image-4.4.77-rockchip-ayufan-118-dbg' is not installed, so not removed
Package 'linux-image-4.4.77-rockchip-ayufan-136-dbg' is not installed, so not removed
The following packages will be REMOVED:
  linux-image-4.4.77-rockchip-ayufan-136*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
  After this operation, 52.5 MB disk space will be freed.
                                                         (Re(Reading database ... 77252 files and directories currently installed.)
Removing linux-image-4.4.77-rockchip-ayufan-136 (0.5.15) ...update-initramfs: Deleting /boot/initrd.img-4.4.77-rockchip-ayufan-136
(Reading database ... 76362 files and directories currently installed.)
Purging configuration files for linux-image-4.4.77-rockchip-ayufan-136 (0.5.15) ...

[  OK  ] G_AGP: linux-image-4.4.77*
root@rock64:~#
root@rock64:~#
root@rock64:~# G_AGI linux-image-4.4.126-rockchip-ayufan-239[  OK  ] APT installation for: linux-image-4.4.126-rockchip-ayufan-239, please wait...

[  OK  ] G_AGI: linux-image-4.4.126-rockchip-ayufan-239
```

```
root@rock64:~# uname -a
Linux rock64 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux

```

Fourdee, 2018-07-04:
@burgyman 

Apologies, have to ask, did you reboot system after installing the kernel?

burgyman, 2018-07-04:
Yes, result above is after reboot :)

MichaIng, 2018-07-04:
Interesting:
`linux-rock64` depends as meta package on the current `linux-image-4.4.126-rockchip-ayufan-239`  and interestingly as well on `linux-rock64-package`, which is the only package we actively install/mark as manual on DietPi-PREP. Since `linux-image-4.4.126-rockchip-ayufan-239` should be the kernel itself, I am thinking what `linux-rock64-package` actually contains. Perhaps some firmware/binary implementations into the system, symlinks of commands, like vcgencmd/tvservice on RPi?

Sadly the kernel headers, which we usually don't need, is also part of the meta package. Bootloader u-boot is included and I guess this is as it should be?

@burgyman 
Since the new kernel got installed, as I believe, only idea is that the other kernel did not get really purged by purging the package, thus both kernel versions are currently available. On the other hand, this clearly sounds like it was: `After this operation, 52.5 MB disk space will be freed`. Could you check:
`ls -l /lib/modules`

Hmm `linux-rock64-package` depends on `device-tree-compiler` as well. Does this mean, that device tree is regenerated based on configs, like on RPi based on config.txt? I am not too deep in this matter, but maybe for this the kernel headers are needed.

Please try `G_AGI linux-rock64` to have everything installed, that seems to be expected by Rock64 dev team. If this solves the issue, we know, that we have to install/mark this one also on DietPi-PREP.

burgyman, 2018-07-04:
```
root@rock64:~# ls -l /lib/modules
total 8
drwxr-xr-x 3 root root 4096 Jun 23 22:29 4.4.126-rockchip-ayufan-239
drwxr-xr-x 2 root root 4096 Jul  4 17:57 4.4.77-rockchip-ayufan-136
root@rock64:~#
root@rock64:~#
root@rock64:~# G_AGI linux-rock64
[  OK  ] Root access verified.
[  OK  ] APT installation for: linux-rock64, please wait...

[  OK  ] G_AGI: linux-rock64
root@rock64:~#
```

After reboot same thing
```
root@rock64:~# uname -a
Linux rock64 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux
root@rock64:~#
```

MichaIng, 2018-07-04:
@burgyman 
Okay, so:
- Their own docs do not work: https://github.com/ayufan-rock64/linux-build/blob/master/recipes/kernel-upgrade.md
- APT purge old kernel does not really remove the old kernel, as `/lib/modules/4.4.77-rockchip-ayufan-136` still exists and is as well still loaded instead of the newer one. Or is the folder (moreless) empty?
`G_TREESIZE /lib/modules`

The question is which version /boot contains and if it is safe to `rm -R /lib/modules/4.4.77-rockchip-ayufan-136` to force leading the new version. Actually if the old one is leaded, I think bootloader settings seems to not have been updated as expected.

@burgyman 
Last test I have in mind would be to reinstall current kernel package again. Perhaps the `linux-rock64` brought some stuff that is needed to make boot loading the updated kernel:
`apt install --reinstall linux-image-4.4.126-rockchip-ayufan-239`
If it doesn't help, I suggest opening some issue on their GitHub repo, ... (quick check) ... ah it is already there: https://github.com/ayufan-rock64/linux-build/issues/214
I will add info and refer to this topic.

burgyman, 2018-07-04:
```
root@rock64:~# G_TREESIZE /lib/modules
36.6 MB /lib/modules
36.6 MB /lib/modules/4.4.126-rockchip-ayufan-239
4.0 KB /lib/modules/4.4.77-rockchip-ayufan-136
root@rock64:~#
```

```
root@rock64:~# apt install --reinstall linux-image-4.4.126-rockchip-ayufan-239
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 0 B/12.9 MB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 76362 files and directories currently installed.)
Preparing to unpack .../linux-image-4.4.126-rockchip-ayufan-239_0.6.44_arm64.deb ...
Unpacking linux-image-4.4.126-rockchip-ayufan-239 (0.6.44) over (0.6.44) ...
Setting up linux-image-4.4.126-rockchip-ayufan-239 (0.6.44) ...
update-initramfs: Generating /boot/initrd.img-4.4.126-rockchip-ayufan-239
```

After reboot:
```
root@rock64:~# uname -a
Linux rock64 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux
root@rock64:~#
```

MichaIng, 2018-07-04:
@burgyman 
Thanks so much for your fast replies 😃. So the old folder is actually empty and `update-initramfs: Generating /boot/initrd.img-4.4.126-rockchip-ayufan-239` looks like installation was complete.

Does /boot still contain the old initramfs file(s)?
`ls -l /boot`

MichaIng, 2018-07-04:
@burgyman 
As advised in the other issue, please try:
```
apt install flash-kernel
flash-kernel
```

I hope `flash-kernel` command somehow explains itself `flash-kernel --help` might otherwise. Can't test it on VM, since it is just available for ARM architecture. That should move the kernel files into the right place of the /boot fs.

burgyman, 2018-07-05:
```
root@rock64:~# cat /boot/efi/extlinux/extlinux.conf
cat: /boot/efi/extlinux/extlinux.conf: No such file or directory
root@rock64:~# apt install flash-kernel
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  devio
Recommended packages:
  u-boot-tools
The following NEW packages will be installed:
  devio flash-kernel
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 61.0 kB of archives.
After this operation, 211 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://cdn-aws.deb.debian.org/debian stretch/main arm64 devio arm64 1.2-1.2+b1 [15.4 kB]
Get:2 https://cdn-aws.deb.debian.org/debian stretch/main arm64 flash-kernel arm64 3.79 [45.6 kB]
Fetched 61.0 kB in 1s (36.4 kB/s)
Preconfiguring packages ...
Selecting previously unselected package devio.
(Reading database ... 76362 files and directories currently installed.)
Preparing to unpack .../devio_1.2-1.2+b1_arm64.deb ...
Unpacking devio (1.2-1.2+b1) ...
Selecting previously unselected package flash-kernel.
Preparing to unpack .../flash-kernel_3.79_arm64.deb ...
Unpacking flash-kernel (3.79) ...
Setting up devio (1.2-1.2+b1) ...
Setting up flash-kernel (3.79) ...

Creating config file /etc/default/flash-kernel with new version
root@rock64:~#
root@rock64:~#
root@rock64:~# flash-kernel
Using DTB: rk3328-rock64.dtb
Installing /usr/lib/linux-image-4.4.126-rockchip-ayufan-239/rockchip/rk3328-rock64.dtb into /boot/dtbs/4.4.126-rockchip-ayufan-239/rk3328-rock64.dtb
Installing new rk3328-rock64.dtb.
flash-kernel: installing version 4.4.126-rockchip-ayufan-239Installing new Image.
Installing new initrd.img.
Installing new dtb.
root@rock64:~#
```

Same after reboot
```
root@rock64:~# uname -a
Linux rock64 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux
root@rock64:~#
```

Flash kernel help
```
root@rock64:~# flash-kernel --help
Using DTB: rk3328-rock64.dtb                                Couldn't find DTB rk3328-rock64.dtb in /usr/lib/linux-image---help or /etc/flash-kernel/dtbs
root@rock64:~#
```

MichaIng, 2018-07-05:
@burgyman 
Okay, reason might be that the new kernel is not installed to the efi partition, even that Rock64 (/boot/efi exists) uses efi? But sorry I can't help further from here on. Someone with own Rock64 to replicate/investigate is needed, I hope @Fourdee finds time, also to just verify that Rock64 uses efi, there stays the old kernel inside efi dir etc. knowing the exact pre-image we use etc.

Thanks so much for your investigation help so far 👍.

> flash-kernel --help

Okay the tool doesn't support the --help option 😉. At least it does it's job automatically.

Fourdee, 2018-07-05:
@MichaIng @burgyman 

I'll take a look when I can, leave it with me.

Fourdee, 2018-07-07:
Test:
- 🈴 Does nothing in relation to kernel update
```
root@DietPi:~# uname -a
Linux DietPi 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux

root@DietPi:~# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following NEW packages will be installed:
  device-tree-compiler
The following packages will be upgraded:
  linux-rock64-package
1 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,656 kB of archives.
After this operation, 508 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:2 http://deb.ayufan.eu/orgs/ayufan-rock64/releases  linux-rock64-package 0.6.44 [1,308 kB]
Get:1 https://cdn-aws.deb.debian.org/debian stretch/main arm64 device-tree-compiler arm64 1.4.2-1 [348 kB]
Fetched 1,656 kB in 2s (721 kB/s)
Selecting previously unselected package device-tree-compiler.
(Reading database ... 33209 files and directories currently installed.)
Preparing to unpack .../device-tree-compiler_1.4.2-1_arm64.deb ...
Unpacking device-tree-compiler (1.4.2-1) ...
Preparing to unpack .../linux-rock64-package_0.6.44_all.deb ...
Unpacking linux-rock64-package (0.6.44) over (0.5.15) ...
Setting up device-tree-compiler (1.4.2-1) ...
Setting up linux-rock64-package (0.6.44) ...

Configuration file '/boot/efi/extlinux/extlinux.conf'
 ==> Deleted (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
 ==> Keeping old config file as default.

```

manual pkg install:
```
root@DietPi:~# apt-get install linux-image-4.4.126-rockchip-ayufan-239
Reading package lists... Done
Building dependency tree
Reading state information... Done
Suggested packages:
  linux-firmware-image-4.4.126-rockchip-ayufan-239
The following NEW packages will be installed:
  linux-image-4.4.126-rockchip-ayufan-239
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 12.9 MB of archives.
After this operation, 66.1 MB of additional disk space will be used.
Get:1 http://deb.ayufan.eu/orgs/ayufan-rock64/releases  linux-image-4.4.126-rockchip-ayufan-239 0.6.44 [12.9 MB]
Fetched 12.9 MB in 3s (3,336 kB/s)
Selecting previously unselected package linux-image-4.4.126-rockchip-ayufan-239.
(Reading database ... 33244 files and directories currently installed.)
Preparing to unpack .../linux-image-4.4.126-rockchip-ayufan-239_0.6.44_arm64.deb ...
Unpacking linux-image-4.4.126-rockchip-ayufan-239 (0.6.44) ...
Setting up linux-image-4.4.126-rockchip-ayufan-239 (0.6.44) ...
update-initramfs: Generating /boot/initrd.img-4.4.126-rockchip-ayufan-239
```

- 🈯️ root@DietPi:~# uname -a
Linux DietPi 4.4.126-rockchip-ayufan-239 #1 SMP Sun May 27 18:38:24 UTC 2018 aarch64 GNU/Linux
```
cp /boot/initrd.img-4.4.126-rockchip-ayufan-239 $(df -h | grep mmcblk0p6 | awk '{print $NF}')/initrd.img
cp /boot/vmlinuz-4.4.126-rockchip-ayufan-239 $(df -h | grep mmcblk0p6 | awk '{print $NF}')/Image
```

Fourdee, 2018-07-07:
Ok, I'll check preimage, see if its our issue:
- ```fstab``` ```LABEL=boot /boot/efi vfat defaults,sync 0 0```

MichaIng, 2018-07-07:
@Fourdee 
```
Configuration file '/boot/efi/extlinux/extlinux.conf'
 ==> Deleted (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
 ==> Keeping old config file as default.
```
- Error makes sense, if EFI partition is not pointed there, but might be our APT force-old-conf rule an issue here as well, e.g. `extlinux.conf` needs edit to force correct kernel upload to EFI partition/handling? Maybe worth to having a look into this file (before and after kernel upgrade without rule), also as `ayufan` asked for it's content in the other thread.

burgyman, 2018-07-08:
@Fourdee not sure I'm following, can I run below to upgrade to latest kernel? 
```
cp /boot/initrd.img-4.4.126-rockchip-ayufan-239 $(df -h | grep mmcblk0p6 | awk '{print $NF}')/initrd.img
cp /boot/vmlinuz-4.4.126-rockchip-ayufan-239 $(df -h | grep mmcblk0p6 | awk '{print $NF}')/Image
```

burgyman, 2018-07-08:
Worked great!
```
root@rock64:~# uname -a
Linux rock64 4.4.126-rockchip-ayufan-239 #1 SMP Sun May 27 18:38:24 UTC 2018 aarch64 GNU/Linux
root@rock64:~#
```

MichaIng, 2018-07-09:
@burgyman 
I don't think it's a kernel bug, since this is a stable release.

`lsblk` does not contain the SD at all?

Fourdee, 2018-07-09:
@burgyman 

Is the SD card plugged into a SD card USB reader, or, the device itself. It could be the kernel has changed the way this functions.

Might be worth checking dmesg for any info (paste whole contents and we'll check):
```
dmesg
```

Might also require a .dtb update, i'll check this later today.

burgyman, 2018-07-09:
@Fourdee
SD card is inserted directly in the device (used mainly for dietpi-backup)

``` 
root@rock64:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.126-rockchip-ayufan-239 (root@b57dd544ff03) (gcc version 7.2.0 (Ubuntu/Linaro 7.2.0-6ubuntu1) ) #1 SMP Sun May 27 18:38:24 UTC 2018
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] earlycon: Early serial console at MMIO32 0xff130000 (options '')
[    0.000000] bootconsole [uart0] enabled
[    0.000000] On node 0 totalpages: 1043968
[    0.000000]   DMA zone: 16312 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 1043968 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] PERCPU: Embedded 21 pages/cpu @ffffffc0fef3a000 s46248 r8192 d31576 u86016
[    0.000000] pcpu-alloc: s46248 r8192 d31576 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 1027656
[    0.000000] Kernel command line: earlycon=uart8250,mmio32,0xff130000 rw root=LABEL=linux-root rootwait rootfstype=ext4 init=/sbin/init coherent_pool=1M ethaddr=92:90:0d:d9:63:59 eth1addr=92:90:0d:d9:63:79 serial=d1d173cbe319b15
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] software IO TLB [mem 0xf6200000-0xfa200000] (64MB) mapped at [ffffffc0f6200000-ffffffc0fa1fffff]
[    0.000000] Memory: 4014268K/4175872K available (11966K kernel code, 1664K rwdata, 4404K rodata, 1152K init, 751K bss, 161604K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
                   vmalloc : 0xffffff8008000000 - 0xffffffbdbfff0000   (   246 GB)
                     .init : 0xffffff8009090000 - 0xffffff80091b0000   (  1152 KB)
                     .text : 0xffffff8008080000 - 0xffffff8008c30000   ( 11968 KB)
                   .rodata : 0xffffff8008c30000 - 0xffffff8009090000   (  4480 KB)
                     .data : 0xffffff80091b0000 - 0xffffff8009350008   (  1665 KB)
                   vmemmap : 0xffffffbdc0000000 - 0xffffffbfc0000000   (     8 GB maximum)
                             0xffffffbdc0008000 - 0xffffffbdc3fc0000   (    63 MB actual)
                   fixed   : 0xffffffbffe7fb000 - 0xffffffbffec00000   (  4116 KB)
                   PCI I/O : 0xffffffbffee00000 - 0xffffffbfffe00000   (    16 MB)
                   memory  : 0xffffffc000200000 - 0xffffffc0ff000000   (  4078 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 64.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=64, nr_cpu_ids=4
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rockchip-sip: function[82000008]: args:[2,0,5] => res:[0,2ed96880,0,0]
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] rockchip_mmc_get_phase: invalid clk rate
[    0.000000] Architected cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000007] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001667] Console: colour dummy device 80x25
[    0.002079] console [tty0] enabled
[    0.002403] bootconsole [uart0] disabled
[    0.002799] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=24000)
[    0.002837] pid_max: default: 32768 minimum: 301
[    0.002977] Security Framework initialized
[... 1369 lines cut ...]
[  138.128471] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[  212.489996] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[  304.168803] EXT4-fs (mmcblk0p7): error count since last fsck: 1
[  304.168852] EXT4-fs (mmcblk0p7): initial error at time 1478193402: ext4_find_entry:1450: inode 2961
[  304.168884] EXT4-fs (mmcblk0p7): last error at time 1478193402: ext4_find_entry:1450: inode 2961
[  326.764983] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[ 9876.896207] TCP: request_sock_TCP: Possible SYN flooding on port 6800. Sending cookies.  Check SNMP counters.
[... cut ...]
[10006.794094] TCP: request_sock_TCP: Possible SYN flooding on port 6801. Sending cookies.  Check SNMP counters.
root@rock64:~#
```

Also noticed that on login I get the following NULL error below
```
───────────────────────────────────────
 DietPi     | 22:00 | Mon 09/07/18
 ───────────────────────────────────────
 v6.11      | Rock64 (aarch64)
 ───────────────────────────────────────
 NULL       | Use dietpi-config to setup a connection
 ───────────────────────────────────────
CPU freq       : 1296.00 MHz
........ 
```

burgyman, 2018-07-26:
@Fourdee new stable release from ayufan for Rock64
https://forum.pine64.org/showthread.php?tid=6309&pid=39292#pid39292

burgyman, 2018-07-26:
Steps to upgrade to latest stable

```
root@rock64:~# uname -a
Linux rock64 4.4.126-rockchip-ayufan-239 #1 SMP Sun May 27 18:38:24 UTC 2018 aarch64 GNU/Linux
```

```
root@rock64:~# sudo apt-get update -y
Hit:1 http://download.mono-project.com/repo/debian stretch InRelease
Get:2 http://deb.ayufan.eu/orgs/ayufan-rock64/releases  InRelease [1,335 B]
Ign:3 https://cdn-aws.deb.debian.org/debian stretch InRelease
Hit:4 https://cdn-aws.deb.debian.org/debian stretch-updates InRelease
Hit:5 https://cdn-aws.deb.debian.org/debian-security stretch/updates InRelease
Hit:6 https://cdn-aws.deb.debian.org/debian stretch-backports InRelease
Hit:7 https://cdn-aws.deb.debian.org/debian stretch Release
Fetched 1,335 B in 5s (250 B/s)
Reading package lists... Done
```

```
root@rock64:~# sudo apt-get install linux-rock64 -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
linux-rock64 is already the newest version (0.7.8).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 linux-rock64 : Depends: u-boot-rockchip-rock64 (= 2017.09-rockchip-ayufan-1017-g7bd50d7e0f)
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

try to fix dependencies:
```
root@rock64:~# apt --fix-broken install
Reading package lists... Done
Building dependency tree
Reading state information... Done
Correcting dependencies... Done
The following package was automatically installed and is no longer required:
  u-boot-rock64
Use 'apt autoremove' to remove it.
The following additional packages will be installed:
  u-boot-rockchip-rock64
The following NEW packages will be installed:
  u-boot-rockchip-rock64
0 upgraded, 1 newly installed, 0 to remove and 161 not upgraded.
4 not fully installed or removed.
Need to get 0 B/365 kB of archives.
After this operation, 1,076 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 97679 files and directories currently installed.)
Preparing to unpack .../u-boot-rockchip-rock64_2017.09-rockchip-ayufan-1017-g7bd50d7e0f_all.deb ...
Unpacking u-boot-rockchip-rock64 (2017.09-rockchip-ayufan-1017-g7bd50d7e0f) ...
dpkg: error processing archive /var/cache/apt/archives/u-boot-rockchip-rock64_2017.09-rockchip-ayufan-1017-g7bd50d7e0f_all.deb (--unpack):
 trying to overwrite '/usr/local/sbin/rock64_upgrade_bootloader.sh', which is also in package u-boot-rock64 0.6.44
Errors were encountered while processing:
 /var/cache/apt/archives/u-boot-rockchip-rock64_2017.09-rockchip-ayufan-1017-g7bd50d7e0f_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

force overwrite:
```
root@rock64:~# apt-get -o Dpkg::Options::="--force-overwrite" install u-boot-rockchip-rock64
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  u-boot-rock64
Use 'apt autoremove' to remove it.
[... 78 lines cut ...]
root@rock64:~#
```

still nothing after reboot:
```
root@rock64:~# uname -a
Linux rock64 4.4.126-rockchip-ayufan-239 #1 SMP Sun May 27 18:38:24 UTC 2018 aarch64 GNU/Linux
root@rock64:~#
```

copy files manually again as instructed before:
```
cp /boot/initrd.img-4.4.132-1072-rockchip-ayufan-ga1d27dba5a2e $(df -h | grep mmcblk0p6 | awk '{print $NF}')/initrd.img
cp /boot/vmlinuz-4.4.132-1072-rockchip-ayufan-ga1d27dba5a2e $(df -h | grep mmcblk0p6 | awk '{print $NF}')/Image
```

upgrade successful after reboot:
```
root@rock64:~# uname -a
Linux rock64 4.4.132-1072-rockchip-ayufan-ga1d27dba5a2e #1 SMP Sat Jul 21 20:18:03 UTC 2018 aarch64 GNU/Linux
root@rock64:~#
```

Issue with SD card was also automatically fixed, SD card is back :)
👍

Fourdee, 2018-07-27:
@burgyman 

Currently finalizing an updated Rock64 image, with following fixes:
- Updated/correct rock64 APT packages installed (should resolve the APT based kernel update issue)
- Corrected ```fstab``` entries to ensure ```/boot/efi``` mount exists, preventing previous issues with manual/auto kernel updates.

Will do some local testing on the image today/tomorrow, if all goes well, will release it tomorrow. Will update you.

Fourdee, 2018-07-27:
Note to self:
```
root@192.168.0.32's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ───────────────────────────────────────
 DietPi     | 00:29 | Fri 27/07/18
 ───────────────────────────────────────
 v6.12      | RockPro64 (aarch64)
 ───────────────────────────────────────
 NULL       | Use dietpi-config to setup a connection
 ───────────────────────────────────────
```

Fourdee, 2018-07-27:
eth lacks ipv4:
```
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 12:72:6b:7b:0a:dc brd ff:ff:ff:ff:ff:ff
    inet6 fe80::1072:6bff:fe7b:adc/64 scope link
       valid_lft forever preferred_lft forever
```
- ```state UNKNOWN``` will fail our ```state UP``` detection for network info.
- Bug in kernel/driver for eth? Or, issue our end with PREP? Will verify with pre-image

Fourdee, 2018-07-27:
- Possible same issue with: https://github.com/Fourdee/DietPi/issues/1812#issuecomment-408547991?

Fourdee, 2018-08-20:
Marking as closed in favor of https://github.com/Fourdee/DietPi/issues/1892

closed 2018-08-20 by Fourdee

Fourdee, 2018-08-21:
Rock64 image updated to 0.7.9:
https://dietpi.com/downloads/images/DietPi_Rock64-ARMv8-Stretch.7z

This should resolve the APT + kernel upgrade issues due to package renamed.