MichaIng/DietPi #1893: Samba sharing issue

mgherghi, 2018-07-04:
Hi all, im having the toughest time mounting my samba share that originates from my merlin modded Asus Ac68U (routers address is 192.168.1.1). My PC finds the shares fine, and without any issue. But once i attempt the same mount in the Pi it fails. I am currently on v6.10 and I am using the drive-manager to add a network drive, but I keep getting this error :
 

│ DietPi-Drive_Manager: mount -t cifs -o username=(edited_out),password=(edited_out)
│ //192.168.1.1/Media /mnt/samba
│ - Exit code: 32
│ - DietPi version: v6.10 | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4
│ - Image creator: DietPi Core Team
│ - Pre-image: Raspbian Lite
│
│ Log file contents:
│ mount error(22): Invalid argument
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)

Fourdee, 2018-07-04:
@mgherghi 

Thanks for the report 👍 

Does the username/password contain any strange characters (eg: ```$\/:;[]#``` and spaces)?

mgherghi, 2018-07-04:
no its just letters and numbers

Fourdee, 2018-07-04:
@mgherghi 

Strange, mounts fine here for Windows share.

Try the following, notice the ```"``` in between the user/password:
```
mkdir -p /mnt/samba
mount -t cifs -o username="CHANGEME",password="CHANGEME"//192.1681.1/Media /mnt/samba
```

Fourdee, 2018-07-04:
Note to self, ```fstab``` does not support user/password entries contained within either ```"``` or ```'```
```
root@DietPi:~# mount -a
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
```

mgherghi, 2018-07-04:
im getting this error :
mount: can't find /mnt/samba in /etc/fstab

Fourdee, 2018-07-04:
@mgherghi 

> mount: can't find /mnt/samba in /etc/fstab

Very strange, fstab entry is not required for manual mounting. You received this error, running these commands?
https://github.com/Fourdee/DietPi/issues/1893#issuecomment-402530163

mgherghi, 2018-07-04:
Yes

Fourdee, 2018-07-04:
@mgherghi 

Please send us a bug report. We'd like to verify fstab and additional settings.
```
dietpi-bugreport
```

Once completed, reply with the reference code and I will take a look.

mgherghi, 2018-07-04:
Reference:
b68fb98c-a6b7-4f41-acf5-b2f86b2eaae0

mgherghi, 2018-07-04:
what should I do?

Fourdee, 2018-07-05:
@mgherghi 

Thanks for sending the bug report 👍 

I've found the following, the samba server may not support version 2/3 of SMB:
```
[  318.574925] No dialect specified on mount. Default has changed to a more secure dialect, SMB2.1 or later (e.g. SMB3), from CIFS (SMB1). To use the less secure SMB1 dialect to access old servers which do not support SMB3 (or SMB2.1) specify vers=1.0 on mount.
[  318.586740] CIFS VFS: cifs_mount failed w/return code = -22
[  363.406560] systemd-fstab-generator[2105]: Ignoring automount option for root device
[  815.700246] No dialect specified on mount. Default has changed to a more secure dialect, SMB2.1 or later (e.g. SMB3), from CIFS (SMB1). To use the less secure SMB1 dialect to access old servers which do not support SMB3 (or SMB2.1) specify vers=1.0 on mount.
[  815.704388] CIFS VFS: cifs_mount failed w/return code = -22
```

Please try mounting the drive, using SMB1.0:
```
mkdir -p /mnt/samba
mount -t cifs -o username="CHANGEME",password="CHANGEME",vers=1.0 //192.1681.1/Media /mnt/samba
```

Fourdee, 2018-07-06:
Notes:
- Try default, then while loop
- vers=3.0
- vers=2.1
- vers=2.0
- vers=1.0

Fourdee, 2018-07-15:
No response, marking as closed.

Should be resolved with v6.12 via SMB version detection during mounting.

closed 2018-07-15 by Fourdee