MichaIng/DietPi #1896: DietPi-Software | Sonarr dies when web interface is accessed

hellfirehd, 2018-07-04:
#### Required Information:
- DietPi version: 6.10
- Distro version	: stretch
- Kernel version: #1 SMP PREEMPT Thu Apr 5 12:46:33 UTC 2018
- SBC device: Odroid XU3/XU4/HC1/HC2 (armv7l)
- Power supply used: 12V 1.5A
- SDcard used: Transcend Premium 32GB
- Hard Drive: Seagate Momentus 5400.6 ST9500325AS

#### Additional Information (if applicable):
- Software title: Sonarr
- Fresh install using dietpi-software
- Can this issue be replicated on a fresh installation of DietPi?
- reference code: bb55358e-9cff-4c09-bf86-6edeeaa4258c

#### Steps to reproduce:
1. Install Sonarr using dietpi-software on ODroid HC2
2. Access the web interface
3. Daemon crashes

#### Expected behaviour:
- Daemon should not be crashing

#### Actual behaviour:
- If the web interface is not accessed the daemon stays up. 

#### Extra details:
- Same behavior if launched as root from console: /user/bin/mono --server /opt/NzbDrone/NzbDrone.exe

Stacktrace:

  at <unknown> <0xffffffff>
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () <0x00033>
  at (wrapper runtime-invoke) <Module>.runtime_invoke_bool (object,intptr,intptr,intptr) [0x0001e] in <71d8ad678db34313b7f718a414dfcb25>:0
/proc/self/maps:
00459000-0070f000 r-xp 00000000 08:01 582634161  /usr/bin/mono-sgen
0071f000-00723000 r-xp 002b6000 08:01 582634161  /usr/bin/mono-sgen
00723000-00726000 rwxp 002ba000 08:01 582634161  /usr/bin/mono-sgen
00726000-01ef3000 rwxp 00000000 00:00 0          [heap]
ab200000-ab300000 rwxp 00000000 00:00 0
ab800000-ab8d0000 rwxp 00000000 00:00 0
ab8d0000-ab900000 ---p 00000000 00:00 0
ab900000-aba00000 rwxp 00000000 00:00 0
abb00000-abc00000 rwxp 00000000 00:00 0
ac100000-ac1fd000 rwxp 00000000 00:00 0
ac1fd000-ac200000 ---p 00000000 00:00 0
ac200000-ac229000 rwxp 00000000 00:00 0
ac229000-ac300000 ---p 00000000 00:00 0
ac300000-ac400000 rwxp 00000000 00:00 0
ac41c000-ac700000 r-xp 00000000 08:01 1827078    /usr/lib/mono/gac/System.Web/4.0.0.0__b03f5f7f11d50a3a/System.Web.dll
ac700000-ac800000 rwxp 00000000 00:00 0
ac900000-aca00000 rwxp 00000000 00:00 0
acb00000-acbf9000 rwxp 00000000 00:00 0
acbf9000-acc00000 ---p 00000000 00:00 0
acd00000-ace00000 rwxp 00000000 00:00 0
acf00000-ad000000 rwxp 00000000 00:00 0
ad0fe000-ad1ff000 rwxp 00000000 00:00 0
ad300000-ad400000 rwxp 00000000 00:00 0
ad500000-ad600000 rwxp 00000000 00:00 0
ad700000-ad729000 rwxp 00000000 00:00 0
ad729000-ad800000 ---p 00000000 00:00 0
ad836000-ad837000 ---p 00000000 00:00 0
ad837000-ad936000 rwxp 00000000 00:00 0
ad936000-ad937000 ---p 00000000 00:00 0
ad937000-ada38000 rwxp 00000000 00:00 0
ada3a000-ada3c000 rwxp 00000000 00:00 0
ada3e000-ada40000 rwxp 00000000 00:00 0
ada42000-ada44000 rwxp 00000000 00:00 0
ada46000-ada48000 rwxp 00000000 00:00 0
ada4a000-ada4c000 rwxp 00000000 00:00 0
[... 545 lines cut ...]
b6fbc000-b6fbd000 ---p 00000000 00:00 0
b6fbd000-b6fbe000 r-xp 00000000 00:00 0
b6fbe000-b6fc1000 rwxp 00000000 00:00 0
b6fc1000-b6fc2000 r-xp 00017000 08:01 582542103  /lib/arm-linux-gnueabihf/ld-2.24.so
b6fc2000-b6fc3000 rwxp 00018000 08:01 582542103  /lib/arm-linux-gnueabihf/ld-2.24.so
bedd7000-bedf8000 rw-p 00000000 00:00 0          [stack]
beeaa000-beeab000 r-xp 00000000 00:00 0          [sigpage]
beeab000-beeac000 r--p 00000000 00:00 0          [vvar]
beeac000-beead000 r-xp 00000000 00:00 0          [vdso]
ffff0000-ffff1000 r-xp 00000000 00:00 0          [vectors]

Native stacktrace:

Debug info from gdb:

mono_gdb_render_native_backtraces not supported on this platform, unable to find gdb or lldb

=================================================================
Got a SIGILL while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries
used by your application.
=================================================================

Aborted
root@media:/opt/NzbDrone#

hellfirehd, 2018-07-04:
Just installed DietPi 6.10 on a 2nd ODroid HC2 with a new (Samsung) SD card.  Let it do all the updates then installed Sonarr.  Same crash happens.

hellfirehd, 2018-07-04:
Might be a mono issue as I am experiencing the same failure with Radarr.

MichaIng, 2018-07-04:
@hellfirehd 
Thanks for your report!

Interesting, we just tested it when tweaking the startup options within the systemd unit a bid: https://github.com/Fourdee/DietPi/pull/1889

Worked well there and the options changed doesn't seem to be the reason, since you tried it as root without these as well.
Thus it most likely is a Odroid and/or Jessie related issue. I will mark for investigation.

hellfirehd, 2018-07-04:
Just tested Sonarr on my RPi 2 Model B (armv7l) and it is working well there.  Sadly I don't have USB Drive for it so it's not temporary solution for me.

Fourdee, 2018-07-05:
#### BR Notes:

XFS
```
/dev/sda1: LABEL="storage" UUID="X" TYPE="xfs" PARTUUID="X"
```

Userdata located on RootFS:
```
drwxrwxr-x  7 dietpi dietpi   79 Jul  4 04:36 dietpi_userdata
```

Service:
```
Jul 04 18:31:00 media mono[1179]:   at System.Collections.Specialized.NameValueCollection.Set (System.String name, System.String value) [0x00000] in <fc308f916aec4e4283e0c1d4b761760a>:0
Jul 04 18:31:01 media mono[1179]: [Fatal] GlobalExceptionHandlers: EPIC FAIL.
Jul 04 18:31:01 media mono[1179]: [v2.0.0.5228] System.NullReferenceException: Object reference not set to an instance of an object.
Jul 04 18:31:01 media mono[1179]:   at System.Collections.Specialized.NameValueCollection.Set (System.String name, System.String value) [0x00000] in <fc308f916aec4e4283e0c1d4b761760a>:0
Jul 04 18:31:01 media mono[1179]: [Warn] MonoDebugCheck: Mono is not running with --debug switch
Jul 04 18:31:01 media mono[1179]: [ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object.
Jul 04 18:31:01 media mono[1179]:   at System.Collections.Specialized.NameValueCollection.Set (System.String name, System.String value) [0x00000] in <fc308f916aec4e4283e0c1d4b761760a>:0
```

Mono ver:
```
5.12.0.226-0xamarin3+debian9b1
```

Fourdee, 2018-07-05:
#### Tests:
- XU4 | need to setup a test system.
- 🈯️ C2
- 🈯️ ASUS TB

Fourdee, 2018-07-05:
@hellfirehd 

I need to setup a XU4 for testing, however, C2 and ASUS TB appear fine. In the mean time please try reinstalling Sonarr:
```
dietpi-software uinstall 144
dietpi-software install 144
```

avraham, 2018-07-05:
Hi,
I experience the same behavior on my Odroid HC1, it doesn't happen all the time though.
I'm running DietPi v6.6

`[FAILED] DietPi-Services | sonarr	failed (Result: signal) since Thu 2018-07-05 09:36:59 PDT; 29s ago
● sonarr.service - Sonarr (NzbDrone) Daemon
   Loaded: loaded (/etc/systemd/system/sonarr.service; enabled)
   Active: failed (Result: signal) since Thu 2018-07-05 09:36:59 PDT; 29s ago
  Process: 707 ExecStart=/usr/bin/mono --debug /opt/NzbDrone/NzbDrone.exe -nobrowser (code=killed, signal=ABRT)
 Main PID: 707 (code=killed, signal=ABRT)

Jul 05 09:36:59 Pi mono[707]: be989000-be9aa000 rwxp 00000000 00:00 0          [stack]
Jul 05 09:36:59 Pi mono[707]: ffff0000-ffff1000 r-xp 00000000 00:00 0          [vectors]
Jul 05 09:36:59 Pi mono[707]: Native stacktrace:
Jul 05 09:36:59 Pi mono[707]: Debug info from gdb:
Jul 05 09:36:59 Pi mono[707]: mono_gdb_render_native_backtraces not supported on this platform, unable to find gdb or lldb
Jul 05 09:36:59 Pi mono[707]: =================================================================
Jul 05 09:36:59 Pi mono[707]: Got a SIGABRT while executing native code. This usually indicates
Jul 05 09:36:59 Pi mono[707]: a fatal error in the mono runtime or one of the native libraries
Jul 05 09:36:59 Pi mono[707]: used by your application.
Jul 05 09:36:59 Pi mono[707]: =================================================================`

Fourdee, 2018-07-05:
@avraham 

Thanks for the report 👍 

This is moving closer and closer to a Mono library/Sonarr code specific issue.

hellfirehd, 2018-07-05:
Immediately after requested reinstall: 
```
root@media:~# systemctl status sonarr
● sonarr.service - Sonarr (NzbDrone) Daemon
   Loaded: loaded (/etc/systemd/system/sonarr.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-07-05 21:40:55 BST; 7s ago
 Main PID: 12542 (mono)
   CGroup: /system.slice/sonarr.service
           └─12542 /usr/bin/mono --optimize=all --server /opt/NzbDrone/NzbDrone.exe -nobrowser

Jul 05 21:40:55 media systemd[1]: Started Sonarr (NzbDrone) Daemon.
Jul 05 21:40:58 media mono[12542]: [Info] Bootstrap: Starting Sonarr - /opt/NzbDrone/NzbDrone.exe - Version 2.0.0.5227
Jul 05 21:41:01 media mono[12542]: [Info] Router: Application mode: Interactive
Jul 05 21:41:02 media mono[12542]: [Info] MigrationLogger: *** Migrating data source=/root/.config/NzbDrone/nzbdrone.db;cache size=-10485760;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
```
After running for about 10 minutes:
```
root@media:~# systemctl status sonarr
● sonarr.service - Sonarr (NzbDrone) Daemon
   Loaded: loaded (/etc/systemd/system/sonarr.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-07-05 21:40:55 BST; 9min ago
 Main PID: 12542 (mono)
   CGroup: /system.slice/sonarr.service
           └─12542 /usr/bin/mono --optimize=all --server /opt/NzbDrone/NzbDrone.exe -nobrowser

Jul 05 21:50:42 media mono[12542]: [Info] ExistingMetadataImporter: Found 0 existing metadata files
Jul 05 21:50:42 media mono[12542]: [Info] ExistingSubtitleImporter: Found 0 existing subtitle files
Jul 05 21:50:42 media mono[12542]: [Info] ExistingOtherExtraImporter: Found 0 existing other extra files
Jul 05 21:50:42 media mono[12542]: [Info] ExistingExtraFileService: Found 0 extra files
Jul 05 21:50:42 media mono[12542]: [Info] RefreshSeriesService: Updating Info for The Flash (2014)
Jul 05 21:50:43 media mono[12542]: [Info] RefreshEpisodeService: Starting episode info refresh for: [279121][The Flash (2014)]
Jul 05 21:50:43 media mono[12542]: [Info] RefreshEpisodeService: Finished episode refresh for series: [279121][The Flash (2014)].
Jul 05 21:50:43 media mono[12542]: [Info] DiskScanService: Scanning disk for The Flash (2014)
Jul 05 21:50:43 media mono[12542]: [Info] MediaCoverService: Downloading Banner for [280619][The Expanse] https://www.thetvdb.com/banners/graphical/280619-g10.jpg
Jul 05 21:50:43 media mono[12542]: [Info] MediaCoverService: Downloading Poster for [280619][The Expanse] https://www.thetvdb.com/banners/posters/5b35d80571416.jpg
```
After first web request: 
```
root@media:~# systemctl status sonarr
● sonarr.service - Sonarr (NzbDrone) Daemon
   Loaded: loaded (/etc/systemd/system/sonarr.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2018-07-05 21:51:49 BST; 9s ago
  Process: 12542 ExecStart=/usr/bin/mono --optimize=all --server /opt/NzbDrone/NzbDrone.exe -nobrowser (code=killed, signal=ABRT)
 Main PID: 12542 (code=killed, signal=ABRT)

Jul 05 21:51:49 media mono[12542]: bec6c000-bec6d000 r-xp 00000000 00:00 0          [vdso]
Jul 05 21:51:49 media mono[12542]: ffff0000-ffff1000 r-xp 00000000 00:00 0          [vectors]
Jul 05 21:51:49 media mono[12542]: Native stacktrace:
Jul 05 21:51:49 media mono[12542]: Debug info from gdb:
Jul 05 21:51:49 media mono[12542]: mono_gdb_render_native_backtraces not supported on this platform, unable to find gdb or lldb
Jul 05 21:51:49 media mono[12542]: =================================================================
Jul 05 21:51:49 media mono[12542]: Got a SIGSEGV while executing native code. This usually indicates
Jul 05 21:51:49 media mono[12542]: a fatal error in the mono runtime or one of the native libraries
Jul 05 21:51:49 media mono[12542]: used by your application.
Jul 05 21:51:49 media mono[12542]: =================================================================
```
I've enabled Debug logging and will report back if it shows anything useful.

hellfirehd, 2018-07-05:
Slightly more interesting fail with Debug logging:
```
root@media:~/.config/NzbDrone/logs# systemctl status sonarr
● sonarr.service - Sonarr (NzbDrone) Daemon
   Loaded: loaded (/etc/systemd/system/sonarr.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2018-07-05 22:00:40 BST; 16s ago
  Process: 13019 ExecStart=/usr/bin/mono --optimize=all --server /opt/NzbDrone/NzbDrone.exe -nobrowser (code=killed, signal=ABRT)
 Main PID: 13019 (code=killed, signal=ABRT)

Jul 05 22:00:40 media mono[13019]: [Debug] Parser: Release Group parsed: thumb
Jul 05 22:00:40 media mono[13019]: [Debug] VideoFileInfoReader: Getting media info from /mnt/media/TV/NCIS/Season 13/NCIS - S13E18 - Scope HDTV-1080p-thumb.jpg
Jul 05 22:00:40 media mono[13019]: EPIC FAIL: System.NullReferenceException: Object reference not set to an instance of an object.
Jul 05 22:00:40 media mono[13019]:   at Nancy.Conventions.StaticContentConventionBuilder.IsWithinContentFolder (System.String contentRootPath, System.String fileName) [0x00000] in <63c44d6f8d7e4f339a987bf9eb94d931>:0
Jul 05 22:00:40 media mono[13019]:   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () <0xb28285b8 + 0x00033> in <71d8ad678db34313b7f718a414dfcb25>:0
Jul 05 22:00:40 media mono[13019]: =================================================================
Jul 05 22:00:40 media mono[13019]: Got a SIGILL while executing native code. This usually indicates
Jul 05 22:00:40 media mono[13019]: a fatal error in the mono runtime or one of the native libraries
Jul 05 22:00:40 media mono[13019]: used by your application.
Jul 05 22:00:40 media mono[13019]: =================================================================
```

Taloth, 2018-07-05:
Hey guys, Sonarr dev here. Just here to drop some observations.

1) I couldn't find it in the report, but check the mono version. If it's 5.12, i'd suggest to try 5.10 also. I haven't heard about crashes on 5.12 but we (and Jackett) have experienced http issues with 5.12.
2) Check if the libmediainfo version is recent, it's a well-known culprit in native memory corruption and crashes. Especially considering hellfirehd's last debug log.   
Sonarr trying to scan a .jpg is unexpected and probably an error on our part, but it shouldn't cause mediainfo to crash.   
I also suspect that mediainfo isn't as extensively tested on armle vs other platforms.
3) We normally run with --debug for diagnostic reasons, but I understand why you'd want to run it a bit more optimized on these kind of devices. Still, it's worth checking if the problem exists without those --server and --optimize=all options. No need to add --debug back in, so normal optimizations would still run.   
On that note, I don't think --server matters much, Sonarr has 3 persistent command threads that do most the work. There is some threadpool usage, but not much. I also wouldn't consider these devices 'servers'.  
4) hellfirehd: Run with Trace log level instead of Debug to get a bit more context around that VideoFileInfoReader. It'll tell us whether mediainfo actually finished doing it's work, or that it crashed during that.

mediainfo is first on my suspect list, followed by --optimize=all. But don't change both at the same time without retesting first.

PS: When posting stdout logs from journalctl, you need more lines. Once mono crashes you'll often get several unrelated 'managed' errors, usually 'NullReferenceException' which tend to obscure the original issue.

hellfirehd, 2018-07-05:
@Taloth, thanks for the tips.  I'll start working through them and see what pops.

What are your thoughts on why it will run fine until the first Web Interface request?

hellfirehd, 2018-07-05:
Set logging to Trace. Removed `--optimize=all` and `--server`.  Didn't touch mediainfo.

Web interface is working and responding.  Everything has been running stable for about 15 minutes.

I was having the same crash with Radarr so I'll see if it will work with the same changes.

Fourdee, 2018-07-06:
@Taloth 

Many thanks for joining this ticket, appreciate the insight and info!

@hellfirehd 

> Set logging to Trace. Removed --optimize=all and --server. Didn't touch mediainfo.
Web interface is working and responding. Everything has been running stable for about 15 minutes.

Excellent, if this proves stable, we will revert the previous patch (which added those options in).

In regards to ```libmediainfo```, we are somewhat limited to whats available in APT:
- Stretch = libmediainfo0v5 (0.7.91-1): https://packages.debian.org/stretch/libmediainfo0v5
- Buster = libmediainfo0v5 (18.03.1-1): https://packages.debian.org/buster/libmediainfo0v5

Its possible the buster package has improved stability and features, however, I doubt it would be possible to install this package on Stretch due to the pre-req's of package dep versions.

Taloth, 2018-07-06:
For mediainfo we usually recommend ppl to use the upstream repo. Coz they continually fix specific edge-cases that can lead to crashes. They do release 18.05 for stretch, but the upstream repo is x86/64 only. Which means you'd have to compile it yourself. Leave it as is till it proves to be a problem.

I checked the default optimizations vs all:

_(DEFAULT is turned on if you run without --debug. All except NO_ALL are turned on with --optimize=all.)_
```
DEFAULT 	peephole	Peephole postpass
DEFAULT 	branch		Branch optimizations
DEFAULT 	inline		Inline method calls
DEFAULT 	cfold		Constant folding
DEFAULT 	consprop	Constant propagation
DEFAULT 	copyprop	Copy propagation
DEFAULT 	deadce		Dead code elimination
DEFAULT 	linears		Linear scan global reg allocation
DEFAULT 	cmov		Conditional moves
DEFAULT 	intrins		Intrinsic method implementations
DEFAULT 	loop		Loop related optimizations
DEFAULT 	aot		Usage of Ahead Of Time compiled code
DEFAULT 	exception	Optimize exception catch blocks
DEFAULT 	gshared		Share generics
DEFAULT 	simd		Simd intrinsics

		sched		Instruction scheduling
		tailc		Tail recursion and tail calls
		fcmov		Fast x86 FP compares
		leaf		Leaf procedures optimizations
		abcrem		Array bound checks removal
		ssapre		SSA based Partial Redundancy Elimination
		ssa		Use plain SSA form
		sse2		SSE2 instructions on x86

NO_ALL		precomp		Precompile all methods before executing Main
NO_ALL		shared		Emit per-domain code
NO_ALL		unsafe		Remove bound checks and perform other dangerous changes
```
fcmov and abcrem would be unwelcome since it changes behavior. And I sincerely doubt any of those non-default optimization would yield a measurable performance improvement.

PS: Don't just revert the commit, coz that would add '--debug' again. Just change it to run without --debug, --optimize, --server. That should be the right trade-off for performance.

hellfirehd, 2018-07-06:
> Excellent, if this proves stable, we will revert the previous patch (which added those options in).

It's been running (and working hard) all night long.  We can give it some more time if you like but I'm satisfied that it's performing as well on the ODroid as it did on my x86_64.

Sadly, the same change did not help Radarr.  It's still crashing like crazy and I suspect it's probably to do with mediainfo.

Taloth, 2018-07-06:
Radarr uses pretty much the same code for calling mediainfo, so get me a ~100 line console log with Trace log level enabled and we should be able to confirm whether mediainfo is involved in those crashes.

Fourdee, 2018-07-07:
@hellfirehd 

Many thanks for confirming resolved 👍 

@Taloth 

> PS: Don't just revert the commit, coz that would add '--debug' again. Just change it to run without --debug, --optimize, --server. That should be the right trade-off for performance.

Yep, i'll get this applied 👍

Fourdee, 2018-07-07:
For v6.12:
- Removed ```--optimize=all --server``` for all our Mono launched applications.

@hellfirehd 

> Radarr uses pretty much the same code for calling mediainfo, so get me a ~100 line console log with Trace log level enabled and we should be able to confirm whether mediainfo is involved in those crashes.

Would be great if you could help assist with this 👍