MichaIng/DietPi #1898: DietPi-Drive_Manager | NFS mounting issue

MichaIng, 2018-07-05:
## Creating a bug report/issue:
Reference: https://dietpi.com/phpbb/viewtopic.php?f=9&t=4169&start=10#p13378

#### Required Information:
- DietPi version	| `6.10`
- Distro version	| `Stretch`
- SBC device		| `RPi2`

#### Additional Information (if applicable):
- Software title	| NFS mount

#### Expected behaviour:

- NFS drive should mount

#### Actual behaviour:

```
root@DietPi:~# mount -t nfs4 -o port=2049 192.168.0.11:/volume1/video /mnt/nfs
mount.nfs4: Protocol not supported
...
root@DietPi:~# mount -t nfs -o port=2049 192.168.0.11:/volume1/video /mnt/nfs
mount.nfs: Protocol not supported
```

#### Extra details:

- Problem got solved with enabling nfs4 support on NAS (Synology).
- We by default add nfs4 to fstab.
- Maybe there is a way to test first or by default add it a more compatible way?
- Sorry can't add much more here, as I never worked with NFS 😉.

MichaIng, 2018-07-05:
@Fourdee 
Maybe we can as well spawn some hint on update, that network drive options have been moved from `dietpi-config` to `dietpi-drive_manager` 🙂.

Fourdee, 2018-07-05:
> Maybe we can as well spawn some hint on update, that network drive options have been moved from dietpi-config to dietpi-drive_manager

Yep 👍 

> Sorry can't add much more here, as I never worked with NFS

Same here.

```auto``` works, unsure if it changes anything, or has any effect.
```
192.168.0.113:/ /mnt/nfs auto auto,_netdev 0  0
```

Fourdee, 2018-07-05:
Hmm

> nfsvers=n
The NFS protocol version number used to contact the server's NFS service. If the server does not support the requested version, the mount request fails. If this option is not specified, the client negotiates a suitable version with the server, trying version 4 first, version 3 second, and version 2 last.

https://linux.die.net/man/5/nfs

MichaIng, 2018-07-05:
@Fourdee 
`The fstype field contains "nfs". Use of the "nfs4" fstype in /etc/fstab is deprecated.` So this we should change at least.

Fourdee, 2018-07-05:
@MichaIng 

> The fstype field contains "nfs". Use of the "nfs4" fstype in /etc/fstab is deprecated. So this we should change at least.

Done, passed testing.

Fourdee, 2018-07-06:
> https://docs.oracle.com/cd/E19253-01/816-4555/rfsrefer-18/index.html

- nfs 2/3 = ```mount -F nfs -r bee:/export/share/man /usr/man```
- nfs 4 = ```mount -F nfs -o vers=4 bee:/export/share/man /usr/man```

Fourdee, 2018-07-06:
@MichaIng 

> Also can't find _netdev option on Debian man page: https://manpages.debian.org/stretch/manpages-de/nfs.5.de.html

http://man7.org/linux/man-pages/man8/mount.8.html

>        _netdev
              The  filesystem  resides  on  a  device  that requires network
              access (used to prevent the system from  attempting  to  mount
              these  filesystems  until  the network has been enabled on the
              system).

MichaIng, 2018-07-06:
@Fourdee 
Jep, also found it on https://wiki.ubuntuusers.de/NFS/, totally reasonable 👍.

Fourdee, 2018-07-07:
@MichaIng 

I'am hoping this user can assist us with testing, hopefully find a solution for NFSv3, then I can implement into drive_man, as per the Samba method.

Fourdee, 2018-07-07:
Notes:

NFS v3 server:
https://unix.stackexchange.com/q/205403/98891
```RPCNFSDCOUNT="8 --no-nfs-version 4"```

duncansimpson, 2018-07-07:
I'm seeing the same thing since upgrading to 6.11 per my post here. 

https://dietpi.com/phpbb/viewtopic.php?f=9&t=4169&start=10#p13427

Happy to provide more details or test fixes.

MichaIng, 2018-07-07:
@duncansimpson 
Thanks for joining here. Good to here again, that it worked pre v6.10.

**Trace again how we added NFS pre v6.10:**
- https://github.com/Fourdee/DietPi/blob/3dd53f29bdcd90b1a51d8efb620bbff2d049524f/dietpi/dietpi-config#L5549
- https://github.com/Fourdee/DietPi/blob/3dd53f29bdcd90b1a51d8efb620bbff2d049524f/dietpi/func/dietpi-set_nfsclient#L44
  - fstab entry: `ip.of.nfs.server:/ /mnt/nfs_client nfs4 auto,_netdev 0  0`, identical to now, besides deprecated `nfs4` entry, which should even cause more issues, esp. in combination with NFS v3.
  - `systemctl daemon-reload`
  - `mount -a`
- Might this daemon-reload simply be needed to reread fstab or something?

**On the other hand I see you guys often have sub paths for NFS server within fstab entry**, e.g. `192.168.1.10:/mnt/user/music_test`
- This can't have been added via `dietpi-config`, as our script handled and added IP only since long time ago. But might this be an issue as well, that some NFS servers just handle access to those sub directories correctly, otherwise refuse connection with no-protocol-support error?
- This would mean we need to allow adding NFS servers with IP and sub path as above.

**Other explanation: APT update of `nfs-common` broke v3 support**
- http://ftp.uk.debian.org/debian/pool/main/n/nfs-utils/
  - Last package update on repo (1.3.4): 21-Feb-2018, on Raspbian most properly later
- Current version: 2.3.2 https://sourceforge.net/projects/nfs/files/nfs-utils/
  - Changelog reveals some fix about version auto negation, but this is just 2.3.1 -> 2.3.2:
```
    Commit 71b807e1 introduce a regression that
    caused v3 not to be tried when v4 was not
    supported by the server during auto negation.
```
- Test could be to install the older package from 2017: http://ftp.uk.debian.org/debian/pool/main/n/nfs-utils/

duncansimpson, 2018-07-09:
@MichaIng 

WRT the subpaths and fstab entries, yes I need to use subpaths in this case, because I have a range of nfs shares on the remote server and I don't need/want all of them visible on my Pi.  I added these entries many months ago, and they have been working fine through numerous upgrades of DietPi until I upgraded from 6.9 to 6.11.  No other changes were made during the most recent upgrade, so I can only suggest that it appears to be related to the APT update of nfs-common.

MichaIng, 2018-07-09:
@duncansimpson 
Not sure, but I think on update the fstab file got recreated, as well as the network drive entries. So the path got removed. Please try to manually re-add it within fstab and `mount -a`.

So we need to allow adding paths for the target server on v6.12, perhaps via separate input field or included within IP input field.

duncansimpson, 2018-07-09:
@MichaIng 

I don't know if the fstab got recreated on upgrade, I didn't check the timestamp on it, however straight after the upgrade I noticed that my remote shares weren't connected.  So I checked fstab and saw my entries _were still there_, intact with the sub paths still there also, so I know the update did not remove the entries and did not remove the subpaths.

`mount -a` doesn't remount the shares.  It also gives the `mount.nfs: Protocol not supported` message for each share (I have 4 in total).

FWIW I also have a mount to a usb local hardrive that I added to fstab which stayed in place after the update.

I later tried the new `dietpi-drive_manager`, which also fails with the same `mount.nfs: Protocol not supported`.  I noticed using that utility _did_ recreate my fstab however.  When it did that, it reordered all the entries, changed the comments, and removed my local USB drive mount from fstab, so I had to re-add that entry manually.

I understand that manually editing fstab is not supported by you guys, and I'm ok with that, however the underlying nfs client seems to be broken since either 6.10 or 6.11.  I'll see if I can find an old SD card tonight and download 6.9 to confirm those entries still work in my specific scenario.

Let me know if there anything else you'd like me to check or test.

MichaIng, 2018-07-09:
@duncansimpson 
Last I can think of: We removed `net-tools` from DietPi core dependencies. Nowadays it is/should be replaced everywhere by `iproute2`, but who knows if somehow old tools/methods use `net-tools` commands, especially as we talk about old NFS versions. Although we did not purge the package, just autoremoved in case it is no dependency of any other installed package, it is worth trying:

```
apt install net-tools
mount -a
```

If this also does not help, my only left idea, as above, is dropped support of bug with latest `nfs-common` patch.
Which device you use?
If non-RPi, please download 1.3.4-2.1 from 2017 from here with correct arch: http://ftp.uk.debian.org/debian/pool/main/n/nfs-utils/
On RPi armhf should work fine as well, but would be better to check for older version on Raspbian repo then.

duncansimpson, 2018-07-09:
Thanks @MichaIng 

I'm at work at the moment (NZ Time), I'll try that tonight (about 8hrs from now). 

FYI I'm using a RPi 3, however I also have an older 2B I can tinker with if necessary, just need to find a spare SD card somewhere.

duncansimpson, 2018-07-10:
@MichaIng 

I tried `apt install net-tools`, no improvement unfortunately:

```
root@DietPi:~# mount -a
mount.nfs: Protocol not supported
mount.nfs: Protocol not supported
mount.nfs: Protocol not supported
mount.nfs: Protocol not supported
root@DietPi:~# apt install net-tools
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  net-tools
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 220 kB of archives.
After this operation, 818 kB of additional disk space will be used.
Get:1 http://mirror.reap.net.nz/raspbian/raspbian stretch/main armhf net-tools armhf 1.60+git20161116.90da8a0-1 [220 kB]
Fetched 220 kB in 1s (216 kB/s)
Selecting previously unselected package net-tools.
(Reading database ... 20705 files and directories currently installed.)
Preparing to unpack .../net-tools_1.60+git20161116.90da8a0-1_armhf.deb ...
Unpacking net-tools (1.60+git20161116.90da8a0-1) ...
Setting up net-tools (1.60+git20161116.90da8a0-1) ...
root@DietPi:~# mount -a
mount.nfs: Protocol not supported
mount.nfs: Protocol not supported
mount.nfs: Protocol not supported
mount.nfs: Protocol not supported
root@DietPi:~#

```

Note as mentioned earlier I have 4 nfs entries in fstab, hence 4 messages.

MichaIng, 2018-07-10:
@duncansimpson 
Okay, so we can sort out `net-tools` 👍.

MichaIng, 2018-07-17:
@duncansimpson 
Which machine/SBC do you use/faced the issue on?
Did you try to manually install/downgrade `nfs-common` package to an earlier version: http://ftp.uk.debian.org/debian/pool/main/n/nfs-utils/

Fourdee, 2018-07-17:
@MichaIng 

The only thing I can suggest (if not already done), is for @duncansimpson  and @mgherghi try mounting all the possible NFS v2/v3 modes, eg:
```
mount -t nfs -o vers=2,port=2049 192.168.0.11:/volume1/video /mnt/nfs
mount -t nfs -o vers=3,port=2049 192.168.0.11:/volume1/video /mnt/nfs
```

Failing that, i'am out of ideas. I have minimal real world NFS experience, only basic configuration and installation.

I did attempt to replicate a v2/v3 server for testing, but unsuccessful: https://github.com/Fourdee/DietPi/issues/1898#issuecomment-403225412. I'll try to give it another stab when time permits.

mgherghi, 2018-07-18:
root@DietPi:/mnt# mount -t nfs -o vers=2,port=2049 192.168.1.1:/Seagates /mnt/nfs
mount: wrong fs type, bad option, bad superblock on 192.168.1.1:/Seagates,
       missing codepage or helper program, or other error
       (for several filesystems (e.g. nfs, cifs) you might
       need a /sbin/mount.<type> helper program)

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
root@DietPi:/mnt# mount -t nfs -o vers=3,port=2049 192.168.1.1:/Seagates /mnt/nfs
mount: wrong fs type, bad option, bad superblock on 192.168.1.1:/Seagates,
       missing codepage or helper program, or other error
       (for several filesystems (e.g. nfs, cifs) you might
       need a /sbin/mount.<type> helper program)

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
root@DietPi:/mnt#

duncansimpson, 2018-07-18:
Hi guys, 

In my case this is all on a Pi 3.

Here's NFSv3 test

```
root@DietPi:~# mount -t nfs -o vers=3 -v 192.168.1.10:/mnt/user/music_test /mnt/test/
mount.nfs: timeout set for Wed Jul 18 19:01:34 2018
mount.nfs: trying text-based options 'vers=3,addr=192.168.1.10'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: trying 192.168.1.10 prog 100003 vers 3 prot TCP port 2049
mount.nfs: prog 100005, trying vers=3, prot=17
mount.nfs: trying 192.168.1.10 prog 100005 vers 3 prot UDP port 33776
mount.nfs: Protocol not supported
root@DietPi:~#
```

And v2 test:

```
root@DietPi:~# mount -t nfs -o vers=2 -v 192.168.1.10:/mnt/user/music_test /mnt/test/
mount.nfs: timeout set for Wed Jul 18 19:04:34 2018
mount.nfs: trying text-based options 'vers=2,addr=192.168.1.10'
mount.nfs: prog 100003, trying vers=2, prot=6
mount.nfs: trying 192.168.1.10 prog 100003 vers 2 prot TCP port 2049
mount.nfs: portmap query retrying: RPC: Program/version mismatch
mount.nfs: prog 100003, trying vers=2, prot=17
mount.nfs: trying 192.168.1.10 prog 100003 vers 2 prot UDP port 2049
mount.nfs: portmap query failed: RPC: Program/version mismatch
mount.nfs: Protocol not supported
root@DietPi:~#
```
Is there a 6.9 image I can revert back to, in order to confirm that still works?

duncansimpson, 2018-07-18:
WRT to the request earlier to rollback to the 2017 version of `nfs-common` which is 1.3.4-2.1.  I can't because that is the version in place already.

```
root@DietPi:~# apt-cache policy nfs-common
nfs-common:
  Installed: 1:1.3.4-2.1
  Candidate: 1:1.3.4-2.1
  Version table:
 *** 1:1.3.4-2.1 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
root@DietPi:~#
```

trying to install the 2018 version, 1.3.4-2.2 fails:

```
root@DietPi:~# apt-get install nfs-common=1:1.3.4-2.2
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Version '1:1.3.4-2.2' for 'nfs-common' was not found
root@DietPi:~#
```

And checking the current version by trying to install it again:

```
root@DietPi:~# apt-get install nfs-common=1:1.3.4-2.1
Reading package lists... Done
Building dependency tree
Reading state information... Done
nfs-common is already the newest version (1:1.3.4-2.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:~#
```

And `apt-get update` doesn't find it either

```
root@DietPi:~# apt-get -u --dry-run upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  ca-certificates dpkg file libmagic-mgc libmagic1 libsystemd0 libudev1 systemd systemd-sysv tzdata udev
11 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Inst dpkg [1.18.24] (1.18.25 Raspbian:stable [armhf])
Conf dpkg (1.18.25 Raspbian:stable [armhf])
Inst libsystemd0 [232-25+deb9u2] (232-25+deb9u4 Raspbian:stable [armhf]) [systemd:armhf ]
Conf libsystemd0 (232-25+deb9u4 Raspbian:stable [armhf]) [systemd:armhf ]
Inst systemd [232-25+deb9u2] (232-25+deb9u4 Raspbian:stable [armhf])
Inst udev [232-25+deb9u2] (232-25+deb9u4 Raspbian:stable [armhf]) []
Inst libudev1 [232-25+deb9u2] (232-25+deb9u4 Raspbian:stable [armhf])
Conf libudev1 (232-25+deb9u4 Raspbian:stable [armhf])
Conf systemd (232-25+deb9u4 Raspbian:stable [armhf])
Inst systemd-sysv [232-25+deb9u2] (232-25+deb9u4 Raspbian:stable [armhf])
Inst tzdata [2018c-0+deb9u1] (2018e-0+deb9u1 Raspbian:stable [all])
Inst file [1:5.30-1+deb9u1] (1:5.30-1+deb9u2 Raspbian:stable [armhf]) []
Inst libmagic1 [1:5.30-1+deb9u1] (1:5.30-1+deb9u2 Raspbian:stable [armhf]) []
Inst libmagic-mgc [1:5.30-1+deb9u1] (1:5.30-1+deb9u2 Raspbian:stable [armhf])
Inst ca-certificates [20161130+nmu1] (20161130+nmu1+deb9u1 Raspbian:stable [all])
Conf udev (232-25+deb9u4 Raspbian:stable [armhf])
Conf systemd-sysv (232-25+deb9u4 Raspbian:stable [armhf])
Conf tzdata (2018e-0+deb9u1 Raspbian:stable [all])
Conf file (1:5.30-1+deb9u2 Raspbian:stable [armhf])
Conf libmagic1 (1:5.30-1+deb9u2 Raspbian:stable [armhf])
Conf libmagic-mgc (1:5.30-1+deb9u2 Raspbian:stable [armhf])
Conf ca-certificates (20161130+nmu1+deb9u1 Raspbian:stable [all])
root@DietPi:~#

```

MichaIng, 2018-07-18:
@duncansimpson 
Okay, Raspbian is one version behind then. Change that this was the reason anyway converged to zero.

To test v6.9 handling please do the following:
```
cp -a /DietPi/dietpi/dietpi-drive_manager ~/
cp -a /etc/fstab ~/
wget https://raw.githubusercontent.com/Fourdee/DietPi/f7170dc0f125c83d9230daa9237076d92ab22d50/dietpi/dietpi-drive_manager -O /DietPi/dietpi/dietpi-drive_manager
wget https://raw.githubusercontent.com/Fourdee/DietPi/f7170dc0f125c83d9230daa9237076d92ab22d50/dietpi/func/dietpi-set_nfsclient -O /DietPi/dietpi/func/dietpi-set_nfsclient
chmod +x /DietPi/dietpi/dietpi-drive_manager /DietPi/dietpi/func/dietpi-set_nfsclient
/DietPi/dietpi/func/dietpi-set_nfsclient
```
Enter IP and in case adjust the /etc/fstab entry with the full path info.
That would recover at least fstab and nfs drive handling of v6.9.
To revert:
```
cp -a ~/dietpi-drive_manager /DietPi/dietpi/dietpi-drive_manager
cp -a ~/fstab /etc/fstab
rm /DietPi/dietpi/func/dietpi-set_nfsclient
rm /boot/dietpi/func/dietpi-set_nfsclient
```

To fully revert to v6.9, if you have no backup somewhere, would be to download and flash the current available RPi DietPi image: https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Stretch.7z
Backup current SD card or simply replace with another one for testing of course.

duncansimpson, 2018-07-18:
> To fully revert to v6.9, if you have no backup somewhere, would be to download and flash the current available RPi DietPi image: https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Stretch.7z
Backup current SD card or simply replace with another one for testing of course.

Thanks @MichaIng.  I have a spare SD card so I'll try that tonight.

duncansimpson, 2018-07-19:
I dug out my spare SD card, put the 6.9 image on there, let it do its thing after first boot, completed the install, then installed nfs-client via dietpi-software and rebooted.

Results of nfs connection test were successful:

```
root@DietPi:~# apt-cache policy nfs-common
nfs-common:
  Installed: 1:1.3.4-2.1
  Candidate: 1:1.3.4-2.1
  Version table:
 *** 1:1.3.4-2.1 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
root@DietPi:~# mount -t nfs -v 192.168.1.10:/mnt/user/music_test /mnt/test/
mount.nfs: timeout set for Thu Jul 19 09:08:00 2018
mount.nfs: trying text-based options 'vers=4.2,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Invalid argument
mount.nfs: trying text-based options 'vers=4.1,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'vers=4.0,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'addr=192.168.1.10'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: trying 192.168.1.10 prog 100003 vers 3 prot TCP port 2049
mount.nfs: prog 100005, trying vers=3, prot=17
mount.nfs: trying 192.168.1.10 prog 100005 vers 3 prot UDP port 33776
root@DietPi:~# ls -l /mnt/test
total 8
drwxrwxrwx 1 99 users   36 Jan  7  2017 2L
drwxrwxrwx 1 99 users 4096 Dec 11  2014 linn 24bits of xmas
drwxrwxrwx 1 99 users 4096 Dec 16  2013 Stereophile Test CD, Vol. 1
drwxrwxrwx 1 99 users   43 Aug  7  2017 Various Artists
root@DietPi:~#
```

Note this is the same version of nfs-common as was present on the 6.11 SD card.

Next I upgraded DietPi on the spare SD card from 6.9 to 6.11 and redid the NFS test.  It now fails, despite the nfs-common package being the same version:

```
 ───────────────────────────────────────
 DietPi     | 09:45 | Thu 19/07/18
 ───────────────────────────────────────
 v6.11      | RPi B (armv6l)
 ───────────────────────────────────────
 eth0       | 192.168.1.21
 ───────────────────────────────────────
 DietPi Core Team   : Daniel Knight (founder), MichaIng, K-Plan
 Image creator      : DietPi Core Team
 Pre-image used     : Raspbian Lite
 Web                : https://DietPi.com | https://twitter.com/dietpi_
 Donate             : https://goo.gl/pzISt9
 DietPi web hosting : Powered by https://MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

root@DietPi:~# apt-cache policy nfs-common
nfs-common:
  Installed: 1:1.3.4-2.1
  Candidate: 1:1.3.4-2.1
  Version table:
 *** 1:1.3.4-2.1 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status
root@DietPi:~# mount -t nfs -v 192.168.1.10:/mnt/user/music_test /mnt/test/
mount.nfs: timeout set for Thu Jul 19 09:48:10 2018
mount.nfs: trying text-based options 'vers=4.2,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Invalid argument
mount.nfs: trying text-based options 'vers=4.1,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'vers=4.0,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'addr=192.168.1.10'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: trying 192.168.1.10 prog 100003 vers 3 prot TCP port 2049
mount.nfs: prog 100005, trying vers=3, prot=17
mount.nfs: trying 192.168.1.10 prog 100005 vers 3 prot UDP port 33776
mount.nfs: Protocol not supported
root@DietPi:~#
```

So that confirms (for me at least) that something is different between 6.9 and 6.11 WRT NFS client connectivity, and yet it appears to be unrelated to the nfs-common package version at least.

duncansimpson, 2018-07-19:
Note the tests I just did were on an old Pi Model B connected to the same switch as the Pi3, so I think that also rules out hardware fault or network issues.

MichaIng, 2018-07-19:
@duncansimpson 
Great test.

That also proves, that it is totally outside of `dietpi-drive_manager` or the way we attempt(ed) the mount. So no need to check in this direction. Nice also to see how `mount.nfs` itself tries all versions from 4.2 down to 3, UDP and TCP.

How did you prevent the automated `dietpi-update` on first boot? Was there some APT upgrade done before you tested it on 6.9, or better were there some packages upgraded on dietpi-update to v6.11? This had no effect on `nfs-common`, as current version (on Raspbian) is dated back a while, but maybe some other involved package upgrade might be the reason?

I will now set up an NFSv3 server myself and play around: https://unix.stackexchange.com/questions/205403/disable-nfsv4-server-on-debian-allow-nfsv3
Unbelievable that the reason is so hard to find here 🤔 😓, and what a chance that it appeared the same version jump where er reworked drive_manager and NFS mount method, and still it is not related to that. At least this made me investigating into the wrong direction for quite a while...

Fourdee already moved this issue to v6.13 milestone. I will give my test today to identify the reason, but as we want to release v6.12 today (due to some serious cron issue), the chance is low that this will be included. However as always, manual fix would be possible 😉.

MichaIng, 2018-07-19:
Took a while until I found `netbase` to be needed for NFS server to work. On Jessie this is not installed as dependency of nfs-kernel-server, as on Stretch+.
Taking into account the config bug within the link above, I have now a NFSv3 only server on a Jessie VM and could replicate the connection issue from another Stretch VM.

Installing `netbase` as well on the NFS client side, the connection fails differently:
```
root@VM-Stretch:~# mount -t nfs -v 192.168.178.28:/ /mnt/VM-Jessie
mount.nfs: timeout set for Thu Jul 19 13:38:58 2018
mount.nfs: trying text-based options 'vers=4.2,addr=192.168.178.28,clientaddr=192.168.178.29'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'vers=4.1,addr=192.168.178.28,clientaddr=192.168.178.29'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'vers=4.0,addr=192.168.178.28,clientaddr=192.168.178.29'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'addr=192.168.178.28'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: trying 192.168.178.28 prog 100003 vers 3 prot TCP port 2049
mount.nfs: prog 100005, trying vers=3, prot=17
mount.nfs: trying 192.168.178.28 prog 100005 vers 3 prot UDP port 46199
mount.nfs: mount(2): Permission denied
mount.nfs: access denied by server while mounting 192.168.178.28:/
```

@duncansimpson 
Could you try as well `apt install netbase`?

MichaIng, 2018-07-19:
Okay added commit. I hope it solves the issue for you guys as well: https://github.com/Fourdee/DietPi/commit/f56c1803c16256a782bfece3ebde8090a810bd47

@Fourdee 
Ideally we somehow need to check within drive_manager if NFS server is v4 or v3 (and below) to additionally let user add path to NFS export conditionally, as:
- With path, client will always connect via NFSv3, even that NFSv4 is supported.
- Without path, client will not connect at all to NFSv3 (permissions issue).

duncansimpson, 2018-07-19:
Thanks @MichaIng, I did a quick test adding `netbase` this morning, on my 6.11 test Pi, and that also fixes the problem for me:

```
root@DietPi:~# apt-get install netbase
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  netbase
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 19.1 kB of archives.
After this operation, 45.1 kB of additional disk space will be used.
Get:1 http://mirror.reap.net.nz/raspbian/raspbian stretch/main armhf netbase all 5.4 [19.1 kB]
Fetched 19.1 kB in 1s (18.3 kB/s)
Selecting previously unselected package netbase.
(Reading database ... 16870 files and directories currently installed.)
Preparing to unpack .../archives/netbase_5.4_all.deb ...
Unpacking netbase (5.4) ...
Setting up netbase (5.4) ...
root@DietPi:~# mount -t nfs -v 192.168.1.10:/mnt/user/music_test /mnt/test/
mount.nfs: timeout set for Thu Jul 19 20:40:53 2018
mount.nfs: trying text-based options 'vers=4.2,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Invalid argument
mount.nfs: trying text-based options 'vers=4.1,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'vers=4.0,addr=192.168.1.10,clientaddr=192.168.1.21'
mount.nfs: mount(2): Protocol not supported
mount.nfs: trying text-based options 'addr=192.168.1.10'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: trying 192.168.1.10 prog 100003 vers 3 prot TCP port 2049
mount.nfs: prog 100005, trying vers=3, prot=17
mount.nfs: trying 192.168.1.10 prog 100005 vers 3 prot UDP port 33776
root@DietPi:~# ls -l /mnt/test
total 8
drwxrwxrwx 1 99 users   36 Jan  7  2017 2L
drwxrwxrwx 1 99 users 4096 Dec 11  2014 linn 24bits of xmas
drwxrwxrwx 1 99 users 4096 Dec 16  2013 Stereophile Test CD, Vol. 1
drwxrwxrwx 1 99 users   43 Aug  7  2017 Various Artists
root@DietPi:~#
```

WRT:

> Just found the system boot hanging, when NFS server is down. Shall we add nofail to network drives as well, otherwise noauto,x-systemd.automount? Just tested, both is enough to avoid hanging. noauto,x-systemd.automount sadly does not auto mount NFS drive on boot, although this is how it should work: https://wiki.archlinux.de/title/Fstab

I also found I had to add `nofail` in the fstab entries I had added manually, to prevent boot hangs if the remote share is down.  So my manual fstab entries use `auto,nofail`.

MichaIng, 2018-07-19:
Okay, `nofail` added to network drives as well. Only thing that is outstanding is:
- NFSv3 needs the path to the export dir added to `mount` command and `fstab` entry.
- NFSv4 needs just `<IP>:/`. If path is added, v4 connection attempts will fails and connection will be v3. If server is v4 only, then connection will completely fail.

Thus, currently `dietpi-drive_manager` cannot be used to add v3 drives directly (uses v4 method). To enable this we would need to identify, which version the server is, properly just via test mounting and catch failure, and in case ask user to verify v3 server enter path as well... Some testing, some good thoughts and ideas needed, thus moved to v6.13 😄.

I will close this issue, as initial problem is solved.

closed 2018-07-19 by MichaIng

Joulinar, 2020-03-26:
@MichaIng 
Not sure if this is the right place or if I should open a new issue but we have a report on our forum about failed NFS mounts

https://dietpi.com/phpbb/viewtopic.php?f=9&t=7422

Drive Manager creates an `/etc/fstab` entry that leads to the fact that the NFS share is mounted during boot. On my test this fails as well because eth interface was not yet online. 

So I see 2 solutions
1. execute NFS mounts after eth interface is online
2. change mount options in `/etc/fstab` to `nfs _netdev,nofail,noauto,x-systemd.automount 0 0`

MichaIng, 2020-03-26:
@Joulinar 
Many thanks for reporting. Jep I saw it on forum as well. Very strange since `noauto` should not make it mount on boot and `_netdev` should force it to wait with mount for network at least. Probably these two options do not work well together in all cases 🤔.

Could you check `journalctl -u mnt-mountpoint.mount; journalctl -u mnt-mountpoint.automount` (replacing "mountpoint" with the actual one of course)? When having the mount timestamps we're probably able to find what is accessing the mount point or might otherwise force the automount.

Joulinar, 2020-03-26:
@MichaIng 
The NFS mount is done right in between assigning IP address using DHCP. Means between `DHCPDISCOVER `and `DHCPOFFER`

```
root@DietPi3:~# journalctl -u mnt-nfs_client.mount
-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Thu 2020-03-26 10:11:41 CET. --
Mär 26 10:10:42 DietPi3 systemd[1]: Mounting /mnt/nfs_client...
Mär 26 10:10:42 DietPi3 mount[427]: mount.nfs: Network is unreachable
Mär 26 10:10:42 DietPi3 systemd[1]: mnt-nfs_client.mount: Mount process exited, code=exited, status=32/n/a
Mär 26 10:10:42 DietPi3 systemd[1]: mnt-nfs_client.mount: Failed with result 'exit-code'.
Mär 26 10:10:42 DietPi3 systemd[1]: Failed to mount /mnt/nfs_client.
```

MichaIng, 2020-03-26:
@Joulinar 
Okay, so my guess is that the `_netdev` overrides the `noauto` option. If you remove _netdev from fstab and leave noauto,x-systemd.automount only, does it work then?

`_netdev` AFAIK result in an After=network.target or After=network-online.target in the `systemctl cat mnt-mountpoint.mount` systemd unit, and sadly `After=network-online.target` does not result in what one expects. The /etc/network/interfaces entries with allow-hotplug are not handled by networking.service but via hotplug udev rules, which create ifup@eth0.service like services once the network adapter is ready. These services are Type=simple (default) which means that network-online.target is already reached when ifup STARTS to do something instead of when it has finished. So in case of DHCP it takes a while until interface setup has finished while the target is reached earlier hence the network mount is attempted too early.

I am thinking if we should override `ifup@<iface>.service` to be Type=oneshot for all configured network interfaces, which gives network-online.target much more meaning and makes our network check in dietpi-boot basically obsolete. We could instead simply add After=network-online.target to dietpi-boot.service then. Would resolve some other related issues.

Debian btw had Type=oneshot (or forking) in the past, but they removed it because in some cases it leads to hanging boot, when some interface fails or hangs on configure. But actually it simply makes sense to hang boot when anything depends on network. It will by default hang for 90 seconds max and we can even adjust this to wait for max 10 seconds instead or similar (AFAIK), e.g. editable by user.
______________
Okay long story short, if removing `_netdev` from fstab works fine already, this is the fix that we can apply for now.

Joulinar, 2020-03-26:
@MichaIng 
just to avoid any confusion. The original fstab entry is `nfs auto,_netdev,nofail`. So it's `auto `instead of `noauto`. I changed it as follow:

from: `nfs auto,_netdev,nofail`
to `nfs _netdev,nofail,noauto,x-systemd.automount`

I used a SAMBA entry as example that is created by Drive Manager

MichaIng, 2020-03-26:
@Joulinar 
Ahhh, dammit we added automount for samba only. Needs to be added for NFS reasonably as well. Okay with automount the issue is solved then?

Joulinar, 2020-03-26:
yep correct, changing to same syntax like samba is fixing it. What I'm not sure about is using the dump and fsck numbers at the end like 

`nfs _netdev,nofail,noauto,x-systemd.automount 0 0`

At least I have seen it on this old forum entry https://dietpi.com/phpbb/viewtopic.php?p=13341#p13341

MichaIng, 2020-03-26:
@Joulinar 
The two numbers can indeed be omitted as they default to 0 anyway then. Okay great, not sure why we did not add this together with Samba that time, it definitely makes sense anyway to not open/keep-up a network share connection before it is actually used.

Joulinar, 2020-03-26:
I guess sooner or later I would need to learn how to create RP's on Git. That way we don't need to wast your time always for such simple thinks.

MichaIng, 2020-03-26:
@Joulinar 
Lol we had this already, but missed the NFS pre-v3 fstab entry while the NFSv3 entry contained the automount + noauto already: https://github.com/MichaIng/DietPi/commit/173589be5b131a015fd31153bef91bcac05ef021