MichaIng/DietPi #1900: [FAILED] DietPi-Software | G_AGI: mariadb-server

zicmama, 2018-07-05:
#### Details:
- Date           | Fri  6 Jul 00:02:58 CEST 2018
- Bug report     | N/A
- DietPi version | v6.10
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1123 SMP Wed Jun 27 17:35:49 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: mariadb-server
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
Fresh install, Adding apps: Kodi / nextcloud / mopidy / syncthing / wordpress / lighttps + Mariadb
Error occurs:

dpkg: error processing package mariadb-server-10.1 (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of mariadb-server:
 mariadb-server depends on mariadb-server-10.1 (>= 10.1.23-9+deb9u1); however:
  Package mariadb-server-10.1 is not configured yet.

dpkg: error processing package mariadb-server (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.24-11+deb9u3) ...
Errors were encountered while processing:
 mariadb-server-10.1
 mariadb-server
E: Sub-process /usr/bin/dpkg returned an error code (1)

[FAILED] DietPi-Software | G_AGI: mariadb-server

##########################"
root@DietPi:~# systemctl status mysql.service
● mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql; generated; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2018-07-06 00:10:33 CEST; 10s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 8014 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)

juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: 0 processes alive and '/usr/bin/mysqladmin --defaults-file=/etc/mysql/debian.cnf ping' resulte
d in
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: [61B blob data]
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: error: 'Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2 "N
o such file or directory")'
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: Check that mysqld is running and that the socket: '/var/run/mysqld/mysqld.sock' exists!
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: 
juil. 06 00:10:33 DietPi mysql[8014]: Starting MariaDB database server: mysqld . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . faile
d!
juil. 06 00:10:33 DietPi systemd[1]: mysql.service: Control process exited, code=exited status=1
juil. 06 00:10:33 DietPi systemd[1]: Failed to start LSB: Start and stop the mysql database server daemon.
juil. 06 00:10:33 DietPi systemd[1]: mysql.service: Unit entered failed state.
juil. 06 00:10:33 DietPi systemd[1]: mysql.service: Failed with result 'exit-code'.

###############################
root@DietPi:~# journalctl
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: 0 processes alive and '/usr/bin/mysqladmin --defaults-file=/etc/mysql/debian.cnf ping' resulte
d in
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: [61B blob data]
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: error: 'Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2 "N
o such file or directory")'
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: Check that mysqld is running and that the socket: '/var/run/mysqld/mysqld.sock' exists!
juil. 06 00:10:33 DietPi /etc/init.d/mysql[8502]: 
juil. 06 00:10:33 DietPi mysql[8014]: Starting MariaDB database server: mysqld . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . faile
d!
juil. 06 00:10:33 DietPi systemd[1]: mysql.service: Control process exited, code=exited status=1
juil. 06 00:10:33 DietPi systemd[1]: Failed to start LSB: Start and stop the mysql database server daemon.
juil. 06 00:10:33 DietPi systemd[1]: mysql.service: Unit entered failed state.
juil. 06 00:10:33 DietPi systemd[1]: mysql.service: Failed with result 'exit-code'.

MichaIng, 2018-07-05:
@zicmama 
Thanks for the report.
- Do you have your dietpi_userdata moved to an external drive?
- If yes, has the external drive a dedicated power supply, or is it powered just via USB from the RPi?
- Could you please try the following:
```
chown mysql:mysql -h /var/lib/mysql
chown mysql:mysql -RL /var/lib/mysql
apt-get install mariadb-server
```

@Fourdee 
Report number 4, I don't think any more that it's an hardware related issue, although it's a fact that we so far were not able to reproduce the error...

zicmama, 2018-07-06:
Yes, dietpi_userdata is moved to an external drive (usb self powered ssd)

```
root@DietPi:~# chown mysql:mysql -h /var/lib/mysql
root@DietPi:~# chown mysql:mysql -RL /var/lib/mysql
root@DietPi:~# apt-get install mariadb-server
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
mariadb-server is already the newest version (10.1.23-9+deb9u1).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
2 partiellement installés ou enlevés.
Après cette opération, 0 o d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] 
Paramétrage de mariadb-server-10.1 (10.1.23-9+deb9u1) ...
dpkg: erreur de traitement du paquet mariadb-server-10.1 (--configure) :
 le sous-processus script post-installation installé a retourné une erreur de sortie d'état 1
dpkg: des problèmes de dépendances empêchent la configuration de mariadb-server :
 mariadb-server dépend de mariadb-server-10.1 (>= 10.1.23-9+deb9u1) ; cependant :
 Le paquet mariadb-server-10.1 n'est pas encore configuré.

dpkg: erreur de traitement du paquet mariadb-server (--configure) :
 problèmes de dépendances - laissé non configuré
Des erreurs ont été rencontrées pendant l'exécution :
 mariadb-server-10.1
 mariadb-server
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

zicmama, 2018-07-06:
I also have voltage alerts
```
root@DietPi:~# dmesg 
[    6.583926] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[   14.561460] Voltage normalised (0x00000000)
[  121.872370] systemd-fstab-generator[2065]: Ignoring automount option for root device
[  122.113357] systemd-fstab-generator[2076]: Ignoring automount option for root device
[  122.403023] systemd-fstab-generator[2087]: Ignoring automount option for root device
[  122.673083] systemd-fstab-generator[2098]: Ignoring automount option for root device
[  122.946617] systemd-fstab-generator[2109]: Ignoring automount option for root device
[  123.196254] systemd-fstab-generator[2120]: Ignoring automount option for root device
[  143.953441] systemd-fstab-generator[2943]: Ignoring automount option for root device
[  180.951565] Under-voltage detected! (0x00050005)
[  193.431467] Voltage normalised (0x00000000)
[  222.551496] Under-voltage detected! (0x00050005)
[  228.791514] Voltage normalised (0x00000000)
[  261.409981] systemd-fstab-generator[3385]: Ignoring automount option for root device
[  881.911520] rpi_firmware_get_throttled: 3 callbacks suppressed
[  881.911527] Under-voltage detected! (0x00050005)
[  886.071511] rpi_firmware_get_throttled: 3 callbacks suppressed
[  886.071517] Voltage normalised (0x00000000)
[15666.561601] Under-voltage detected! (0x00050005)
[15676.951510] Voltage normalised (0x00000000)
[15697.751586] Under-voltage detected! (0x00050005)
[15703.991546] Voltage normalised (0x00000000)
[16296.791555] Under-voltage detected! (0x00050005)
[16300.951556] Voltage normalised (0x00000000)
[62828.471717] Under-voltage detected! (0x00050005)
[62840.951677] Voltage normalised (0x00000000)
[62843.031713] Under-voltage detected! (0x00050005)
[62855.511757] Voltage normalised (0x00000000)
[62870.071711] Under-voltage detected! (0x00050005)
[62876.311848] Voltage normalised (0x00000000)
```

MichaIng, 2018-07-06:
@zicmama 
Okay the under-voltage error then again points to the direction of insufficient power supply, as we thought on the first bug reports of this kind.

With "self-powered" you mean the SSD has an own power supply, or does it get it's power over USB?

If it has no own power supply, then I suggest you plug the SSD via docking station with own power plug. Maximum current, that the Pi can supply via USB is `1.2A`, which is not very much, enough for USB sticks usually but not for e.g. 2,5" drives. Also the power supply of the Pi itself should be one of quality. The really cheap ones or from old phones e.g. might from the numbers allow enough current, but cannot hold the voltage stable on fast rising load. So when the power load rises fast, the voltage drops (maybe just for short) and leads to random hardware parts failing, e.g. USB drives.

If the external drive has an own power supply already, then clearly the Pi's power supply needs to be changed.
I am not sure, how USB current is used, in case of dedicated power supply, would be a test to disable "Max USB current" in "dietpi-config > Performance Options". Perhaps somehow the drive leaches power from the Pi via USB, even that is doesn't need to. At least it is worth testing 😉.

Fourdee, 2018-07-16:
I'll mark this as closed, as the issue is hardware (PSU/SD) related, and, we are unable to replicate in our testing.

However @zicmama , please reopen if required.

closed 2018-07-16 by Fourdee

zicmama, 2018-07-22:
I also have voltage alerts
root@DietPi:~# dmesg 
[    6.583926] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[   14.561460] Voltage normalised (0x00000000)
[  121.872370] systemd-fstab-generator[2065]: Ignoring automount option for root device
[  122.113357] systemd-fstab-generator[2076]: Ignoring automount option for root device
[  122.403023] systemd-fstab-generator[2087]: Ignoring automount option for root device
[  122.673083] systemd-fstab-generator[2098]: Ignoring automount option for root device
[  122.946617] systemd-fstab-generator[2109]: Ignoring automount option for root device
[  123.196254] systemd-fstab-generator[2120]: Ignoring automount option for root device
[  143.953441] systemd-fstab-generator[2943]: Ignoring automount option for root device
[  180.951565] Under-voltage detected! (0x00050005)
[  193.431467] Voltage normalised (0x00000000)
[  222.551496] Under-voltage detected! (0x00050005)
[  228.791514] Voltage normalised (0x00000000)
[  261.409981] systemd-fstab-generator[3385]: Ignoring automount option for root device
[  881.911520] rpi_firmware_get_throttled: 3 callbacks suppressed
[  881.911527] Under-voltage detected! (0x00050005)
[  886.071511] rpi_firmware_get_throttled: 3 callbacks suppressed
[  886.071517] Voltage normalised (0x00000000)
[15666.561601] Under-voltage detected! (0x00050005)
[15676.951510] Voltage normalised (0x00000000)
[15697.751586] Under-voltage detected! (0x00050005)
[15703.991546] Voltage normalised (0x00000000)
[16296.791555] Under-voltage detected! (0x00050005)
[16300.951556] Voltage normalised (0x00000000)
[62828.471717] Under-voltage detected! (0x00050005)
[62840.951677] Voltage normalised (0x00000000)
[62843.031713] Under-voltage detected! (0x00050005)
[62855.511757] Voltage normalised (0x00000000)
[62870.071711] Under-voltage detected! (0x00050005)
[62876.311848] Voltage normalised (0x00000000)

zicmama, 2018-07-22:
Installed with a 2.5 Amp power. And I still get same error...
my install list
[image]

MichaIng, 2018-07-22:
@zicmama 
If the SSD is 2,5" and is not attached via docking station with dedicated power plug, then I guess it is nearly impossible to run it successfully on RPi, regardless the RPi power supply. The SBC can simply not forward the needed power to the drive, to have it run stable. USB-only powered works reliable for USB sticks and external SD cards only.

I personally use something similar like this for my HDDs attached to RPi: https://szjmpics.photoebucket.com/image/watermark/EB/22/us/121070/normal/121070-11.jpg