MichaIng/DietPi #1902: SW Request: Rocket Chat and Nextcloud integration

kevkha, 2018-07-06:
Hello and thanks for DietPi. I was able to get Nextcloud up and running with DietPi on RPi3. Looks like Rocket Chat and Nextcloud can be integrated according to this https://github.com/RocketChat/Rocket.Chat/issues/7791

It would be nice to have Rocket Chat added to DietPi-Software.
Thanks in advance.

MichaIng, 2018-07-06:
@kevkha 
Thanks for your request. Huiii, it seems that there are several different installation methods depending on the device/system. We could need to stick with a method, that is similar on all devices: https://rocket.chat/docs/installation/manual-installation/debian/

First we need to add `MongoDB`, the other prereqs are already supported by `dietpi-software`.
The following steps need much testing with all devices, as manual building/node/npm often depends on distro version, CPU architecture and APT repo.

The integration into Nextcloud according to your link is so far more experimental, so until there is an official app, I think we would leave this to end user.

But as mentioned within most other software request issues, we are currently due to man power concentration on maintaining available software titles and DietPi core itself. Also as `MongoDB` will be needed first step, I will mark this as low priority for now.

kevkha, 2018-07-06:
Thanks @MichaIng. Sounds good. I can help with deployment and testing as needed.

MichaIng, 2018-08-02:
MongoDB only officially provided for `amd64` devices: 
https://docs.mongodb.com/manual/tutorial/install-mongodb-on-debian/
https://repo.mongodb.org/apt/debian/dists/stretch/mongodb-org/4.0/main/

Debian maintained package are available for `arm64` as well at least: https://packages.debian.org/de/stretch/mongodb-server

On Buster repo there are packages for all architectures, but would need testing: https://packages.debian.org/de/buster/mongodb-server

Ah, Raspbian provides an armhf package, although old version:
```
2018-08-02 17:32:06 root@micha:/var/log# apt search mongodb-server
Sorting... Done
Full Text Search... Done
mongodb-server/testing 1:2.4.14-4 armhf
  object/document-oriented database (server package)
```

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+8