MichaIng/DietPi #1905: Harddrives automount problem

roelfoubert, 2018-07-07:
I changed to boot my pi on usb. But now the other drives connected via usb wont be mounted. they are connected but not selectable in dietpi-drive_manager. How can I fix this?

[Disks.txt](https://github.com/Fourdee/DietPi/files/2172714/Disks.txt)

Fourdee, 2018-07-07:
@roelfoubert 

Hi,

Thanks for the report 👍 

We'll need to get some more info.

Please paste results:
```
blkid
df -h
cat /etc/fstab
```

roelfoubert, 2018-07-07:
The drives I use are the same I used before I boot on a usb drive so I know they work.
One is another usb stick and the other is an usb powerd external harddrive.
[drives 2.txt](https://github.com/Fourdee/DietPi/files/2172947/drives.2.txt)

roelfoubert, 2018-07-07:
I copied the partitions from my old sd card to my usb with gparted and everything worked like previous

Fourdee, 2018-07-07:
@roelfoubert 

Thanks for additional logs 👍 

Its picking the items up as unmounted:
```
#UUID=7a6c1e4f-4747-00f8-73ca-64b68bfe6a93 /mnt/7a6c1e4f-4747-00f8-73ca-64b68bfe6a93 auto defaults,x-systemd.automount,noatime,rw,nofail 0 0
#UUID=f1d4d67e-2e08-4ac4-8143-828eb95b42ec /mnt/f1d4d67e-2e08-4ac4-8143-828eb95b42ec auto defaults,x-systemd.automount,noatime,rw,nofail 0 0
```

If you run ```dietpi-drive_manager```, do the drives appear in the list (if not, please can you take a screenshot). If they appear in the list, simply select each drive, and select ```Mount```.

DietPi-Drive_Manager no longer automatically mounts newly detected drives. They must be done manually.

roelfoubert, 2018-07-07:
The drives didn't appear in the drive manager menu but I can't take a screenshot because I fixed it by cloning a previous sd card

Fourdee, 2018-07-07:
@roelfoubert 

Ok, i'll try to setup a device that matches your setup, see if we can replicate.

Note to self:
- Lacks primary partition ```/dev/sdX``` on all drives?

Fourdee, 2018-07-07:
@roelfoubert 

I've enabled some debugging output for drive_manager, please can you download the updated script:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/faf662194bccd0530f4d01abac458af07cd7e3a8/dietpi/dietpi-drive_manager -O /DietPi/dietpi/dietpi-drive_manager
```

Then run in debug mode:
```
DEBUG=1 dietpi-drive_manager
```

Once completed, it will print out lots of drive information. Please copy and paste everything you see into this ticket. With this info, would should be able to debug/resolve much quicker.

roelfoubert, 2018-07-07:
Now everything works so I don't think you will see the problem in the logs. I wanted to restore a backup on a new installation on a usb stick and then I had the problem.

Fourdee, 2018-07-08:
@roelfoubert 

> Now everything works so I don't think you will see the problem in the logs. I wanted to restore a backup on a new installation on a usb stick and then I had the problem.

Ah I see. Yep, this will restore a lower version of DietPi, and, your previous ```/etc/fstab``` entries which may of caused the issue.

Ideally, backups should be only created/restored on the same system. Restoring a backup of another system with different hardware/devices, may create issues if they do not exist on the restored system.

I'll mark this as closed, please reopen if required.

closed 2018-07-08 by Fourdee