MichaIng/DietPi #1910: remove ssl-certification

Solmester, 2018-07-08:
Where do "dietpi-letsencrypt" store the certificate? i need to remove it.

MichaIng, 2018-07-08:
@Solmester 
`/etc/letsencrypt` default of certbot.
But note that webservers will fail afterwards. You would need to recover non-SSL configuration in case.

Solmester, 2018-07-08:
I am looking in the Configuration folder of lighttpd ( /etc/lighttpd/lighttpd.conf) and don't see where to remove ssl, it it the right path?

MichaIng, 2018-07-08:
On Lighttpd to revert:
```
rm /etc/lighttpd/conf-enabled/letsencrypt.conf
rm /etc/lighttpd/conf-enabled/redirect.conf
lighttpd-disable-mod dietpi-hsts
rm /etc/lighttpd/conf-available/99-dietpi-hsts.conf
rm /etc/systemd/system/certbot.service.d/dietpi-lighttpd.conf
systemctl restart lighttpd
```

Solmester, 2018-07-08:
Thank you very much for the help!

Solmester, 2018-07-08:
Btw: now it force me to use https on "https://192.168.1.13/owncloud" but not on "http://192.168.1.13/wordpress", is there a setting i need to change?

MichaIng, 2018-07-09:
@Solmester 
Hmm, did you enable HSTS (did `lighttpd-disable-mod dietpi-hsts`/`rm /etc/lighttpd/conf-available/99-dietpi-hsts.conf` complain file/mod not being available)?

HSTS makes the browser remember that it must always connect to this page via HTTPS. Please try to use another browser to access ownCloud and see if it works there.

Or did you change anything within ownCloud settings, / `/var/www/owncloud/config/config.php` or (e.g. CLI URL) after switching to HTTPS? But actually there should be nothing that forces HTTPS on ownCloud side.
Perhaps it's also just browser cache related, try t clear it, e.g. also just for the affected page with `<ctrl>+<F5>`.