MichaIng/DietPi #1911: HomeAssistant Nmap tracker stop working after update diepi

lupa18, 2018-07-08:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.11
- Distro version	| 9.4
- Kernel version	| Linux DietPi 4.4.77-rockchip-ayufan-136 #1 SMP Thu Oct 12 09:14:48 UTC 2017 aarch64 GNU/Linux
- SBC device		| Rock64 (aarch64)

#### Steps to reproduce:
1. Update dietpi
2. (net-tools is uninstalled)

#### Expected behaviour:
nmap trakcer should work

#### Actual behaviour:
nmap not working anymore because net-tools are a dependency for it

#### Extra details:

**Traceback:**
```
2018-07-08 17:28:36 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/home/homeassistant/.pyenv/versions/3.6.3/envs/homeassistant-3.6.3/lib/python3.6/site-packages/homeassistant/components/device_tracker/__init__.py", line 694, in async_device_tracker_scan
    found_devices = yield from scanner.async_scan_devices()
  File "/home/homeassistant/.pyenv/versions/3.6.3/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/homeassistant/.pyenv/versions/3.6.3/envs/homeassistant-3.6.3/lib/python3.6/site-packages/homeassistant/components/device_tracker/nmap_tracker.py", line 81, in scan_devices
    self._update_info()
  File "/home/homeassistant/.pyenv/versions/3.6.3/envs/homeassistant-3.6.3/lib/python3.6/site-packages/homeassistant/components/device_tracker/nmap_tracker.py", line 142, in _update_info
    mac = info['addresses'].get('mac') or _arp(ipv4)
  File "/home/homeassistant/.pyenv/versions/3.6.3/envs/homeassistant-3.6.3/lib/python3.6/site-packages/homeassistant/components/device_tracker/nmap_tracker.py", line 53, in _arp
    arp = subprocess.Popen(cmd, stdout=subprocess.PIPE)
  File "/home/homeassistant/.pyenv/versions/3.6.3/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/home/homeassistant/.pyenv/versions/3.6.3/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'arp': 'arp'
```

Fourdee, 2018-07-09:
@lupa18 

Apologies, we'll get this resolved for v6.12.

In the mean time, please install net-tools:
```
G_AGI net-tools
```

MichaIng, 2018-07-17:
Fixed: https://github.com/Fourdee/DietPi/commit/f359dad0c26c379bca37ac50ef17966a03c8619f

Fourdee, 2018-07-17:
@MichaIng 

Test install ARMv7, looks good 👍 
```
● home-assistant.service - Home Assistant
   Loaded: loaded (/etc/systemd/system/home-assistant.service; disabled; vendor
preset: enabled)
   Active: active (running) since Tue 2018-07-17 21:46:13 BST; 1min 5
0s ago
 Main PID: 30807 (homeassistant-s)
```

Fourdee, 2018-07-17:
@MichaIng 

I'll mark this as closed, appears completed (https://github.com/Fourdee/DietPi/issues/1911#issuecomment-405722726). Please reopen if required 👍

closed 2018-07-17 by Fourdee