MichaIng/DietPi #1913: DietPi-Software | Pydio: Install rework needed, currently intrusive Apache only config

MichaIng, 2018-07-09:
Pydio currently configures Apache2 webserver only and PHP + Apache configuration files in an intrusive way. Needed is:
- Download from always up-to-day source: https://download.pydio.com/pub/core/ci/pydio-latest.tar.gz
  - tar.gz due to smaller download size and possible surviving user permissions
- Webserver dependent pydio-only configuration file, containing php_(admin_)value fcgi output_buffering value, FollowSymLink and other configuration, on /var/www/pydio only basis.
- Assure PHP modules are enabled: https://pydio.com/en/docs/v8/pydio-requirements
  - Interesting openssl in opposition of mcrypt (not available on Debian Buster any more)? I think we can remove mcrypt completely.

Priority low due to low install count, however it might be related to the currently suboptimal installation method: https://dietpi.com/survey
- At least Pydio is still actively developed 👍

MichaIng, 2018-11-18:
Pydio APT repo is available, but for x86 only: https://download.pydio.com/pub/linux/debian/dists/stretch/main/

Stay with archive, to not further split/complicate install method.

Fourdee, 2018-11-19:
ASUS TB Stretch install + GUI use test:
- 🈯️ 

Loving the file drag + drop feature in GUI 😃 

Thanks @MichaIng

Although I still received the following:
[image]
```
root@DietPi:~# cat /etc/lighttpd/conf-enabled/99-dietpi-pydio.conf
$HTTP["url"] =~ "^/pydio($|/)" {
        # Deny access to pydio/data dir
        $HTTP["url"] =~ "^/pydio/data" {
                url.access-deny = ("")
        }

        # Disable PHP output buffering, recommended by Pydio
        setenv.add-environment += (
                "PHP_ADMIN_VALUE" => "output_buffering=Off",
        )

}
```

Fourdee, 2018-11-24:
Hmm, appears to be working fine. Pydio setup does not seem to actually check the URL:
- http://192.168.0.118/pydio/data/
```
403 - Forbidden
```

Fourdee, 2018-11-24:
Output buffer message disappears with:
```
/etc/php/7.0/mods-available/dietpi.ini
output_buffering=Off
```

Fourdee, 2018-11-24:
🈴 mysql.service starting with mariadb.service, causes 2+ minute wait on ```dietpi-services start```
```
root@DietPi:~# systemctl status mysql -l
● mysql.service - LSB: Start and stop the mysql database server daemo
n
   Loaded: loaded (/etc/init.d/mysql; generated; vendor preset: enabled)
   Active: active (running) since Sat 2018-11-24 04:52:43 GMT; 32s ag
o

Fourdee, 2018-11-24:
Ok, retest of Pydio setup page results + reboot:
- 🈯️ Lighttpd
- 🈯️ Nginx
- 🈯️ Apache2

Testing passed. Completed.

closed 2018-11-24 by Fourdee