MichaIng/DietPi #1914: run-parts: /etc/cron.hourly/dietpi exited with return code 1

sergiud, 2018-07-10:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `6.11`
- Distro version	| stretch
- Kernel version	| `4.14.32+`
- SBC device		| `Odroid XU3/XU4/HC1/HC2 (armv7l)`
- Power supply used	| 15V 4A
- SDcard used		| eMMC

#### Additional Information (if applicable):
- Software title	| cron

#### Steps to reproduce:

1. Install OpenMediaVault
2. Enable e-mail notifications

#### Expected behaviour:
No errors.

#### Actual behaviour:
Error:
```bash
run-parts: /etc/cron.hourly/dietpi exited with return code 1
```

MichaIng, 2018-07-10:
@sergiud 
Thank you for your report.

Let's see, which part failed, call the following and post results:
`/DietPi/dietpi/func/run_ntpd 1`

Which logging mode did you choose, hourly clearing (mode -1) or hourly clearing with persistent storage (mode -2)?
`/DietPi/dietpi/dietpi-logclear 1` in case of mode -1, which will just clear all log files
`/DietPi/dietpi/dietpi-logclear 0` in case of mode -2, which will first save current log files on the disk, then clears the log files (in ramlog).

sergiud, 2018-07-10:
`/DietPi/dietpi/func/run_ntpd 1` result:

```bash
[  OK  ] Root access verified.
[  OK  ] DietPi-Run_ntpd | systemctl restart systemd-timesyncd
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (1/60)
[  OK  ] DietPi-Run_ntpd | NTPD: systemd-timesyncd synced
[  OK  ] NTPD: time sync | Completed
```

I use the default logging mode. More specifically, Log System in `dietpi-software` states `DietPi-Ramlog #1`. @MichaIng  Is this the info you are looking for?

MichaIng, 2018-07-10:
In this case run: `/DietPi/dietpi/dietpi-logclear 1`

sergiud, 2018-07-10:
Result:
```bash
[  OK  ] Root access verified.
[ INFO ] DietPi-Logclear | 
 - Log file directory      | /var/log
 - Processed files         | 138
 - Excluded files          | 0
 - Cleared log files       | 8
 - Unsupported files       | 0
 - Deleted files           | 0
 - Space cleared           | 14 KB
```

sergiud, 2018-07-10:
Additional info: running `sudo /etc/cron.hourly/dietpi ; echo $?` outputs 0. Hence, I'm not sure where the error stems from. The only time I can reproduce the problem is if I run `/etc/cron.hourly/dietpi` not as root.

MichaIng, 2018-07-10:
@sergiud 
Thanks for the info.

Hmm the /etc/cron.* folders are definitely run as root:
```
root@VM-Jessie:~# cat /etc/crontab
#Please use dietpi-cron to change cron start times
SHELL=/bin/sh
PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin

# m h dom mon dow user  command
*/30 * * * *   root    cd / && run-parts --report /etc/cron.minutely
17 *    * * *   root    cd / && run-parts --report /etc/cron.hourly
25 1    * * *   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.daily )
47 1    * * 7   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.weekly )
52 1    1 * *   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.monthly )
```

You could manually run what cron does and see if this works: `run-parts --report /etc/cron.hourly`

We have another opened issue with cron and one of our scripts. I don't think there is a connection, but just in case I link it here: https://github.com/Fourdee/DietPi/issues/1901

I saw you installed OMV. We dropped official support for this, as we faced much issues with DietPi reported by users that were/are hard to fix, the way OMV works. Maybe it doesn't like it's log files being cleared, otherwise it works running manually 🤔. Maybe OMV has a cron job that interferes with dietpi cron job, clearing the logs. The above run-parts command would then throw the error as well.

To check, if there are actually OMV logs:
`ls -al /var/log`

If this is indeed the case, besides some manual workarounds, or perhaps finding a way to change OMVs logging behaviour, I don't see another good solution than disabling dietpi-ramlog:
```
dietpi-software uninstall 103
systemctl disable dietpi-ramlog
```

sergiud, 2018-07-12:
I will test the proposed change shortly. Thank you!

There are also the following issue as well, possibly related to this one. Namely,

```
/etc/cron.daily/logrotate:
error: skipping "/var/log/mysql/mysql.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/mysql/mysql-slow.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/mysql/mariadb-slow.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/mysql/error.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/nginx/access.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/nginx/error.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/nginx/gitlab_access.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/nginx/gitlab_error.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/nginx/openmediavault-webgui_access.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/nginx/openmediavault-webgui_error.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/redis/redis-server.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/unattended-upgrades/unattended-upgrades.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/unattended-upgrades/unattended-upgrades-dpkg.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
error: skipping "/var/log/unattended-upgrades/unattended-upgrades-shutdown.log" because parent directory has insecure permissions (It's world writable or writable by group which is not "root") Set "su" directive in config file to tell logrotate which user/group should be used for rotation.
run-parts: /etc/cron.daily/logrotate exited with return code 1
```

I never changed the rights for `/var/log`, so I wonder what can cause this whether the issues are related.

MichaIng, 2018-07-14:
@sergiud 
Yeah, the logrotate issue is well known. I also found the log files being created by the software with a certain user/group and `/etc/logrotate.d/<name>.log` addition not containing the needed user/group entries to handle this. Not sure why the package maintainers never took care about this, as it's within their responsibility 😉.

But also `dietpi-ramlog` needs some review. We pre-create some softwares log folders, which should not be needed and may destroy user permissions on them.

MichaIng, 2018-07-15:
@sergiud 
Fast fix for initial issue, at least it finally appears that this is the reason for all issues with cron calling DietPi scripts: `wget https://raw.githubusercontent.com/Fourdee/DietPi/82ac7b32d32dca9db4fdb824c7ead80174844090/dietpi/func/dietpi-globals -O /DietPi/dietpi/func/dietpi-globals`

MichaIng, 2018-07-15:
@sergiud 
To clean up a bid I opened a new issue, please try to apply the fix mentioned there and report back, if it solves all issues: https://github.com/Fourdee/DietPi/issues/1923

I will close this issue in favour of the new one.

closed 2018-07-15 by MichaIng