MichaIng/DietPi #1915: Odroid C2 Unable to find expected entry 'main/binary-arm64/Packages' in Release File

gittusmaximus, 2018-07-12:
#### Details:
- Date           | Wed 11 Jul 13:21:35 -12 2018
- Bug report     | N/A
- DietPi version | v6.11
- Img creator    | ETCHER
- Pre-image      | n/a
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017
- Distro         | jessie (index=3)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
1. Installing through dietpi-software
2. Occurred lots when updating from 6.9 to 6.11 (kept trying to rerun last command and version updated but problem still persists in dietpi-software installs. 

#### Expected behaviour:
Things should be installed. 

#### Actual behaviour:
Doesn't install and throws this (as shown in log file below):
W: Failed to fetch http://deb.debian.org/debian-security/dists/jessie/updates/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

#### Extra details:
I know I messed with adding some 32bit libraries so I could try mess around with Boinc but I cannot remember how I did this or where the instructions are. 

I also tried using the solutions provided here: https://github.com/Fourdee/DietPi/issues/653
but I got a 404 Not Found when running the first command. I then looked up the repo and found that the specified "libc6_2.19-18+deb8u6_arm64.deb" does not exist there anymore, I saw that "deb8u10_arm64.deb" did so I amended the commands thinking this might help but nope. 

I eventually managed to install build-essential through apt-get install whereas prior to running the above commands I was unable to. 

I've also tried changing APT mirrors but this fails to work either. 

I tried to create a bugreport but this occurred:
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[......] DietPi-Bugreport | (1/5) Testing connection to dietpi.com, please wait.[  OK  ] DietPi-Bugreport | Connection test: dietpi.com
stat: cannot stat ‘6dba9225-2d2d-430b-85f7-3dfb164b9148.7z’: No such file or directory
/DietPi/dietpi/dietpi-bugreport: line 154: ((: <= 10000000 : syntax error: operand expected (error token is "<= 10000000 ")
[FAILED] DietPi-Bugreport | The bug report upload archive appears to be unexpected large. Please inspect and in case clean up the locations to be uploaded, as their size should never be that large:
CMD_OUT.txt
G_ERROR_HANDLER_GITREPORT
/var/log/*
/DietPi/*
/boot/dietpi.txt
/boot/config.txt
/boot/dietpi/*
/tmp/.G*
/var/tmp/dietpi/logs/*
/var/lib/dietpi/*
/etc/bash.bashrc
/etc/bashrc.d/*
/root/.bashrc
/home/*/.bashrc
/etc/profile
/etc/profile.d/*
/root/.profile
/home/*/.profile
/etc/rc.local
/etc/X11/xorg.conf
/etc/asound.conf
/etc/network/interfaces
/etc/fstab
/etc/sysctl.conf
/etc/sysctl.d/*
/etc/init.d/*
/etc/systemd/system/*
/lib/systemd/system/*
/etc/apt/sources.list
/etc/apt/sources.list.d/*

- ...

#### Additional logs:
```
Log file contents:
Hit http://deb.debian.org jessie/updates InRelease
Ign http://ftp.debian.org jessie InRelease
Hit http://fuzon.co.uk all InRelease
Hit http://ftp.debian.org jessie-updates InRelease
Hit http://fuzon.co.uk jessie InRelease
Hit http://ftp.debian.org jessie-backports InRelease
Hit http://ftp.debian.org jessie Release.gpg
Get:1 http://ftp.debian.org jessie-updates/main arm64 Packages/DiffIndex [11.8 kB]
Hit http://ftp.debian.org jessie-updates/contrib arm64 Packages
Get:2 http://ftp.debian.org jessie-updates/non-free arm64 Packages/DiffIndex [736 B]
Hit http://ftp.debian.org jessie-updates/contrib Translation-en
Get:3 http://ftp.debian.org jessie-updates/main Translation-en/DiffIndex [3,688 B]
Get:4 http://ftp.debian.org jessie-updates/non-free Translation-en/DiffIndex [736 B]
Get:5 http://ftp.debian.org jessie-backports/main arm64 Packages/DiffIndex [27.8 kB]
Get:6 http://ftp.debian.org jessie-backports/contrib arm64 Packages/DiffIndex [23.9 kB]
Get:7 http://ftp.debian.org jessie-backports/non-free arm64 Packages/DiffIndex [8,530 B]
Get:8 http://ftp.debian.org jessie-backports/contrib Translation-en/DiffIndex [7,960 B]
Get:9 http://ftp.debian.org jessie-backports/main Translation-en/DiffIndex [27.8 kB]
Get:10 http://ftp.debian.org jessie-backports/non-free Translation-en/DiffIndex [19.1 kB]
Hit http://fuzon.co.uk all/c2 arm64 Packages
Hit http://ftp.debian.org jessie Release
Hit http://fuzon.co.uk all/main arm64 Packages
Hit http://fuzon.co.uk all/testing arm64 Packages
Hit http://ftp.debian.org jessie/main arm64 Packages
Hit http://ftp.debian.org jessie/contrib arm64 Packages
Hit http://fuzon.co.uk jessie/backports arm64 Packages
Hit http://ftp.debian.org jessie/non-free arm64 Packages
Hit http://fuzon.co.uk jessie/main arm64 Packages
Hit http://ftp.debian.org jessie/contrib Translation-en
Hit http://ftp.debian.org jessie/main Translation-en
Hit http://ftp.debian.org jessie/non-free Translation-en
Ign http://fuzon.co.uk all/c2 Translation-en_GB
Ign http://fuzon.co.uk all/c2 Translation-en
Ign http://fuzon.co.uk all/main Translation-en_GB
Ign http://fuzon.co.uk all/main Translation-en
Ign http://fuzon.co.uk all/testing Translation-en_GB
Ign http://fuzon.co.uk all/testing Translation-en
Ign http://fuzon.co.uk jessie/backports Translation-en_GB
Ign http://fuzon.co.uk jessie/backports Translation-en
Ign http://fuzon.co.uk jessie/main Translation-en_GB
Ign http://fuzon.co.uk jessie/main Translation-en
Fetched 132 kB in 18s (7,049 B/s)
W: Failed to fetch http://deb.debian.org/debian-security/dists/jessie/updates/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

BerndKohl, 2018-07-13:
Having the same problem!
Please fix as I am no longer able to update Dietpi or any other piece of software in my Odroid.

MichaIng, 2018-07-14:
@gittusmaximus 
Thanks for your report.
> W: Failed to fetch http://deb.debian.org/debian-security/dists/jessie/updates/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

Indeed `binary-arm64` subfolder and `InRelease` entries are not available within the Debian security repos:
http://security.debian.org/debian-security/dists/jessie/updates/main/
http://security.debian.org/dists/jessie/updates/main/

But `binary-all` instead, which should be successfully used here.

@BerndKohl 
Just to be sure, you also use an ARMv8/aarch64/arm64 device, right?
To check: `echo $G_HW_ARCH`

It seems to be due to changes within Debian repo structure, although I cannot believe they produce such an error across all arm64 devices 🤔.

Just to sort that our: Could you both try: `apt-get clean && apt-get update`

BerndKohl, 2018-07-14:
echo $G_HW_ARCH
3

apt-get clean && apt-get update
Ign http://fuzon.co.uk jessie/main Translation-en                                                                               
Fetched 756 kB in 12s (59.1 kB/s)                                                                                               
W: Failed to fetch http://deb.debian.org/debian-security/dists/jessie/updates/InRelease  Unable to find expected entry 'main/binary-arm64/Packages' in Release file (Wrong sources.list entry or malformed file)

E: Some index files failed to download. They have been ignored, or old ones used instead.

Fourdee, 2018-07-14:
Unable to replicate on C2 with Stretch. Assuming this is Jessie limited issue.

> https://cdn-aws.deb.debian.org/debian-security/dists/stretch/updates/InRelease

🈯️ Architectures: amd64 arm64 armel armhf i386 mips mips64el mipsel ppc64el s390x

> http://deb.debian.org/debian-security/dists/jessie/updates/InRelease

🈴 Seems they removed ARM64 binaries | Architectures: amd64 armel armhf i386

Fourdee, 2018-07-14:
This is an external bug, however, we'll try to find a solution.

A very quick workaround, would be to remove the offending entry from ```/etc/apt/sources/list```:
```
deb http://security.debian.org jessie/updates  main contrib non-free
```

BerndKohl, 2018-07-14:
sorry for this silly question: but how do I do this? Simply putting a #  in front of the offending entry didn't do the trick...

BerndKohl, 2018-07-14:
ok, I got it. Had to remove this entry:
> deb http://deb.debian.org/debian-security/ jessie/updates main contrib non-free

Is it safe now to call DietPi-update or will it break something?

gittusmaximus, 2018-07-14:
> Just to sort that our: Could you both try: apt-get clean && apt-get update
No change in issue, same as BerndKohl

removed offending line from sources.list
and apt-get update works without sending that error message. 

Just tried installing new software and the issue seems to have gone. 
ie G_AGUP doesn't send exit code 100 and it installs new software. 

```
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[  OK  ] DietPi-Software | The system will now reboot.
 This completes the DietPi-Software installation.
```

Thanks very much @Fourdee  and @MichaIng  , appreciate your time and help fixing this problem. 
Will be sticking with your OS for quite some time now!

BerndKohl, 2018-07-15:
Successfully upgraded DietPi from 6.9 to 6.11 after having removed the offending line.

Thanks for your support on this!

MichaIng, 2018-07-15:
@gittusmaximus @BerndKohl 
The offending line contains the the Debian security update repository: https://www.debian.org/security/
It provides security relevant APT package updates a bid earlier than the main repo/mirror and allows admins of extremely importantly running servers to e.g. update security patches only, while leaving all other APT packages untouched to reduce risk of server downs. But as long as you don't want to actively achieve this, the security repo is indeed from minor importance, as you will receive the same updates max. a day later from main repo 👍.

rickers1960, 2018-07-16:
ok, I got it. Had to remove this entry:

deb http://deb.debian.org/debian-security/ jessie/updates main contrib non-free

Apologies but I am a complete novice, can someone help explain how to remove this entry in order that i may update my C2,         I am putty.exe to connect to my C2. 

thanks in advance .

rickers1960, 2018-07-16:
Thanks for the quick reply , I am logging in with root .

I have removed the problem line but still unable to complete update . 

Please see attached file 

[DietPi.docx](https://github.com/Fourdee/DietPi/files/2198096/DietPi.docx)

Any ideas on what is wrong ?

MichaIng, 2018-07-16:
@rickers1960 
Connection to fuzon mirror of meverics repo is failing.

@Fourdee 
I can't ping the domain, is this correct?
https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L601

Fourdee, 2018-07-16:
@MichaIng 

Its back up and running, it runs on the older MVS account system, Paypal payments are manual. Error my end:
[image]

rickers1960, 2018-07-16:
Thanks to both , I have now updated to 6.11

MichaIng, 2018-07-16:
@Fourdee 
Maybe add Meveric APT mirror to `dietpi.com`, so on the long hand you can close the other domain+MVS?

Fourdee, 2018-07-16:
### Workaround:

```
sed -i '/debian-security/d' /etc/apt/sources.list
G_AGUP
```

Fourdee, 2018-07-16:
@MichaIng 

> Maybe add Meveric APT mirror to dietpi.com, so on the long hand you can close the other domain+MVS?

Yep, issue is HDD space, and, I also give full access to Meveric for this server (least we can do to support his work over the years):
```
root@fuzon:~# df -h
Filesystem         Size  Used Avail Use% Mounted on
/dev/ploop14827p1   50G   33G   15G  69% /
```

Fourdee, 2018-07-16:
Completed.

closed 2018-07-16 by Fourdee