MichaIng/DietPi #1918: Lxqt install on rpi3/+

LeeMenHin, 2018-07-12:
It is possible to install Lxqt onboth device ？ Because seem lxqt modern then lxde more light weight and less resources use 。Please help if it possible to add in dietpi-software（sorry about my English）（A user from Hong Kong）

MichaIng, 2018-07-12:
@LeeMenHin 
Thanks for your request.

Indeed LXQt seems to be interesting. Lubuntu 18.10 switches as well. Some complaints are that it is a bid buggy.

I will do some test installation and see if it works well with our software offer. But there is much to do besides, so could take a time from my side 😉.

LeeMenHin, 2018-07-13:
thx a lot bro always support you。 If here i can help

Fourdee, 2018-07-15:
Qt a great library, however, it has a vast number of pre-reqs, could result in a bulkier installation than current.

MichaIng, 2018-07-17:
Just tried it out on VM. Works fine so far. 

**About the package size:**
RPi (LAMP server + full featured Nextcloud installed):
- `apt install lxqt xserver-xorg xinit` => `479 MB of additional disk space will be used.`
- `apt install lxqt-core xserver-xorg xinit` => `366 MB of additional disk space will be used.`
- `apt install lxde xserver-xorg xinit` => `401 MB of additional disk space will be used.`
- `apt install lxde-core xserver-xorg xinit` => `363 MB of additional disk space will be used.`

But on VM (most clean image) the other way round:
- `apt install lxqt xserver-xorg xinit` => `288 MB of additional disk space will be used.`
- `apt install lxqt-core xserver-xorg xinit` => `281 MB of additional disk space will be used.`
- `apt install lxde xserver-xorg xinit` => `330 MB of additional disk space will be used.`
- `apt install lxde-core xserver-xorg xinit` => `300 MB of additional disk space will be used.`

Interesting, that it takes so much more space on RPi, even that more system/library packages should be already installed.

On VM, desktops are anyway a bid slow, so hard to compare, but the difference at least is not large.
I would not replace LXDE, but add LXQT to our Desktop offer. But it needs to be integrated and tested with our desktop software offer of course.

Fourdee, 2018-07-17:
@MichaIng 

> LXDE, but add LXQT 

Yep agree 👍  Offer it as an additional selection.

LeeMenHin, 2018-07-21:
Very nice to hear about working on . Stay and wait
~

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+9

MichaIng, 2019-04-16:
Thanks to @Fourdee we will have LXQt available with v6.23: https://github.com/MichaIng/DietPi/commit/697e33217bc5850841373f6f59eafc1abbd42bf7
Many thanks for your great effort 😃. Perhaps we can take the chance and add a desktop preference similar to webserver preference?

Currently when a software is chosen that requires a desktop, but no desktop is or was chosen before, LXDE will be installed by default, which is the best, because most lightweight, default IMO. But would be great to give user the choice before?

Fourdee, 2019-04-19:
:u6307: Available in DietPi v6.23:
[image]

MichaIng, 2019-04-19:
@Fourdee 
Many thanks. Will directly try it out on notebook 😃. I reopen the issue to not forget adding this to beta testing phase to perhaps gain some feedback for fine tuning or something.

Fourdee, 2019-04-19:
@MichaIng 

Testing passed for:
- :u6307: x86:64 native with auto start
- :u6307: RPi with TightVNC server

MichaIng, 2019-04-19:
@Fourdee 
Yeah, good point to test Buster now as well on a regular basis. Going through all software titles is still a task to do, but has the advantage that all web applications must stand PHP7.3, and that goes hand-in-hand with the decision to move Stretch systems to either PHP7.2 or PHP7.3. Nextcloud 16 is already on RC3 and will not support PHP7.0 anymore... Damn thing since for sure some users will run into the issue to have an incompatible version installed until we release v6.23 🙈.

Fourdee, 2019-04-20:
Changes:
- LXQT theme for both Stretch and Buster
- Speedcrunch installed by default (QT based calculator)
- ```apt-mark auto``` for uninstall of non core desktop items (eg: additions ```firefox-esr```)

MichaIng, 2019-04-29:
@Fourdee 
On Buster `G_CONFIG_INJECT 'use_trash=' 'use_trash=0' /etc/xdg/libfm/libfm.conf` fails since the file does not exist.

And: https://github.com/MichaIng/DietPi/issues/2747

MichaIng, 2019-04-30:
About `libfm.conf`:
- On Buster [pcmanfm-qt](https://packages.debian.org/buster/pcmanfm-qt) does not depend on libfm-modules which depends on libfm-data recursively which ships the file:
```
root@VM-Stretch:~# dpkg -S /etc/xdg/libfm/libfm.conf
libfm-data: /etc/xdg/libfm/libfm.conf
```
- On Stretch this is the case: https://packages.debian.org/stretch/pcmanfm-qt
- So I guess on Buster the trashbin is implemented differently, so we skip this adjustment simply. We can find out in a future version how to turn off trash bin by default on Buster, if this is not yet the case.

Fourdee, 2019-05-04:
@MichaIng 

Featherpad also alternative option, seems to replace leafpad?
[image]
 
https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=featherpad

MichaIng, 2019-05-04:
@Fourdee 
Featherpad is Qt only.

On LXDE mousepad is chosen automatically as replacement (alternative dependency), so to be consistent we should just apply/follow this on all desktops?

Ah I see, mousepad ships GTK dependencies that are not native on Qt based desktops, right?
Yeah so then I agree to go with the one that integrates more natively 👍.

closed 2019-05-12 by MichaIng