MichaIng/DietPi #1923: Important | Issues with cron jobs calling DietPi scripts

MichaIng, 2018-07-15:
Related issues:
- https://github.com/Fourdee/DietPi/issues/1901
- https://github.com/Fourdee/DietPi/issues/1914
- https://github.com/Fourdee/DietPi/issues/1920

This is a fix everyone should apply, as it breaks DietPi's own cron jobs to work (clearing `/var/log` hourly + time sync). Currently most cron jobs, executing DietPi scripts, will exit with error. Malicious line is https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-globals#L266, which leads to `tput` error exiting the whole script. On v6.9 `local lines; (( lines=(${#input_string}+6)/$(tput cols) ))` had same error output but without exiting the script, for whatever reason 😉. If some shell/bash expert knows why the first solution exits the whole script, please share, as this is interesting to know and might help to prevent further errors on this scripting bases.

Fast fix to apply, which will be part of v6.12 as well: `wget https://raw.githubusercontent.com/Fourdee/DietPi/0f18aa4dc0af8ab910a0173dce8849d5b53c30b0/dietpi/func/dietpi-globals -O /DietPi/dietpi/func/dietpi-globals`

sergiud, 2018-07-16:
Great! This eliminated hourly error notifications by OMV for me.

kmakisara, 2018-07-16:
Time sync hourly works without problems.

eljotx, 2018-07-16:
Nice! After fix, cron with dietpi-backup (#1901) is working well.

Fourdee, 2018-07-16:
Ok, lets mark this as closed. Resolved for v6.12 (great work @MichaIng 👍 ). In the mean time, solution = https://github.com/Fourdee/DietPi/issues/1923#issue-341325599

closed 2018-07-16 by Fourdee

aspinks, 2019-04-19:
i know this is old but i ran this:
wget https://raw.githubusercontent.com/Fourdee/DietPi/0f18aa4dc0af8ab910a0173dce8849d5b53c30b0/dietpi/func/dietpi-globals -O /DietPi/dietpi/func/dietpi-globals
now when i try to install it goes to the drive manager.  i exit that and get this:
 OK  ] DietPi-Software | Installing Sonarr: automatically download TV shows
[  OK  ] DietPi-Software | Free space check: path=/ | available=24428 MB | required=500 MB

then it goes to the command prompt.

MichaIng, 2019-04-19:
See: https://github.com/MichaIng/DietPi/issues/2724