MichaIng/DietPi #1924: DietPi-RAMlog | Additional work for disabling

Fourdee, 2018-07-15:
> https://github.com/Fourdee/DietPi/pull/1919#issuecomment-405108537

Fourdee, 2018-07-17:
https://github.com/Fourdee/DietPi/commit/e85f2efa39bd7d3867f06a804149c47c45c3170e

MichaIng, 2018-07-18:
@Fourdee 
I will test.

https://manpages.debian.org/stretch/systemd/systemd.service.5.en.html#OPTIONS
So for our needs, forking and oneshot both have the effect (compared to default=simple), that ExecStart needs to finish before systemd starts follow-up units. In our case of dietpi-ramlog/ramdisk services, oneshot actually fits better, since we do not start a daemon within the scripts. But without PIDFile info for systemd to stop on shutdown, I guess both types have practically the same effect.
And yeah I remember you saying the nice green dot on systemd status gives one the impression that something is active (actually tmpfs mount) 😉.
**Ah more importantly, good question would be if ExecStop is called for oneshot scripts, or worse, it is called directly after ExecStart has finished.** Jep we stay with forking 👍!

According to the above, I think the disable service should be better oneshot, so systemd does not go on until all it's tasks are finished.

Fourdee, 2018-07-18:
Hmm, think we could simplify this a little. With disabling ```dietpi-ramlog``` we risk something going wrong for the user and their logs.

Basically, its part of the DietPi system. Maybe we should simply detect if running in RAM or DISK mode, and inform the user in ```dietpi-ramlog``` about which mode is running?

Fourdee, 2018-07-18:
@MichaIng 

Yep agree, I just need more coffee in my system 🤣 

> https://github.com/Fourdee/DietPi/issues/1924#issuecomment-405766368

Resolved with:
https://github.com/Fourdee/DietPi/commit/ac6e524cf30aa4087a0ddef9b9765499c7d6f052

Fourdee, 2018-07-18:
Hmm,

ramlog > full
full < ramlog

- 1st test 🈯️ 
- 2nd test 🈴 service still reported as enabled, but does not run? | ```/var/tmp/dietpi/logs/dietpi-ramlog_store/``` still exists.
```
root@DietPi:~# systemctl status dietpi-ramlog
● dietpi-ramlog.service - DietPi-RAMlog
   Loaded: loaded (/etc/systemd/system/dietpi-ramlog.service; enabled; vendor pr
eset: enabled)
```

Fourdee, 2018-07-18:
@MichaIng 

> Require=dietpi-ramlog in combination with systemctl stop dietpi-ramlog as first command leads to service terminates itself.

Yep confirmed. 👍 

> The service does not require ramdisk either, does it?

systemctl stop dietpi-ramdisk does, uses ```/DietPi``` in the service

Just doing some further tests, will send commit in a few.

MichaIng, 2018-07-18:
@Fourdee 
Ah yes sure, so `Require=dietpi-ramdisk` should stay.

Fourdee, 2018-07-18:
Good to go: https://github.com/Fourdee/DietPi/commit/00fc31b3970d335f685e060e16e11ae40e7da2be

Passed testing my end: https://github.com/Fourdee/DietPi/issues/1924#issuecomment-405993612

MichaIng, 2018-07-18:
@Fourdee 
Jep works fine now here as well 👍.

Fourdee, 2018-07-18:
Completed.

closed 2018-07-18 by Fourdee