MichaIng/DietPi #1925: Bug

MOHELLEBI, 2018-07-16:
#### Details:
- Date           | Mon 16 Jul 08:32:17 CEST 2018
- Bug report     | N/A
- DietPi version | v6.11
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1123 SMP Wed Jun 27 17:35:49 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: bash-doc -s
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Package 'bash-doc' has no installation candidate
```

MichaIng, 2018-07-16:
@MOHELLEBI 
Please try:
```
apt clean
G_AGUP
G_AGUG
```
Maybe it is also a temporary issue with the Raspbian repo.
On Raspbian Buster it works here, but have no RPi on Stretch to assure.

Did you switch your APT sources? You can reset it via:
`dietpi-config` > `Network Options: Misc` > `APT mirror` > there choose, if in doubt, `Global (raspbian.raspberrypi.org)`

MichaIng, 2018-07-27:
I will mark this as closed, guess time solved the issue, as we watched other RaspberryPi repo issues the last days which are resolved now.

Feel free to reopen, if required.

closed 2018-07-27 by MichaIng

jcw, 2019-08-19:
FWIW, it's happening again now. This is a clean install, no way to get past this at this point, I guess.
I'll retry later.