MichaIng/DietPi #1927: META | DietPi-Benchmark optional uploads

Fourdee, 2018-07-16:
Similar to dietpi-survey. Have an option to run:
- RootFS transfer rate
- CPU performance
- RAM performance

Then option to upload scores to http://dietpi.com/benchmark
Simple database to compare your scores against HW models etc.

@MichaIng 
Thoughts?

Fourdee, 2018-07-16:
REF: https://github.com/Fourdee/DietPi/issues/1253#issuecomment-346881878

MichaIng, 2018-07-16:
@Fourdee 
Nice idea. We can combine this with `dietpi-survey`, e.g. `dietpi-benchmark` allows user to choose upload, then creates some flag (which could be simply the data within /tmp/dietpi-benchmark/upload.txt) and starts dietpi-survey. dietpi-survey then checks for the flag and in case adds benchmark results die survey upload.

Hmm, but needs different handling, as relation between device data and results is needed 🤔.

Fourdee, 2018-07-16:
@MichaIng 

Excellent 👍 

Once https://github.com/Fourdee/DietPi/issues/1768 is completed, i'll make a start on this.

LexiconCode, 2018-07-17:
Consider implementing this into the stats for each device under the download section of dietpi.com
It's very hard to get a picture of real-world performance of a SBC before purchase not to mention accessory hardware performance in relation to a particular SBC.

Fourdee, 2018-07-17:
@LexiconCode 

Great idea, love it 👍 We'll definitely consider doing this.

My PHP experience is somewhat limited, so we may have to simply link the URL in the downloads section.

Fourdee, 2018-08-22:
@MichaIng 
Benchmark suite code done, just need to impliment how we are to upload this to dietpi.com. Use of survey and send, if the generated results file exists?
[image]

LexiconCode, 2018-08-22:
Is it possible for rootfs benchmark to determine what type of hardware like MMC, hard drive, SD card?

Fourdee, 2018-08-22:
@LexiconCode 

> Is it possible for rootfs benchmark to determine what type of hardware like MMC, hard drive, SD card?

Unsure, however, good suggestion, i'll take a look 👍

MichaIng, 2018-08-23:
@Fourdee 
Ah sorry, was a bid busy since last weekend, missed that one.

First I agree with @LexiconCode that the values have more meaning, if connected with related hardware info.
- On SBCs CPU and RAM are fixed, thus here $G_HW_MODEL[_DESCRIPTION] would be sufficient.
- But on native PC + VM we should find out which hardware is in use, e.g. `grep -m1 '^model name' <<< "$(</proc/cpuinfo)" | sed 's/model name[[:blank:]]*: //'`?
- But couldn't find a way to get memory bandwidth or name info, besides using additional tools: `lshw` or `dmidecode`
- For disk speed and all machines are at least some basic drive and connection info helpful to sort/interpret the values, as mentioned. mmc can be identified via root mount /dev/* name, but no idea how to get all these infos without additional tools.
  - `cat /sys/block/mmcblk0/device/name` does not always exist
  - `cat /sys/block/sda/device/model` sometimes empty, also does not always exist
  - Couldn't find USB version or something so far.

About file scheme:
- If @userdeveloper98 finds the time to add it to the new survey page, theoretically it doesn't matter too much.
- But just to stay flexible it is never wrong to add it in a shell readable way again. But compared to the java database, we need to think how to present the data. E.g.:
  - Calculate average values for each device, also to support our performance rating. Then the values would need to be added e.g. like this:
`aBENCH_CPU[$G_HW_MODEL_DESCRIPTION]=$(( ${aBENCH_CPU[$G_HW_MODEL_DESCRIPTION]:=0} + $CPU_BENCH_RESULT )) && (( ${aDEVICE_COUNT[$G_HW_MODEL_DESCRIPTION]}++))`
After looping through all upload files:
`for i in ${#aDEVICE_COUNT[@]}; do ${aBENCH_CPU_AVG[$i]}=$(( ${aBENCH_CPU[$i]} / ${aDEVICE_COUNT[$i]} )); done`
  - Show a result spread over all devices:
((aBENCH_CPU[$CPU_BENCH_RESULT]++))
The result would need to be rounded first, grouped into reasonable score intervals, so we can present it in some sort of histogram.
- Or we do both 😄.
- But sourcing these files then does not contain any additional hardware info, for this I see no other chance then creating $UNIQUE_ID indexed arrays and adding the above mentioned infos as well to allow differentiation.

Fourdee, 2018-08-25:
@MichaIng 

We could probably do this using a 2D array?
```
((aBENCH_CPU_TIME[$G_HW_MODEL_DESCRIPTION,$(sed -n 1p $fp_dietpi_benchmark)]++))
```

That way, results are linked to the ```$G_HW_MODEL_DESCRIPTION```

```
# -------------------------
# DietPi-Benchmark
# -------------------------
((aBENCH_CPU_TIME[Odroid C2 (aarch64),17.15]++))
((aBENCH_ROOTFS_WRITE[Odroid C2 (aarch64),26.6]++))
((aBENCH_ROOTFS_READ[Odroid C2 (aarch64),110]++))
((aBENCH_RAM_WRITE[Odroid C2 (aarch64),473]++))
((aBENCH_RAM_READ[Odroid C2 (aarch64),480]++))
```

MichaIng, 2018-10-01:
@Fourdee 
Niiiice.

Maybe we can merge the results as additional columns into the "Devices" install count table, what you think?

Fourdee, 2018-10-02:
@MichaIng 

> Maybe we can merge the results as additional columns into the "Devices" install count table, what you think?

Good idea 👍 

Only issue is, i'll need to create about 4/5 of these tables, as, adding all the info on 1 table makes it too squashed.

I'll have a play bit later.

Fourdee, 2018-10-02:
Some issues with server, needs a reboot:
```
[10535890.095824] TCP: request_sock_TCP: Possible SYN flooding on port 443. Sending cookies.  Check SNMP counters.
[10538980.248728] TCP: request_sock_TCP: Possible SYN flooding on port 80. Sending cookies.  Check SNMP counters.
[12131781.922070] INFO: task jbd2/sda1-8:410 blocked for more than 120 seconds.
[12131781.922074]       Not tainted 4.9.0-6-amd64 #1 Debian 4.9.82-1+deb9u3
[12131781.922074] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[12131781.922076] jbd2/sda1-8     D    0   410      2 0x00000000
[12131781.922098]  ffff8c6cbacac000 0000000000000000 ffff8c6c769030c0 ffff8c6cbfc18940
[12131781.922124]  ffffffff9da11500 ffffabcbc04b7ca0 ffffffff9d40c649 ffff8c6c76e7e888
[12131781.922126]  0000000000000246 ffff8c6cbfc18940 ffffabcbc04b7d80 ffff8c6c769030c0
[12131781.922129] Call Trace:
[12131781.922252]  [<ffffffff9d40c649>] ? __schedule+0x239/0x6f0
[12131781.922269]  [<ffffffff9ceb9c50>] ? prepare_to_wait_event+0xf0/0xf0
[12131781.922271]  [<ffffffff9d40cb32>] ? schedule+0x32/0x80
[12131781.922326]  [<ffffffffc02f428f>] ? jbd2_journal_commit_transaction+0x25f/0x17b0 [jbd2]
[12131781.922329]  [<ffffffff9ceb9c50>] ? prepare_to_wait_event+0xf0/0xf0
[12131781.922333]  [<ffffffffc02f9c02>] ? kjournald2+0xc2/0x260 [jbd2]
[12131781.922335]  [<ffffffff9ceb9c50>] ? prepare_to_wait_event+0xf0/0xf0
[12131781.922338]  [<ffffffffc02f9b40>] ? commit_timeout+0x10/0x10 [jbd2]
[12131781.922346]  [<ffffffff9ce970c9>] ? kthread+0xd9/0xf0
[12131781.922349]  [<ffffffff9ce96ff0>] ? kthread_park+0x60/0x60
[12131781.922355]  [<ffffffff9ce7c3d0>] ? SyS_exit_group+0x10/0x10
[12131781.922357]  [<ffffffff9d411537>] ? ret_from_fork+0x57/0x70
[12131781.922370] INFO: task mysqld:24660 blocked for more than 120 seconds.
[12131781.922371]       Not tainted 4.9.0-6-amd64 #1 Debian 4.9.82-1+deb9u3
[12131781.922372] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
```

Fourdee, 2018-10-02:
@MichaIng 

Server offline for backup.

Might also upgrade it to 4cores 4GB 😉

MichaIng, 2018-10-02:
@Fourdee 
> Also really want it this way, but can't figure it out yet:

Use `colspan` to spread a cell over multiple columns. `rowspan` could be used to spread the first "Device" cell over the first two lines then:

```
<table>
  <tr>
    <th rowspan="2">Device:</th>
    <th colspan="3">CPU time</th>
  </tr>
  <tr>
    <th>avg</td>
    <th>min</td>
    <th>max</td>
  </tr>
  <tr>
    <td>RPi</td>
    <td>123</td>
    <td>56</td>
    <td>203</td>
  </tr>
</table>
```

Fourdee, 2018-10-02:
Boom boom:
[image]

MichaIng, 2018-10-02:
@Fourdee 
Looks nice.

However this could be presented much nicer on the new java survey page with nice charts.
Maybe @userdeveloper98 finds some time to implement this by times? 😜

Fourdee, 2018-10-02:
Color me up sally:
[image]

Fourdee, 2018-10-02:
@MichaIng 

> However this could be presented much nicer on the new java survey page with nice charts.
Maybe @userdeveloper98 finds some time to implement this by times? 😜

Yep agree, we'll need to wait and see if @userdeveloper98  is up for it :)

Fourdee, 2018-10-02:
Getting alot of this during update:
```
TOSTART_OPTION[]: bad array subscript
/tmp/dietpi-survey_report/11f218f5-2ba7-4835-af54-b3e1da5f2e91.txt: line 17: aSOFTWARE[${aSOFTWARE_NAME6_10[]}]++: bad substitution
```

MichaIng, 2018-10-02:
@Fourdee 
Jep known issues.

On earlier versions for it was possible that survey added an empty string as installed software ID. I guess it was due to first run survey, if not .installed file was created yet, or at least nothing yet marked as installed.

The other issue is due to DietPi subversion integer being wrong sometimes in updater. So DietPi is actually v6.10+, thus survey uploads new survey file scheme, but version string added is wrong, so no special aSOFTWARE_NAME array exist for this.

I anyway though that we could clean the survey files a bid, removing those that were not changed for time X. But not urgent, since the errors above do not break anything, just a few software installs are dropped.

Fourdee, 2018-10-02:
```
6103e95d-3a09-4729-b7dd-6bd9f3cb2e7c.txt: line 17: aSOFTWARE[]: bad array subscript
((aSOFTWARE[${aSOFTWARE_NAME_12[5]}]++))
```

Fourdee, 2018-10-02:
@MichaIng 

>  since the errors above do not break anything

Adding checks for bad files and skipping them 👍 

>  since the errors above do not break anything

Might possibly be the cause why only 1 benchmark result is showing, i've done 4, unsure yet:
https://dietpi.com/survey/#benchmark

Fourdee, 2018-10-02:
[image]

Completed, all issues resolved 🈯️ 

You can run a quick benchmark and upload with:
```
/DietPi/dietpi/func/dietpi-benchmark 2
```

View results:
https://dietpi.com/survey#benchmark

MichaIng, 2018-10-04:
@Fourdee 
What do you think about creating results file in a shell source-able way, so it can be directly `cat` into the dietpi-survey upload file and sourced by other scripts as well. Generally I like to do this for most of our data/settings/results files, as sourcing should always be much faster then using `sed` or `grep` to catch content. Also the same values are then forced to have the same variable names across scripts, to have increased consistency.
https://github.com/Fourdee/DietPi/blob/dev/dietpi/func/dietpi-benchmark#L253-L261

MichaIng, 2018-10-06:
@Fourdee 
Currently upload is done automatically, so no "optional uploads". Can we have a switch to have results saved locally or upload them? Or two separate selections for bench only and upload only?

I am thinking as well, if it is okay to bench single devices as well, e.g. CPU only, but then we need to handle empty entries in report.

Fourdee, 2018-10-07:
@MichaIng 

> I am thinking as well, if it is okay to bench single devices as well, e.g. CPU only, but then we need to handle empty entries in report.

The whole purpose of this was to reduce available options, and, support benchmark uploads via 1 menu item.

I removed the option to manually run CPU/rootfs/RAM as its contained within the suite option, which, we want users to run as share/view their results.

> Currently upload is done automatically, so no "optional uploads"

Yep, intended, its part of the survey. However, we should change the "Ok" to run bench, have it a "Yes/No":
[image]

> And currently on VM, 1° CPU temperature is shown.

Yep, i'll see if I can clean it up.

Fourdee, 2018-10-07:
🈯️ 

[image]

MichaIng, 2018-10-07:
@Fourdee 
> The whole purpose of this was to reduce available options, and, support benchmark uploads via 1 menu item.

Hmm, I still think we should allow to run benchmarks without upload. Even that our survey is anonym and safe, I don't like it from end user perspective to be forced to transfer any data. Nothing against reducing options for simplicity, but there should be always a choice, if it's about sharing your data.
E.g. I purged free version of 3dmark instantly from my Windows, after recognizing that bench results are automatically uploaded 😉.

Fourdee, 2018-10-07:
@MichaIng 

Thought the same eariler.

2 tests:
- Online
- Offline

I'll make the change.

Fourdee, 2018-10-07:
Completed.

closed 2018-10-07 by Fourdee