MichaIng/DietPi #1928: DietPi-Software | Missing Minio details for installation options

LexiconCode, 2018-07-16:
Missing Minio details for installation options [DietPi-Software | Details for ALL installation options](https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=9121#p9121)

Details like...

> Login URL: http://192.168.0.100:9000/minio/login
> API key:
> Password:

Fourdee, 2018-07-16:
@LexiconCode 

Thanks for the report 👍 

Yep confirmed. I lack any experience with Minio, so we'll open up for investigations.

Fourdee, 2018-09-30:
Docker info needs doing:
- https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=110#p9120

MichaIng, 2020-09-27:
Brief Docker docs added, mostly because of the recent switch to journald-based logging: https://dietpi.com/phpbb/viewtopic.php?p=9120#p9120

MichaIng, 2020-11-11:
This is now handled at the dedicated DietPi-Docs repository: https://github.com/MichaIng/DietPi-Docs
Minio S3 docs are TBD there as well, however aligning and complementing lists is on the ToDo 🙂.

closed 2020-11-11 by MichaIng