MichaIng/DietPi #1931: First run

amk221, 2018-07-17:
## Creating a bug report/issue:

First run complains about requiring sudo

#### Required Information:
- DietPi version	| `6.11`
- Distro version	| `stretchcat`
- Kernel version	| `Linux DietPi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux`
- SBC device		| `RPi 3 Model B (armv7l)` or (EG: RPi3)
- Power supply used / SDcard used		| Whatever comes with DigiOne Player

#### Steps to reproduce:
1. Flash a card with DietPi_RPi-ARMv6-Stretch.7z
2. Insert into Pi
3. Turn on
4. ssh in dietpi/dietpi

#### Expected behaviour:

- First run script should just run?

#### Actual behaviour:

```
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ───────────────────────────────────────
 DietPi     | 12:44 | Tue 17/07/18
 ───────────────────────────────────────
 v6.11      | RPi 3 Model B (armv7l)
 ───────────────────────────────────────
 eth0       | 192.168.1.154
 ───────────────────────────────────────
tee: /var/tmp/dietpi/logs/dietpi-firstrun-setup.log: Permission denied
[FAILED] Root privileges required. Please run the command with "G_SUDO", or, "sudo -i".
dietpi@DietPi:~$
```

amk221, 2018-07-17:
Apologies, I should have used `DietPi_RPi-ARMv7-Stretch_AlloGUI.7z` instead of `DietPi_RPi-ARMv6-Stretch.7z`. Works fine with that.

closed 2018-07-17 by amk221

MichaIng, 2018-07-17:
@amk221 
Yeah a bid confusing, since usually for RPi2/3 you as well use ARMv6 images/packages due to Raspbian, but in this case ARMv6 is really just for RPi1.