MichaIng/DietPi #1933: DietPi-Software | Nextcloud - Web server fails to properly redirect to request app password from nextcloud client

LexiconCode, 2018-07-17:
#### Required Information:
- DietPi version	|  6.11
- Distro version	| debian stretch
- SBC device		| Odroid C2
- Power supply used	| Official power supply
- SDcard used		| SanDisk ultra

#### Additional Information (if applicable):
- Software title	| Nextcloud
- Was the software title installed freshly or updated/migrated?
Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce:
After install nextcloud and certbot
1. Start desktop next cloud client that is not been set up.
2. Enter https://Domain/nextcloud
3. click on `click here to request app password from the web interface`
4. The redirect goes to `https://Domain/nextcloud/apps/files/?dir=/&fileid=127`

#### Expected behaviour:

The redirect should automatically generate at password and device username to which a user accepts.
I don't remember the exact URL name that should be expected.

#### Actual behaviour:
-  The redirect goes to `https://Domain/nextcloud/apps/files/?dir=/&fileid=127` instead of requesting app password.

#### Extra details:
- Using Nginx Web server.
- configured to force https via certbot

MichaIng, 2018-07-17:
@LexiconCode 
Thanks for your report. I try to replicate.

Which Nextcloud desktop client you use? Linux, Windows or MacOS? Current version I assume?
I try to replicate on Stretch VM.

As we use (mostly) the Nginx configuration from https://docs.nextcloud.com, it is possible that this needs to be addressed to https://github.com/nextcloud/ directly. But let's investigate first.

LexiconCode, 2018-07-17:
@MichaIng 
I'm using the Windows client the latest version. As long as the client supports the feature I assume it would be the same unexpected behavior across all client.

LexiconCode, 2018-07-17:
I apologize I'm using the Odroid N2 not the C2.

LexiconCode, 2018-07-17:
I can confirm this issue exists on Odroid N2 and Odroid C2 running the same configuration.

MichaIng, 2018-07-17:
Can confirm issue as well on Lighttpd and Apache2:
- Initial link: `http://192.168.178.29/nextcloud/index.php/settings/personal?sectionid=security#apppasswords`
- Redirects to `http://192.168.178.29/nextcloud/index.php/apps/files/?dir=/&fileid=2`
- Should redirect to `http://192.168.178.29/nextcloud/index.php/settings/user/security`

I believe this is since the merge of user and admin settings. The initial link at least contains the old path scheme.

MichaIng, 2018-07-29:
GitHub issue opened: https://github.com/nextcloud/desktop/issues/514

closed 2018-08-27 by MichaIng