MichaIng/DietPi #1935: Update nexcloud

ludji49, 2018-07-18:
Création d'un rapport de bogue ou d'un problème :

###### Informations requises :
- Version DietPi | `paste -sd '' /DietPi/dietpi/.version`. 6.11
- Version Distro | `echo $G_DISTRO_NAME` ou `cat /etc/debian_version`.
- Version du noyau | `uname -a`.
- Dispositif SBC | `echo $G_HW_MODEL_DESCRIPTION` EG : RPi3
- Alimentation utilisée | EG : 5V 1A RAVpower
- Carte SD utilisée | EG : SanDisk ultra 

Renseignements supplémentaires (s'il y a lieu) :
- Titre du logiciel |  Nextcloud
- Le titre du logiciel a-t-il été installé récemment ou a-t-il été mis à jour ou migré ? yes
- Ce numéro peut-il être reproduit sur une nouvelle installation de DietPi ? no
- ID `dietpi-bugreport` ID

###### Étapes pour se reproduire :
>>le numéro.
1. ...
2. ...

###### Comportement attendu :
Que DEVRAIT-il se passer ? -->
- ...

###### Comportement réel :
Qu'est-ce qui se passe ? -->
- ...

Traduit avec www.DeepL.com/Translator

ludji49, 2018-07-18:
[image]

MichaIng, 2018-07-18:
@ludji49 
`.well-known` folders are added by various web services and scripts, nothing to worry about.
Simply move it outside your Nextcloud folder and inside again after update has finished:
`mv /path/to/nextcloud/.well-known /tmp/`
Update Nextcloud via web updater
`mv /tmp/.well-known /path/to/nextcloud/`

MichaIng, 2018-07-19:
@ludji49 
1. Copy the command into your putty terminal:
`mv /path/to/nextcloud/.well-known /tmp/`
2. Update Nextcloud as your tried
3. Copy next command into your putty terminal:
`mv /tmp/.well-known /path/to/nextcloud/`

ludji49, 2018-07-19:
there is an error on the writing of the command
[image]

MichaIng, 2018-07-19:
@ludji49 
Ah sorry, `/path/to/nextcloud/` should be replaced with the path to your nextcloud directory. On DietPi it should be then:

`mv /var/www/nextcloud/.well-known /tmp/`

and after Nextcloud update: `mv /tmp/.well-known /var/www/nextcloud/`

ludji49, 2018-07-19:
Update done, but for the next update I have to go through this command 
mv /var/www/nextcloud/.well-known /tmp/
and after Nextcloud update: mv /tmp/.well-known /var/www/nextcloud/

[image]

MichaIng, 2018-07-19:
@ludji49 
That's true.

One of the negative sides of having webroot within /nextcloud directory. Other software tends to place file into webroot, which Nextcloud then identifies as integrity check violation. They already added much of the well known files/folders to be excluded for this, but for some reason `.well-known` still breaks updates.

You could check what is inside and if it's needed: `ls -al /var/www/nextcloud/.well-known`

ludji49, 2018-07-20:
Thank you very much .......

MichaIng, 2018-07-26:
I mark this as closed, as initial issue is answered/solved. Please reopen, if required.

closed 2018-07-26 by MichaIng