MichaIng/DietPi #1937: Auto Setup does not work

okoestner, 2018-07-18:
I try to auto setup my raspi with the above copied dietpi.txt. The setup terminates with an error when trying to install the phpmyadmin package.

I am installing the following image: DietPi_v6.8_RPi-ARMv6-Stretch.img

dietpi.txt
===========

#Modifications to /boot/* files, will not be saved when powered on.
#Please ensure you edit from the DietPi-RamDisk location /DietPi/*

#NB: This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
#NB: Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.

#------------------------------------------------------------------------------------------------------
# DietPi-Automation
# All settings below will be applied on the 1st boot of DietPi, ONCE.
#------------------------------------------------------------------------------------------------------

##### Networking Options #####

#	If both Ethernet and Wifi are enabled, Wifi will take priority and Ethernet will be disabled.
#		1=enabled
AUTO_SETUP_NET_ETHERNET_ENABLED=1
AUTO_SETUP_NET_WIFI_ENABLED=1

#	Enter your Wifi details below, if applicable (Case Sensitive).
AUTO_SETUP_NET_WIFI_SSID=MyWiFi
AUTO_SETUP_NET_WIFI_KEY=MySecret
#	available | WPA-PSK / WPA-EAP / NONE | WEP=No longer supported
AUTO_SETUP_NET_WIFI_KEYMGR=WPA-PSK

#  For WPA2 Enterprise uncomment the following entries and set accordingly
#AUTO_SETUP_NET_WIFI_PROTO=RSN
#AUTO_SETUP_NET_WIFI_PAIRWISE=CCMP
#AUTO_SETUP_NET_WIFI_AUTH_ALG=OPEN
#AUTO_SETUP_NET_WIFI_EAP=PEAP
#AUTO_SETUP_NET_WIFI_IDENTITY=<userid>
#AUTO_SETUP_NET_WIFI_PASSWORD=<userpassword>
#
# If your WPA2 Enterprise requries a certificate copy the cert to this /boot directory and enter the name on the following line
#AUTO_SETUP_NET_WIFI_CERT=wpa2ent_cert.cer
# Uncomment either or both lines below and set accordingly for your WPA2 Enterprise needs.
#AUTO_SETUP_NET_WIFI_PHASE1=
#AUTO_SETUP_NET_WIFI_PHASE2=auth=MSCHAPV2

#	Enter your Static Network details below, if applicable.
AUTO_SETUP_NET_USESTATIC=0
Static_IP=192.168.0.100
Static_Mask=255.255.255.0
Static_Gateway=192.168.0.1
Static_DNS=8.8.8.8

#	Hostname
AUTO_SETUP_NET_HOSTNAME=horst

#	Force ethernet speeds
#		NB: This is mainly aimed towards Pine A64's which may have a HW issue that causes unstable 1Gbit link.
#		0=automatic speed | 10 = 10mbit, 100 = 100mbit etc
AUTO_SETUP_NET_ETH_FORCE_SPEED=0

##### Misc Options #####

#	Size of swapfile to generate (MB)
#		0=Disabled | 1=auto (2GB-RAM = size) | 2+=manual
AUTO_SETUP_SWAPFILE_SIZE=1
#		Optional swapfile location
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap

##### Software Automation Options #####

[... 169 lines cut ...]

#Optional username for ownCloud/Nextcloud admin account, default is 'admin'. Applied during installation.
SOFTWARE_OWNCLOUD_NEXTCLOUD_USERNAME=admin

#Optional data directory for ownCloud, default is '/mnt/dietpi_userdata/owncloud_data'. Applied during installation.
#	This option is for advanced users. For full compatibility, please keep this options defaults, and, use dietpi-drive_manager to move the DietPi user data location.
SOFTWARE_OWNCLOUD_DATADIR=/mnt/dietpi_userdata/owncloud_data

#Optional data directory for Nextcloud, default is '/mnt/dietpi_userdata/nextcloud_data'. Applied during installation.
#	This option is for advanced users. For full compatibility, please keep this options defaults, and, use dietpi-drive_manager to move the DietPi user data location.
SOFTWARE_NEXTCLOUD_DATADIR=/mnt/dietpi_userdata/nextcloud_data

#Wifi Hotspot
SOFTWARE_WIFI_HOTSPOT_SSID=DietPi-HotSpot
#	minimum of 8 characters
SOFTWARE_WIFI_HOTSPOT_KEY=dietpihotspot
SOFTWARE_WIFI_HOTSPOT_CHANNEL=3

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Dev settings
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=master
DEV_GITOWNER=Fourdee
AUTO_SETUP_GLOBAL_PASSWORD=dietpi

Fourdee, 2018-07-18:
@okoestner 

Hi,

Thanks for the report 👍 

This is an known limitation/issue when installing phpmyadmin and lighttpd at the same time:
https://github.com/Fourdee/DietPi/issues/316#issuecomment-217469692

---

Until either we (or external sources) are able to resolve the issue, please try running with nginx/apache 2 stacks:
```
#set nginx, change
AUTO_SETUP_WEB_SERVER_INDEX=-1
```

You can then remove the lighttpd stack, DietPi will automatically install webserver stacks as needed, based on the above preference.:
```
#AUTO_SETUP_INSTALL_SOFTWARE_ID=81
```

MichaIng, 2018-07-19:
@okoestner 
It works, if you install Lighttpd stack first, after this is done and reboot, your can manually install phpMyAdmin. This is the workaround with user notification that we do on interactive installs. But of course on non-interactive installs this does not show up.

@Fourdee 
Maybe we should silently deselect phpMyAdmin on automated first run installs? Currently this is just effective when software is selected within whiptail menu: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L15390-L15399

Also I can't believe it that this is still not fixed on repo side. Maybe we should send a bug report about this: https://www.debian.org/Bugs/Reporting ?

Fourdee, 2018-07-19:
@MichaIng 

> Maybe we should silently deselect phpMyAdmin on automated first run installs? Currently this is just effective when software is selected within whiptail menu: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L15390-L15399

Not a bad idea 👍 , at least until we can do the below.

> Also I can't believe it that this is still not fixed on repo side. Maybe we should send a bug report about this: https://www.debian.org/Bugs/Reporting ?

Yep, although, its been a year or two, we'd need to review and re-investigate it fresh again, before we do.

MichaIng, 2018-08-08:
I will close this in favour of https://github.com/Fourdee/DietPi/issues/316
But added comment there to recheck and in case solve this for automated installs as well.

closed 2018-08-08 by MichaIng