MichaIng/DietPi #1939: Roon bridge not working on v6.12

Kreeblah, 2018-07-20:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.12
- Distro version	| stretch
- Kernel version	| Linux DietPi 3.16.56+ #1 SMP PREEMPT Wed Apr 18 16:59:34 CEST 2018 aarch64 GNU/Linux
- SBC device		| Odroid C2 (aarch64)
- Power supply used	| https://ameridroid.com/products/5v-2a-power-supply
- SDcard used		| https://ameridroid.com/products/emmc-black-module-c2-linux-red-box?variant=7245907460130

#### Additional Information (if applicable):
- Software title	| Roon Bridge
- Was the software title installed freshly or updated/migrated? Both
- Can this issue be replicated on a fresh installation of DietPi? Yes
- ddce5205-bb46-48c6-b6a9-39dd074f93a8 ID

#### Steps to reproduce:

1. Set up a sound card (in my case, the HiFi Shield+) and install ALSA, as usual
2. Install the Roon Bridge package from the DietPi optimized list
3. Check the Roon server's networked audio zones

#### Expected behaviour:

- The ODROID C2 should be listed in the available networked audio zones in the Roon server.

#### Actual behaviour:

- The Roon server doesn't list the bridge at all

#### Extra details:

- This was working on v6.11, but stopped working when I updated to v6.12.  I tried a fresh install just in case something went wrong with the update process, but I'm still seeing the same issue.  I'm also not really sure where the bridge is writing its logs since I don't see them in /var/log, but I do notice it starting and stopping processes like the RoonBridgeHelper and Roon bridge processor if I run ps a few times.

MichaIng, 2018-07-20:
@Kreeblah 
I just recognized and emergency fixed an issue, where `/var/log` ramlog entry was added to fstab during update, even that this was not selected before.

Did you use `dietpi-ramlog` before? If not, please manually remove the entry in `/etc/fstab` again and reboot to recover the old logs.

But that should actually not prevent the service from starting and especially not break something on fresh install.

Okay actually I might found the issue: https://github.com/Fourdee/DietPi/pull/1938/files#diff-d92a6ee04e02fd2a2dc23d5bec3e6a98R8896
We changed user to "roon" but did not change something about the log directory. As this is still /var/log, "roon" user should not have permissions to create log files there.

Please try the following:
```
mkdir -p /var/log/roon
chown roon:dietpi /var/log/roon
rm $G_FP_DIETPI_USERDATA/roon/RoonBridge/Logs
rm $G_FP_DIETPI_USERDATA/roon/RAATServer/Logs
ln -sf /var/log/roon $G_FP_DIETPI_USERDATA/roon/RoonBridge/Logs
ln -sf /var/log/roon $G_FP_DIETPI_USERDATA/roon/RAATServer/Logs
systemctl restart roonbridge
```

Kreeblah, 2018-07-20:
@MichaIng - Yep, that did it.  After doing what you suggested with making that directory with the right permissions and linking it, the Roon bridge software works just fine.

MichaIng, 2018-07-20:
Thanks for the feedback. Hmm I am thinking to fix this within current master as well 🤔. Actually not good practice to change branch after release...

closed 2018-07-20 by MichaIng