MichaIng/DietPi #1940: nfs mounting not working

mgherghi, 2018-07-20:
so after the most recent update i think nfsv3 is working but when i mount the share it gives me the following error:

mount.nfs: access denied by server while mounting 192.168.1.1:/

mgherghi, 2018-07-20:
everything is setup properly on the server side, because i have another client setup using the same method

MichaIng, 2018-07-20:
@mgherghi 
Jep that is a known issue. Hmm we should have made this a bid clearer in changelog 🤔:
See: https://github.com/Fourdee/DietPi/issues/1898#issuecomment-406437446

So you manually need to add the path to the NFS export to the clients `/etc/fstab` entry. We will add full NFSv3 functionality to `dietpi-drive_manager` with v6.13. As it is a bid hacky to get this to work, we couldn't resolve it for v6.12 any more. It was already much luck inside to find the actual reason for general NFSv3 support last yesterday night due to changes on APT repo side (obviously), so forgive us the not yet fully completed solution 😉.

I leave this issue opened and rename it, so we use it for full NFSv3 implementation into dietpi-drive_manager

MichaIng, 2018-07-20:
@mgherghi 
That depends on the NFS server, where the location of the shared folder is found within file system.

On default DietPi NFS server setup it would be:

`192.168.1.1:/mnt/dietpi_userdata` instead of just `192.168.1.1:/`

Which other client setup you have and how exactly did you add the NFS drive there?

MichaIng, 2018-08-08:
@Fourdee 
Did you test on your machines? Can this be closed?

Fourdee, 2018-08-09:
```
sh: 0: getcwd() failed: No such file or directory
Selecting previously unselected package netbase.
(Reading database ... 11685 files and directories currently installed.)
Preparing to unpack .../0-netbase_5.4_all.deb ...
Unpacking netbase (5.4) ...
Selecting previously unselected package libevent-2.0-5:arm64.
Preparing to unpack .../1-libevent-2.0-5_2.0.21-stable-3_arm64.deb ...
Unpacking libevent-2.0-5:arm64 (2.0.21-stable-3) ...
Selecting previously unselected package libnfsidmap2:arm64.
Preparing to unpack .../2-libnfsidmap2_0.25-5.1_arm64.deb ...
Unpacking libnfsidmap2:arm64 (0.25-5.1) ...
Selecting previously unselected package libtirpc1:arm64.
Preparing to unpack .../3-libtirpc1_0.2.5-1.2_arm64.deb ...
Unpacking libtirpc1:arm64 (0.2.5-1.2) ...
Selecting previously unselected package keyutils.
Preparing to unpack .../4-keyutils_1.5.9-9_arm64.deb ...
Unpacking keyutils (1.5.9-9) ...
Selecting previously unselected package rpcbind.
Preparing to unpack .../5-rpcbind_0.2.3-0.6_arm64.deb ...
Unpacking rpcbind (0.2.3-0.6) ...
Selecting previously unselected package nfs-common.
Preparing to unpack .../6-nfs-common_1%3a1.3.4-2.1_arm64.deb ...
Unpacking nfs-common (1:1.3.4-2.1) ...
Setting up libnfsidmap2:arm64 (0.25-5.1) ...
Setting up keyutils (1.5.9-9) ...
Processing triggers for libc-bin (2.24-11+deb9u3) ...
Setting up libtirpc1:arm64 (0.2.5-1.2) ...
Processing triggers for systemd (232-25+deb9u4) ...
Setting up libevent-2.0-5:arm64 (2.0.21-stable-3) ...
Setting up rpcbind (0.2.3-0.6) ...
Created symlink /etc/systemd/system/multi-user.target.wants/rpcbind.service → /lib/systemd/system/rpcbind.service.
Created symlink /etc/systemd/system/sockets.target.wants/rpcbind.socket → /lib/systemd/system/rpcbind.socket.
Setting up netbase (5.4) ...
Setting up nfs-common (1:1.3.4-2.1) ...

Creating config file /etc/idmapd.conf with new version
Adding system user `statd' (UID 106) ...
Adding new user `statd' (UID 106) with group `nogroup' ...
Not creating home directory `/var/lib/nfs'.
Created symlink /etc/systemd/system/multi-user.target.wants/nfs-client.target → /lib/systemd/system/nfs-client.target.
Created symlink /etc/systemd/system/remote-fs.target.wants/nfs-client.target → /lib/systemd/system/nfs-client.target.
nfs-utils.service is a disabled or a static unit, not starting it.
Processing triggers for libc-bin (2.24-11+deb9u3) ...
Processing triggers for systemd (232-25+deb9u4) ...

[  OK  ] DietPi-Drive_Manager | G_AGI: nfs-common netbase
/DietPi/dietpi/dietpi-drive_manager: line 1876: .mount_out: No such file or directory
grep: .mount_out: No such file or directory
```
Stuck in loop, asking me for details