MichaIng/DietPi #1941: Roon Bridge service migration to own user - Files permissions prevent start of service

camelreef, 2018-07-20:
#### Required Information:
- DietPi version	| 6.12
- Distro version	| stretch
- Kernel version	| Linux RPi-Salon 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux
- SBC device		| RPi 2 Model B (armv7l)
- Power supply used	| Official RPi PS, 2.5A 5V
- SDcard used		| SanDisk ultra

#### Additional Information (if applicable):
- Software title	| Roon Bridge
- Was the software title installed freshly or updated/migrated? | It was migrated, but an uninstall, deleting logs and reinstall changes nothing
- Can this issue be replicated on a fresh installation of DietPi? | Don't know, not tried, but probably
- `dietpi-bugreport` ID d963ca2e-6db8-441a-ab40-00d779e10e93

#### Steps to reproduce:
1. Install Roon Bridge from dietpi-software

#### Expected behaviour:
- the Roon Bridge service should be starting normally

#### Actual behaviour:
- The Roon Bridge service fails to start because of failure to open/move files due to wrong file permissions

#### Extra details:
journalctl logs extract | https://pastebin.com/9jESKE4P

MichaIng, 2018-07-20:
Ahhh, forgot the master mirror on dietpi.com: https://dietpi.com/downloads/dietpi-update_mirror/master/
Will bring it up to date!

camelreef, 2018-07-20:
YAY! Glad to have been a trigger!
Thanks for DietPi, I love it!

Used at home this way:
Redhill, UK - http://www.youplala.net/wp-content/uploads/2018/02/2018021201-Multimedia-Setup.pdf
One additional baby bedroom with SqueezeLite and motionEye
All with netdata monitoring

And in other places these ways:

Around Seattle, USA - https://www.youplala.net/wp-content/uploads/2017/12/2017121501-Music-System-Diagrams.png

Dar Es Salaam, Tanzania (before in Dubai, UAE) all migrated to DietPi for the Bridges since - https://www.youplala.net/wp-content/uploads/2017/12/2017011405-Home-setup.png?w=1000

MichaIng, 2018-07-20:
@camelreef 
I really just use a tiny part of SBC potential. It is always nice, involving with DietPi, to get known with the full potential of the software offer. Great what roon, kodi etc. developers do and great that we can help a bid to spread this nice software to people 😍.

Fourdee, 2018-07-20:
@MichaIng 

Great work on the fast fix for this 👍 

Very strange, how this passed local testing for me without issues during dev. I must of missed something, so apologies from my side.

Ok, lets do a hotfix today/tomorrow, just to ensure all users are updated with the hotfix.

MichaIng, 2018-07-20:
@Fourdee 
Yeah, I was hoping to prevent a needed hotfix version, but forgot that DietPi code is downloaded from `dietpi.com`. There should not be much affected due to numbers of RoonBridge installs and updates over the night, but at least all which updates during the night will face an error when installing RoonBridge (new) and some will have tmpfs on `/var/log` who don't chose/uninstalled dietpi-ramlog.

I think, if no other issues appear now, hotfix update is not needed, but lets see.

Fourdee, 2018-07-20:
> I think, if no other issues appear now, hotfix update is not needed, but lets see.

@MichaIng 

Yep, agree 👍

Fourdee, 2018-07-20:
REF: https://community.roonlabs.com/t/dietpi-roonbridge-issues/46535/8?u=dan_knight

MichaIng, 2018-07-27:
I will mark this as closed. @camelreef Thanks again for reporting, I guess this saved many others from running to the same issue 👍.

closed 2018-07-27 by MichaIng