MichaIng/DietPi #1942: process tool applies changes only to the father process

antonellocaroli, 2018-07-20:
## Creating a feature request:

process tool applies changes only to the father process

#### Is your feature request related to a problem? Please describe:

- ...

antonellocaroli, 2018-07-21:
in fact I was trying processtool on another distro (gentoo / open-rc), maybe that's the fact.

that dietpi uses system ... and type for squeezelite there is only one process ...
instead I find 4

Fourdee, 2018-07-21:
@antonellocaroli 

> in fact I was trying processtool on another distro (gentoo / open-rc),

Hi, 

All our scripts require Debian currently to function correctly. It's also not advised to use single DietPi scripts, without 1st running PREP on the system, as DietPi is a complete script system and expects/requires additional script functions contained within other files.

Regardless,

We check for all available processes that match ```aPROCESS_NAME```:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-process_tool#L74-L75

Then we apply the process tool settings to the PID's pulled from the match:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-process_tool#L83

Check ``ps ax``` output, that is what we use to obtain available PID's before matching:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-process_tool#L59

---

As mentioned above, we do not support the distro's you are using at this time. As such, i'll mark this as closed, however, please continue discussions if required.

Fourdee, 2018-07-21:
@MichaIng 
Known issue?

MichaIng, 2018-07-21:
@Fourdee 
Ah sorry, I remember saw once process tool failing to set nice for a related process (logging daemon, other sub/side process), where the name was totally different. Thought this is related. But hum yeah this is a different case. Didn't know SPID and these kind of sub processes. So far, thinking about webserver, PHP etc, all had different PIDs. Hmm yeah lets investigate and see if we can apply process handling for all children of one PID.

Fourdee, 2018-07-21:
🈴 taskset:
```
root@DietPi:~# taskset -pc 1 12385
pid 12385's current affinity list: 0-3
pid 12385's new affinity list: 1

root@DietPi:~# taskset -ac 1 12385
taskset: failed to execute 12385: No such file or directory
```

Fourdee, 2018-07-21:
@antonellocaroli 

Apologies for slamming the "we dont support your distro" initially. From past experience we have to clearly define what we support, else, it becomes impossible.

Great find on this bug and many thanks for reporting it 👍 

The updated script is available here, which now applies to all child threads:
https://raw.githubusercontent.com/Fourdee/DietPi/926ebe115c8f64b775b39877e621614dce3e3d13/dietpi/dietpi-process_tool

If you could confirm this resolves the issue i'd be grateful.

antonellocaroli, 2018-07-21:
Thanks for the change,

now everything seems to work, except for affinity that seems to go on the child processes but not on that father.
[image]

> Apologies for slamming the "we dont support your distro" initially. From past experience we have to clearly define what we support, else, it becomes impossible.

No problem!!! I understand perfectly

I generally use something like that:
```
task_tid=($(ps -C $app -L --no-headers -o tid))

taskset -cp $core ${task_tid[0]}
taskset -cp $core ${task_tid[1]}
taskset -cp $core ${task_tid[2]}
taskset -cp $core ${task_tid[3]}
taskset -cp $core ${task_tid[4]}
```

antonellocaroli, 2018-07-22:
now and then from failed
[image]
[image]

Fourdee, 2018-07-22:
Seems the PID group apply fails on some threads/processes:
```
[  OK  ] DietPi-Process_tool | Plex Media Server (16296) : Affinity  0-3
[  OK  ] DietPi-Process_tool | Plex Media Server (16296) : Scheduler SCHED_OTHER 0
renice: failed to get priority for 16596 (process group ID): No such process
[FAILED] DietPi-Process_tool | Plex DLNA Server (16596) : Nice      0
```

Fourdee, 2018-07-22:
#### Testing https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-process_tool

[image]

```
root@DietPi:~# dietpi-process_tool 1

 DietPi-Process_tool
─────────────────────────────────────────────────────
 Mode: Apply

[  OK  ] DietPi-Process_tool | Cron (26136) : Nice      2
[  OK  ] DietPi-Process_tool | Cron (26136) : Affinity  0
[  OK  ] DietPi-Process_tool | Cron (26136) : Scheduler SCHED_BATCH 0
[  OK  ] DietPi-Process_tool | DHCP Client (652) : Nice      2
[  OK  ] DietPi-Process_tool | DHCP Client (652) : Affinity  1
[  OK  ] DietPi-Process_tool | DHCP Client (652) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | OpenSSH Server (487) : Nice      3
[  OK  ] DietPi-Process_tool | OpenSSH Server (487) : Affinity  2
[  OK  ] DietPi-Process_tool | OpenSSH Server (487) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | OpenSSH Server (6523) : Nice      3
[  OK  ] DietPi-Process_tool | OpenSSH Server (6523) : Affinity  2
[  OK  ] DietPi-Process_tool | OpenSSH Server (6523) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | OpenSSH Server (16678) : Nice      3
[  OK  ] DietPi-Process_tool | OpenSSH Server (16678) : Affinity  2
[  OK  ] DietPi-Process_tool | OpenSSH Server (16678) : Scheduler SCHED_OTHER 0
[  OK  ] DietPi-Process_tool | Squeezelite (26157) : Nice      5
[  OK  ] DietPi-Process_tool | Squeezelite (26157) : Affinity  3
[  OK  ] DietPi-Process_tool | Squeezelite (26157) : Scheduler SCHED_FIFO 30
[  OK  ] DietPi-Process_tool | Completed
```

parent = 26157
threads = 26157 - 26160

[image]

```
root@DietPi:~# taskset -cp 26160
pid 26160's current affinity list: 3
root@DietPi:~# taskset -cp 26159
pid 26159's current affinity list: 3
root@DietPi:~# taskset -cp 26158
pid 26158's current affinity list: 3
root@DietPi:~# taskset -cp 26157
pid 26157's current affinity list: 3
```

```
root@DietPi:~# chrt -v -p 26158
pid 26158's current scheduling policy: SCHED_FIFO
pid 26158's current scheduling priority: 30
root@DietPi:~# chrt -v -p 26157
pid 26157's current scheduling policy: SCHED_FIFO
pid 26157's current scheduling priority: 30
```

@antonellocaroli 

Should be good to go 👍 Many thanks for the ```ps``` tip

antonellocaroli, 2018-07-23:
thank you,

now it seems to work perfectly !!!

great

Fourdee, 2018-07-23:
@antonellocaroli 

> thank you,
now it seems to work perfectly !!!

Excellent, been testing it locally aswell, all seems good. 

I'll mark this as completed. Thanks again for reporting this issue and your assistance 👍

closed 2018-07-23 by Fourdee

antonellocaroli, 2018-07-24:
Personally, I will try to also move the system interrupts (USB and ethernet) on a specific core ....
what do you think?