MichaIng/DietPi #1943: Installing nextcloud, fail on mariadb install

Neoderek, 2018-07-20:
#### Details:
- Date           | Fri 20 Jul 17:52:55 BST 2018
- Bug report     | N/A
- DietPi version | v6.11
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1123 SMP Wed Jun 27 17:35:49 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: mariadb-server
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ... I was installing nextcloud from software optimized, 
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ... Looking the unistalling section this are my installed software:

0    OpenSSH Client:           
2    NULL:                                   
5    ALSA: linux sound system                                                        
6    Xserver: linux display system                                                        
7    FFmpeg: audio & visual libary                                                                  
17   Git Client: git clone etc                                                                     
23   LXDE: ultra lightweight desktop                                                      
31   Kodi: the media centre for linux                                                           
 42   Plex Media Server: web interface media streaming server                                 
44   Transmission: bittorrent server with web interface (c)                                     
66   RPi-Monitor: web interface system stats                                                  
67   NoIp: url website address for your device                                              
81   LLSP: lighttpd | sqlite  | php                                                         
84   Lighttpd: webserver                                                                      
87   SQlite: database                                                               
89   PHP: webserver                                                                        
93   Pi-hole: block adverts for any device on your network               
94   ProFTP: lightweight ftp server                                                         
103  DietPi-Ramlog: minimal, optimized logging                        
105  OpenSSH Server:      
120  RealVNC Server: desktop for remote connection                                                                 
150  Mono: runtime libraries and repo                                              
152  Avahi-Daemon: hostname broadcast (mac, pc bonjour) 

I have other programs installed, out of "software optimization"
https://github.com/JustArchi/ArchiSteamFarm
https://github.com/MichMich/MagicMirror
https://github.com/Unitech/pm2
#### Additional logs:
```
Log file contents:
Setting up mariadb-server-10.1 (10.1.23-9+deb9u1) ...
chown: changing ownership of '/var/lib/mysql': Read-only file system
touch: cannot touch '/var/lib/mysql/debian-10.1.flag': Read-only file system
dpkg: error processing package mariadb-server-10.1 (--configure):
 subprocess installed post-installation script returned error exit status 1
dpkg: dependency problems prevent configuration of mariadb-server:
 mariadb-server depends on mariadb-server-10.1 (>= 10.1.23-9+deb9u1); however:
  Package mariadb-server-10.1 is not configured yet.

dpkg: error processing package mariadb-server (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 mariadb-server-10.1
 mariadb-server
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Fourdee, 2018-07-20:
@Neoderek 

Hi,

Thanks for the report 👍 

```
chown: changing ownership of '/var/lib/mysql': Read-only file system
```

The system is currently running as read only. If you have not specifically set this yourself, I estimate a hardware issue causing filesystem failure.

Please post the following in your reply:
- PSU used (make and model)
- SD card used (brand and size)
- ```dmesg | grep ext```
- USB devices attached (EG: external HDD's)
- If using external HDD's are they self powered (have their own PSU)?

Neoderek, 2018-07-21:
Hi Fourdee, 

I don´t do nothing special to change my sistem to read only.

    Any recent power cuts, and/or, hard resets of power (eg: pulling the power cable out when powered on)? ** Can be possible, and i had internet problems last weeks.**
    PSU used (make and model) **Raspberry pi official psu**
    SD card used (brand and size) It's samsumg, 10 category.
    dmesg | grep ext

> dietpi@DietPi:~$ dmesg | grep ext
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=27504eef-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
                     .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.006276] CPU: Virtualization extensions available.
[    1.274091] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[   40.901714] Adding 1097724k swap on /var/swap.  Priority:-2 extents:6 across:2506752k SSFS

    USB devices attached (EG: external HDD's) Only 1 HDD
    If using external HDD's are they self powered (have their own PSU)? The hdd is only conected to the raspberry pi

Fourdee, 2018-07-21:
@Neoderek 

Thanks for the info 👍 

The issue is with the rootFS. Read-Only is applied by the kernel automatically when an issue occurs to prevent damage.

So it would appear either a power issue has occurred, or, the SD card itself is failing.

As you are using the official PSU (i assume 5V/2.5A version?), this should be more than enough to power USB drive (1.2A max shared across all USB sockets).

I cant see any errors with ```dmesg```, however, we should check the following:
```
[ 1.274091] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
```

Please can you paste results, I just want to ensure its not set to read only:
```
cat /etc/fstab
```

---

Failing the above, it may be advisable to re-write the DietPi image and start from scratch on a fresh SD card.

Neoderek, 2018-07-27:
sorry for all this time whit not answer. 

I resolved the problem. I unistall mariadb manually and i check the manager-drive options i see a option for read-only. 2 times fail to uncheck the option, but... at 3rd i change it and now i can install whitout problems nextcloud.

The problem was the hdd. 

This is the actually fstab: 

root@DietPi:~# cat /etc/fstab
#-----------------------------------------------------------
#NETWORK
#-----------------------------------------------------------
#Please use DietPi-Drive_Manager to setup network mounts

#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
proc    /proc           proc    defaults                                                                                                                        0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs /var/log tmpfs defaults,size=50m,noatime,nodev,nosuid,mode=1777 0 0
tmpfs   /DietPi         tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777                                        0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#INTERNAL DRIVES
#-----------------------------------------------------------
PARTUUID=27504eef-02 / auto defaults,noatime,rw 0 1
PARTUUID=27504eef-01 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=007e509f-11d7-4c11-8a96-c48966bfef63 /mnt/external auto defaults,noatime,rw,nofail,x-systemd.automount 0 0

Thanks for your work! U help a lot to us (noobies)

MichaIng, 2018-07-27:
@Neoderek 
Great to here, thanks for the feedback. I mark this issue as closed. Feel free to reopen, if there are further questions/problems.

closed 2018-07-27 by MichaIng