MichaIng/DietPi #1946: Update MiniDLNA with additional features

burgyman, 2018-07-21:
Hi,
On top of DietPi I have installed [Entware](https://entware.net/about.html) because of the improved version of MiniDLNA:
- supports on-the-fly Thumbnail extraction and generation (requires ffmpeg)
- option to select a PNG image to use for an icon
- db rescan and rebuild

Package list: http://bin.entware.net/armv7sf-k3.2/Packages.html

Any chance to update MiniDLNA to the same version or with similar set of features?
Thanks.

Fourdee, 2018-07-21:
@burgyman 

Many thanks for the request 👍 

Currently all devices will install MiniDLNA from APT.

For this to be possible, we need to compile, provide and maintain binaries for:
- ARMv6
- ARMv7
- ARMv8
- x86_64

As such, i'll mark this as low priority for now.

---

In regards to refresh/rescan DB, this can be done already:
https://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p49
```
rm /mnt/dietpi_userdata/.MiniDLNA_Cache/* && systemctl restart minidlna
```

MichaIng, 2019-10-30:
Debian Buster ships the most current upstream version, which is also available for Stretch via backports: https://packages.debian.org/buster/minidlna

RPi2/3/4 Stretch systems can as well apply Debian Stretch backports, if really required, so only RPi1/Zero (ARMv6 ones) require an own build.

Since otherwise the project seems dead (no new upstream since 2017: https://sourceforge.net/projects/minidlna/files/minidlna/), this single build would stay current, as for now.

However, since we ship Raspbian Buster for a while now, and especially RPi1/Zero users are strongly advised to migrate to Buster (due to many other Stretch-ARMv6 related issues, PHP7.3 only via Buster repo and such), I'm not going to invest time to compile and host this. If someone else did it anyway or wants to do it, I'm however happy to host and add it do our default install for RPi1+Zero Stretch systems. Until then I mark this issue as closed.

closed 2019-10-30 by MichaIng