MichaIng/DietPi #1947: /etc/cron.hourly/dietpi fails to remove /var/lib/dietpi/dietpi-ramlog/storage/

sergiud, 2018-07-21:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `6.12`
- Distro version	| `stretch`
- Kernel version	| `4.14.32+`
- SBC device		| Odroid XU3/XU4/HC1/HC2 (armv7l)
- Power supply used	|  15V 5A
- SDcard used		| eMMC

#### Additional Information (if applicable):
- Software title	| cron

#### Steps to reproduce:

1. Enable notifications in OpenMediaVault
2. Update from 6.11 to 6.12

#### Expected behaviour:

-  No error notification

#### Actual behaviour:

- Failure to execute hourly `dietpi` script

#### Extra details:

The removal of `/var/lib/dietpi/dietpi-ramlog/storage/` fails only a single time after the update from 6.11 to 6.12.

The `rm` calls here:

https://github.com/Fourdee/DietPi/blob/12efbfa207928764441e49c0588655986315723d/rootfs/etc/cron.hourly/dietpi#L35

and here

https://github.com/Fourdee/DietPi/blob/12efbfa207928764441e49c0588655986315723d/rootfs/etc/cron.hourly/dietpi#L42

are missing a `[[ -d /var/lib/dietpi/dietpi-ramlog/storage ]]` check and instead try to remove the directory in question unconditionally.

MichaIng, 2018-07-21:
@sergiud 
Thanks for your report.

We changed dietpi-ramlog storage location but forgot to adjust this within cron job: https://github.com/Fourdee/DietPi/commit/e85f2efa39bd7d3867f06a804149c47c45c3170e#diff-9beb868c8b43c2fb4363fc06b6fab912

The error message is harmless, this part of the cron job uncritical, just intended to be failsafe. Adding directory existence check is a good idea anyway 👍.

We will fix this for v6.13

MichaIng, 2018-07-22:
Okay we simply removed the ramlog storage update from cron job. The idea of writing (even that those are just empty files => each 4k block) files to disk, while ramlog is intended to reduce disk I/O is not worth it. /var/log file/folder structure will be stored on reboot, which is scheduled after every dietpi-software and dietpi-update run.

I will mark the issue as closed, as the fix is very trivial now :).

closed 2018-07-22 by MichaIng