MichaIng/DietPi #1950: dietpi-globals: line 27: cd: HOME not set

mrwhy-orig, 2018-07-22:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.12

#### Additional Information (if applicable):
- Software title	|  Chromium Kiosk
DietPi upgraded from 6.4 to 6.12

#### Steps to reproduce:

Activate Chromium in Autostart options

#### Expected behaviour:
Open Chromium at startup

#### Actual behaviour:
Console is displaying the HOME not set error and asks for Login. With Enter the login is done automatically to root without any Password.

#### Extra details:
I have changed the default password. I'd like to Chromium with restricted user. Although if i start chromium from command line on Raspi with 7" display attached, chromium gets display on 2/3 of the display, not fullscreen

MichaIng, 2018-07-22:
@mrwhy-orig 
Thanks for your report.
Jep already found and fixing this for v6.13: https://github.com/Fourdee/DietPi/pull/1945

At least it is harmless, you can ignore these messages, or do:
`sed -Ei 's/^[[:blank:]]*cd \$HOME/\t\[\[ \$HOME \]\] \&\& cd \$HOME/' /DietPi/dietpi/func/dietpi-globals`

mrwhy-orig, 2018-07-23:
@MichaIng, ignoring these message isn't an option, because the chromium kiosk mode startup fails without the fix.
But your change is working. Thx!

closed 2018-07-23 by mrwhy-orig