MichaIng/DietPi #1952: Installing plex media server Failed

Tuaz, 2018-07-22:
#### Details:
- Date           | Sun 22 Jul 12:53:57 BST 2018
- Bug report     | N/A
- DietPi version | v6.12
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | #1 SMP PREEMPT Wed Apr 18 16:59:34 CEST 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: binutils:armhf plexmediaserver-installer:armhf
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

- I was installing Plex media server from software optimized

#### Expected behaviour:

- ...

#### Actual behaviour:
- [FAILED] DietPi-Software | G_AGI: binutils:armhf plexmediaserver-installer:armhf

#### Extra details:

- Installed software
- 46 qBitTorrent
- 144 Sonarr
- 147 Jackett
- 115 Webmin
- 96 Samba
- 117 PiVPN

#### Additional logs:
```
Log file contents:
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

MichaIng, 2018-07-22:
@Tuaz 
Thanks for your report. I try to replicate.

Fourdee, 2018-07-22:
Unable to replicate with local testing. We'll need to debug user system directly.

@Tuaz 

Lets try running the commands manually, please paste the output:
```
apt-get install binutils:armhf plexmediaserver-installer:armhf
```

> E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

None of the installed software has held packages (to my knowledge), please also try running APT fix, please paste output aswell:
```
apt-get install -f
```

Tuaz, 2018-07-22:
apt-get install binutils:armhf plexmediaserver-installer:armhf
```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libmono-system-configuration4.0-cil : Depends: libmono-system-security4.0-cil (>= 1.0) but it is not going to be installed
 libmono-system-data4.0-cil : Depends: libmono-system-enterpriseservices4.0-cil (>= 1.0) but it is not going to be installed
                              Depends: libmono-system-numerics4.0-cil (>= 1.0) but it is not going to be installed
                              Depends: libmono-system-transactions4.0-cil (>= 1.0) but it is not going to be installed
 libmono-system-web4.0-cil : Depends: libmono-sqlite4.0-cil (>= 3.0.6) but it is not going to be installed
                             Depends: libmono-system-componentmodel-dataannotations4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
                             Depends: libmono-system-drawing4.0-cil (>= 3.0.6) but it is not going to be installed
                             Depends: libmono-system-enterpriseservices4.0-cil (>= 1.0) but it is not going to be installed
                             Depends: libmono-system-runtime-serialization-formatters-soap4.0-cil (>= 1.0) but it is not going to be installed
                             Depends: libmono-system-web-applicationservices4.0-cil (>= 1.0) but it is not going to be installed
                             Depends: libmono-system-web-services4.0-cil (>= 1.0) but it is not going to be installed
 mono-runtime : Depends: mono-runtime-sgen (= 5.12.0.226-0xamarin3+debian9b1) but it is not going to be installed
 mono-xsp4 : Depends: mono-xsp4-base (= 4.5-0xamarin3+debian9b1) but it is not going to be installed
 monodoc-manual : Depends: monodoc-browser but it is not going to be installed or
                           monodoc-http but it is not going to be installed or
                           monodoc-viewer
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

```
apt-mark showhold
`shows nothing`

apt-get install -f
`0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

Fourdee, 2018-07-22:
@Tuaz 

Thanks 👍 

Appears mono is the issue (yet again 🤣 )
Try reinstalling mono:
```
 dietpi-software uninstall 150
 dietpi-software install 150
```

failing that, please send us a bug report so we can investigate system configuration:
```
dietpi-bugreport
```

Tuaz, 2018-07-22:
It worked after reinstalling 150, but i send you bug report because i miss read you not sure if any useful now :)

closed 2018-07-23 by Tuaz