MichaIng/DietPi #1953: Failed to install software

GulyFMG, 2018-07-22:
#### Details:
- Date           | Sun 22 Jul 14:01:15 BST 2018
- Bug report     | N/A
- DietPi version | v6.12
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | #1123 SMP Wed Jun 27 17:35:49 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

New install of 6.12 trying to install kodi, radarr and sonarr using dietpi-software to do so.

#### Expected behaviour:

- ...

#### Actual behaviour:

- Fails to install any software

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Hit:1 http://download.mono-project.com/repo/debian raspbianstretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:3 https://apt.sonarr.tv develop InRelease [13.3 kB]
Hit:4 https://archive.raspberrypi.org/debian stretch InRelease
Get:5 https://apt.sonarr.tv develop/main armhf Packages [105 kB]
Ign:5 https://apt.sonarr.tv develop/main armhf Packages
Get:5 https://apt.sonarr.tv develop/main armhf Packages [124 kB]
Ign:5 https://apt.sonarr.tv develop/main armhf Packages
Get:5 https://apt.sonarr.tv develop/main armhf Packages [839 kB]
Err:5 https://apt.sonarr.tv develop/main armhf Packages
  GnuTLS recv error (-9): A TLS packet with unexpected length was received.
Fetched 13.3 kB in 2s (5,512 B/s)
Reading package lists...
E: Failed to fetch https://apt.sonarr.tv/dists/develop/main/binary-armhf/Packages  GnuTLS recv error (-9): A TLS packet with unexpected length was received.
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

--------------------------------------------------------------------

[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

MichaIng, 2018-07-22:
@GulyFMG 
Thanks for your report. I will do some test installation here as well.

@Fourdee 
Is it intended that we use the development branch instead of master (stable)?
https://github.com/Sonarr/Sonarr/wiki/Installation#linux

MichaIng, 2018-07-22:
@GulyFMG 
To fix:
`sed -i 's|https://apt.sonarr.tv|http://apt.sonarr.tv|g' /DietPi/dietpi/dietpi-software`
and retry installation.

GulyFMG, 2018-07-22:
@MichaIng Thanks for the workaround finished the institucional successfully.

closed 2018-07-22 by GulyFMG

MichaIng, 2018-07-22:
@GulyFMG 
Great to hear, thanks for feedback 🙂.