MichaIng/DietPi #1955: ODROID WiFi Adapter 3 Won't Work as HotSpot

stevejenkins, 2018-07-22:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.12
- Distro version	| stretch
- Kernel version	| Linux DietPi 3.16.56+ #1 SMP PREEMPT Wed Apr 18 16:59:34 CEST 2018 aarch64 GNU/Linux

- SBC device		| Odroid C2 (aarch64)
- Power supply used	|  5.2V 1.6A (Amazon Echo power adapter)
- SDcard used		| SanDisk ultra

#### Additional Information (if applicable):
- Software title	| WiFi HotSpot
- Was the software title installed freshly or updated/migrated: Fresh
- Can this issue be replicated on a fresh installation of DietPi: Yes
- dietpi-bugreport: f3724239-6ba3-47d3-bdc9-a2245528deb0

#### Steps to reproduce:
1. Connect ODROID WiFi Adapter 3 (Realtek RTL8188CUS)
2. Configure adapter via dietpi-config
3. Install WiFi HotSpot via dietpi-software

#### Expected behaviour:
WiFi SSID should broadcast

#### Actual behaviour:
WiFi SSID does not broadcast and wlan0 does not appear to have correct settings

#### Extra details:
    root@DietPi:~# iwconfig wlan0
    wlan0     IEEE 802.11bgn  ESSID:off/any  
              Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm   
              Retry short limit:7   RTS thr=2347 B   Fragment thr:off
              Encryption key:off
              Power Management:off`

    root@DietPi:~# cat /etc/hostapd/hostapd.conf
    interface=wlan0
    driver=rtl871xdrv
    ssid=MobilePi
    hw_mode=g
    channel=9
    macaddr_acl=0
    auth_algs=1
    ssid=MobilePi

    root@DietPi:~# hostapd /etc/hostapd/hostapd.conf -dd
    random: Trying to read entropy from /dev/random
    Configuration file: /etc/hostapd/hostapd.conf
    drv->ifindex=4
    l2_sock_recv==l2_sock_xmit=0x0x48c3b0
    ioctl[SIOCSIWMODE]: Invalid argument
    Could not set interface to mode(3)!
    Could not set interface to master mode!
    rtl871xdrv driver initialization failed.
    hostapd_interface_deinit_free(0x48b950)
    hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
    hostapd_interface_deinit(0x48b950)
    wlan0: interface state UNINITIALIZED->DISABLED
    hostapd_bss_deinit: deinit bss wlan0
    wlan0: AP-DISABLED 
    hostapd_cleanup(hapd=0x48cba0 (wlan0))
    hostapd_free_hapd_data: Interface wlan0 wasn't started
    hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
    hostapd_interface_free(0x48b950)
    hostapd_interface_free: free hapd 0x48cba0
    hostapd_cleanup_iface(0x48b950)
    hostapd_cleanup_iface_partial(0x48b950)
    hostapd_cleanup_iface: free iface=0x48b950

Fourdee, 2018-07-23:
Thanks for the report 👍 

Confirmed bug:
```
Bus 001 Device 003: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wi reless Adapter [Realtek RTL8188CUS]
```

stevejenkins, 2018-07-23:
Cool. In the meantime any word on chipsets that are known to work with ODROID + DietPi in Hotspot mode?

Fourdee, 2018-07-25:
@stevejenkins 

> FYI, tested a different adapter with same results.

Strange, it may be a kernel related issue. We'll need to do some investigations and further testing when possible.

stevejenkins, 2018-07-25:
Sounds good. This isn't a mission-critical device for me. I built it as a mobile network speed tester (that's why I went ODROID instead of RPi to get a true GigE Ethernet adapter), and wanted to be able to connect to it wirelessly and run the tests from a tablet. So I'm happy to install any additional drivers, kernel versions, or alpha DietPi bits to get to the bottom of it. Thanks!

MichaIng, 2019-04-22:
@stevejenkins 
It has been a long time. However if this issue is still not solved for your:
- We ship a custom hostapd package/binary for RTL8188C* chips, which at least was required. This regular package did not support any of those chips.
- Turns out that at least in some cases (Edimax RTL8188CUS on RPi) the official Debian/Raspbian package now supports these chips.

At least it is worth trying:
```
G_AGI --reinstall hostapd
G_CONFIG_INJECT 'driver=' 'driver=nl80211' /etc/hostapd/hostapd.conf
hostapd /etc/hostapd/hostapd.conf -dd
```

Ref:
- https://github.com/MichaIng/DietPi/issues/2531
- https://github.com/MichaIng/DietPi/issues/2673

MichaIng, 2019-08-01:
Ref: https://dietpi.com/phpbb/viewtopic.php?p=18984#p18984

MichaIng, 2020-11-29:
I mark this issue as closed now. Basically it is known that not all WiFi adapters support hotspot mode and there is not much we can do about this.

closed 2020-11-29 by MichaIng