MichaIng/DietPi #1956: Helios4 Support Request

g-provost, 2018-07-23:
## Creating an image request:

#### Give us some formal device information:
- Device name | Helios4
- Official product URL | https://kobol.io/helios4
- GitHub resource URL (if available) | https://github.com/helios-4 and https://github.com/armbian/build
- Image download URL | https://wiki.kobol.io/download/

#### Is the SBC officially supported by the Debian installer?
- No. But note that upstream support for Kernel and U-Boot mainline is almost completed.
Linux Kernel Mainline support : 80% upstream (ongoing effort soon to be 100%)
U-Boot Mainline support : 100% upstream

#### If not, is a reliable 3rd party Debian image available for this SBC?
- Official supported by Armbian with hardware 100% supported

#### If not, are there install instructions for Debian available?
https://wiki.kobol.io/install
https://www.armbian.com/helios4/

MichaIng, 2018-07-23:
@g-provost 
Thanks for your request. Sadly do not offer images, which are based on ARMbian due to constant friction we had (and by times still have) with their devs. To solve/avoid this, the only solution was to completely remove any need for ARMbian from DietPi.

But as this is open source, you are allowed to create your own image, by running our DietPi image creation script on the fresh ARMbian image: https://raw.githubusercontent.com/Fourdee/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh

This should work on all not too much outdated Debian based images, however we cannot guarantee it in every case of manual system manipulations or special firmware needs.

As soon as it's possible to use the official Debian installer on it or having another non ARMbian reliant image source, we will have another look.

g-provost, 2018-08-13:
@MichaIng Can you elaborate on the aspects of the Armbian builds that don't fit to make it a baseline for DietPi ? This could help.

Because the following statement when we are talking about Arm SBC is not really helping / encouraging.
> As soon as it's possible to use the official Debian installer on it or having another non ARMbian reliant image source, we will have another look.

Fourdee, 2018-08-19:
@g-provost 

> Can you elaborate on the aspects of the Armbian builds that don't fit to make it a baseline for DietPi ?

The issue is not the ARMbian build of the image. The issue is with ARMbian themselves in relation to us. We simply cannot work together for a better cause. We've tried, been trolled, had arguments and I finally pulled their images from our lineup to put a end to it.
To reduce friction and pointless childish trolling, we no longer provide their images, or, have anything to do with them.

In regards to how we create DietPi images:
We rely on an existing pre-image with working uboot/kernel and modules. If the RootFS is not already Debian, we replace it with our own.

As their is no current non-ARMbian based images available for this device, and, we do not compile images from scratch (its not what we do, and, would eat up 99% of our time). I will mark it as closed.

closed 2018-08-19 by Fourdee