MichaIng/DietPi #1957: Qbittorrent settings dont save on restart

mgherghi, 2018-07-23:
for some reason i cant get the settings on the app to stick after a reboot.

Fourdee, 2018-07-23:
@mgherghi 

Thanks for the report 👍 

We'll open for investigations.

Fourdee, 2018-07-23:
Config file is not being updated by webUI:
```
/home/qbittorrent/.config/qBittorrent/qBittorrent.conf
```

I see the issue:
```
root@DietPi:~# ls -lha /home/qbittorrent/
total 28K
drwxr-xr-x 4 qbittorrent qbittorrent 4.0K Jul 23 20:23 .
drwxr-xr-x 4 root        root        4.0K Jul 23 20:22 ..
-rw-r--r-- 1 qbittorrent qbittorrent  220 May 15  2017 .bash_logout
-rw-r--r-- 1 qbittorrent qbittorrent 3.5K May 15  2017 .bashrc
drwxr-xr-x 3 root        root        4.0K Jul 23 20:22 .config
drwxr-xr-x 3 qbittorrent dietpi      4.0K Jul 23 20:23 .local
-rw-r--r-- 1 qbittorrent qbittorrent  675 May 15  2017 .profile
```

---

## Solution:
```
chown -R qbittorrent:dietpi /home/qbittorrent
dietpi-services restart
```

Fourdee, 2018-07-23:
@mgherghi 

We'll get this resolved for v6.13, however, you can fix it now by using the following commands:
https://github.com/Fourdee/DietPi/issues/1957#issuecomment-407173961

Fourdee, 2018-07-23:
Completed, patched in v6.13.

closed 2018-07-23 by Fourdee