MichaIng/DietPi #1958: Emby Server shutdown when idle .

mgherghi, 2018-07-23:
I have tried to install the emby server from the dietpi-software menu (v3.4.1) and then update the app according to the website (emby.com) to v3.5.  I feel that if i do that update or just install the emby server directly from the website (emby.com) , it does not work properly. Specifically when the server is idle for more than 10 minutes i need to restart the rpi or type in 'dietpi-services restart" otherwise i am unable to access it through a remote connection.  I would like to stay up to date with the emby updates as there are certain bugs that have been fixed in the newest version of emby..  What should I do ?   

I am on dietpi v6.12

Fourdee, 2018-07-23:
@mgherghi 

Hi,

Currently, we use the http://download.opensuse.org repo for installing Emby packages. Updates are applied from the repo itself.

The upgrade guide on Emby.com most likely does not accommodate for this repo installation, or, there is a bug in the latest version causing the fail after 10 minutes.

When Emby fails after 10 minutes, please run this command and paste results:
```
systemctl status emby-server -l
```

We may be able to debug.

mgherghi, 2018-07-24:
root@DietPi:~# systemctl status emby-server -l
● emby-server.service - Emby Server is a personal media server with a
pps on just about every device.
   Loaded: loaded (/usr/lib/systemd/system/emby-server.service; enabled; vendor
preset: enabled)
   Active: active (running) since Tue 2018-07-24 02:04:23 BST; 5h 12m
in ago
 Main PID: 18530 (EmbyServer)
   CGroup: /system.slice/emby-server.service
           └─18530 /opt/emby-server/system/EmbyServer -programdata /var/lib/emby
 -ffmpeg /opt/emby-server/bin/ffmpeg -ffprobe /opt/emby-server/bin/ffprobe -rest
artexitcode 3 -updatepackage emby-server-deb_{version}_armhf.deb

Jul 24 03:09:39 DietPi emby-server[18530]:   libpostproc    55.  1.100 / 55.  1.
100
Jul 24 03:09:39 DietPi emby-server[18530]: Input #0, matroska,webm, from 'file:/
mnt/Seagate/Some.Kind.Of.Beautiful.2014.720p.WEBRip.x264.AC3-m2g/Some.Kind.Of.Be
autiful.2014.720p.WEBRip.x264.AC3-m2g.mkv':
Jul 24 03:09:39 DietPi emby-server[18530]:   Metadata:
Jul 24 03:09:39 DietPi emby-server[18530]:     encoder         : libmkv 0.6.4.1
Jul 24 03:09:39 DietPi emby-server[18530]:   Duration: 01:39:19.96, start: 0.000
000, bitrate: 2884 kb/s
Jul 24 03:09:39 DietPi emby-server[18530]:     Chapter #0:0: start 0.000000, end
 5959.958500
Jul 24 03:09:39 DietPi emby-server[18530]:     Metadata:
Jul 24 03:09:39 DietPi emby-server[18530]:       title           : Chapter 1
Jul 24 03:09:39 DietPi emby-server[18530]:     Stream #0:0(eng): Video: h264 (Ma
in), yuv420p(tv, bt709, progressive), 1280x534 [SAR 1:1 DAR 640:267], 24 fps, 24
 tbr, 1k tbn, 180k tbc (default)
Jul 24 03:09:39 DietPi emby-server[18530]:     Stream #0:1(eng): Audio: ac3, 480
00 Hz, 5.1(side), fltp, 384 kb/s (default)

mgherghi, 2018-07-24:
its like emby goes to sleep and it cant wakeup unless i restart the service, i even tried the "wake server" option

Mgherghina, 2018-07-24:
@fourdee

Fourdee, 2018-07-24:
@mgherghi 

> its like emby goes to sleep and it cant wakeup unless i restart the service, i even tried the "wake server" option

Strange, service looks fine, its running and nothing of note in the log.

Was the service status command used after the web UI is dis-functional?

The only thing I can think of is updating to v3.5, via the guide, introduces an instability (or bug) with the program itself. 
Ideally, to ensure stability and full support from our end, please stick with the version DietPi installs via the added APT repo.

As this is outside our scripts and support, i'll mark the issue as closed, however, please continue discussions if required.

closed 2018-07-24 by Fourdee