MichaIng/DietPi #1959: This error when trying to update software

Roxas72, 2018-07-24:
Hello guys, this is my first time opening an issue so sorry in advance if I'm missing something 

#### Details:
- Date           | Tue 24 Jul 16:06:41 BST 2018
- Bug report     | N/A
- DietPi version | v6.11
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Kernel version | #1123 SMP Wed Jun 27 17:35:49 BST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:2 https://archive.raspberrypi.org/debian stretch InRelease
Err:3 https://archive.raspberrypi.org/debian stretch Release
  503  Service Unavailable
Hit:4 http://dev2day.de/pms stretch InRelease
Reading package lists...
E: The repository 'https://archive.raspberrypi.org/debian stretch Release' does no longer have a Release file.
```

MichaIng, 2018-07-24:
@Roxas72 
Thanks for your report.

Everything available here: https://archive.raspberrypi.org/debian/dists/stretch/
Maybe it was a temporary issue do to repo mirroring.

Please retry to manually update the package lists: `apt update`

Roxas72, 2018-07-24:
Thank you @MichaIng for your answer.
It seems though that after rebooting the RPI the system info shows the software as v6.12, the updated one. How is it possible? Because the update should have failed..

MichaIng, 2018-07-24:
@Roxas72 
Hmm, just tested in my case and as the the new version string is written to disk after all update steps are done, also after reboot it shows the old version 🤔.

If you are sure that the update didn't finish, you can rerun it via `dietpi-update -1`.

Roxas72, 2018-07-24:
@MichaIng well for the time being I'm gonna leave it as it is, as long as it works! 
Can I trouble you in this same thread with another problem I happen to have? It's a service-related one (SickRage), can you help me?

Fourdee, 2018-07-24:
@Roxas72 

> Can I trouble you in this same thread with another problem I happen to have? It's a service-related one (SickRage), can you help me?

We'll help if we can, please post the issue.

Roxas72, 2018-07-24:
@Fourdee alright thanks guys! 
So basically after trying to update the software (and failing, a couple of times, aaand to be honest unplugging the power) I've noticed that Sickrage wasn't working anymore. With a little bit of investigation, or in other words `dietpi-services status` , here's what it says: 
```
[FAILED] DietPi-Services | sickrage	failed (Result: exit-code) since Tue 2018-07-24 18:32:42 BST; 3h 54min ago
● sickrage.service - SickRage
   Loaded: loaded (/etc/systemd/system/sickrage.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-07-24 18:32:42 BST; 3h 54min ago
  Process: 7396 ExecStart=/usr/bin/python /etc/sickrage/SickBeard.py -q --daemon --nolaunch --datadir=/mnt/dietpi_userdata/sickrage (code=exited, status=2)

Jul 24 18:32:41 DietPi systemd[1]: Failed to start SickRage.
Jul 24 18:32:41 DietPi systemd[1]: sickrage.service: Unit entered failed state.
Jul 24 18:32:41 DietPi systemd[1]: sickrage.service: Failed with result 'exit-code'.
Jul 24 18:32:42 DietPi systemd[1]: sickrage.service: Service hold-off time over, scheduling restart.
Jul 24 18:32:42 DietPi systemd[1]: Stopped SickRage.
Jul 24 18:32:42 DietPi systemd[1]: sickrage.service: Start request repeated too quickly.
Jul 24 18:32:42 DietPi systemd[1]: Failed to start SickRage.
Jul 24 18:32:42 DietPi systemd[1]: sickrage.service: Unit entered failed state.
Jul 24 18:32:42 DietPi systemd[1]: sickrage.service: Failed with result 'exit-code'.
```
I have tried to restart the service but nothing, any other suggestions other than the dumb and easy one I've chosen?

Fourdee, 2018-07-24:
@Roxas72 

Appears the service is stuck in a fail loop.

Try running the command manually:
-  stop services
```
dietpi-services stop
```
- Run the command and check for any errors/output
```
/usr/bin/python /etc/sickrage/SickBeard.py -q --daemon --nolaunch --datadir=/mnt/dietpi_userdata/sickrage
```

MichaIng, 2018-07-24:
@Roxas72 
Btw.:
In such cases of service fail loops, you can see more (possibly the initial issue) via `journalctl`, then what `dietpi-services status` respectively `systemctl status <servicename>` shows. The latter show just the tail of all their output.

Roxas72, 2018-07-25:
@Fourdee 
This is what the command gave me: 
```
/usr/bin/python: can't open file '/etc/sickrage/SickBeard.py': [Errno 2] No such file or directory
``` 
I might not be an expert, but that's pretty odd.. how does the directory not exist since it was working till yesterday?

Fourdee, 2018-07-25:
@Roxas72 

Thanks for the info 👍 

> /etc/sickrage/SickBeard.py

This is the old install location, as of v6.12 its changed to:
```
$G_FP_DIETPI_USERDATA/sickrage/SickBeard.py
```

So we probably assume from this that the update which failed on your device, is the cause.

#### Ok, probably the best thing to do is simply rerun the update. If any errors occurs, please paste them in your next post so we can review/check.
```
dietpi-update -1
```

Roxas72, 2018-07-25:
@Fourdee 
Thank you very much, the update solved the problem!! 
I know it was an easy issue but you guys were really helpful, I appreciate it!
Keep up the good work, DietPi is definitely the best os I've ever tried on my RPI!

Fourdee, 2018-07-25:
> Thank you very much, the update solved the problem!!

Excellent, great to hear 👍 

I'll mark this as resolved.

closed 2018-07-25 by Fourdee