MichaIng/DietPi #1960: Temporary RaspberryPi APT repository issues

MichaIng, 2018-07-25:
https://archive.raspberrypi.org/

APT update currently fails, which also leads to `dietpi-update` and `dietpi-software` (G_AGUP) failing.

amk221, 2018-07-26:
I got this today, re-running `dietpi-update` says the upgrade worked ok, (6.11 -> 6.12), even though the above failed. 

How can I re-run `dietpi-update` to re-attempt this, if it thinks everything ok?

MichaIng, 2018-07-26:
@amk221 
Very strange, I already read this somewhere else, that even in case of APT failure, the update went on, or at least the version string changed. I cannot replicate this. In case of APT error, at least here, a whiptail menu pops up, showing error details and offering retry or configure the APT mirrors, the script does not go on and the update (including version string) is not fully applied.
But I also found slightly different APT errors currently on the repos. Perhaps the error changed into a "warning" in between, which allows the script to go on (error code 0).
E.g. yesterday I found:
```
W: Failed to fetch https://archive.raspberrypi.org/debian/dists/stretch/InRelease  503  Service Unavailable [IP: 93.93.135.188 443]
W: Some index files failed to download. They have been ignored, or old ones used instead.
E: Problem parsing dependency 2
E: Error occurred while processing kodi-pvr-hts (NewVersion2)
E: Problem with MergeList /var/lib/apt/lists/archive.raspberrypi.org_debian_dists_stretch_main_binary-armhf_Packages
E: The package lists or status file could not be parsed or opened.
```
Today the first two warnings disappeared, just the dependency error remained. I haven't installed Kodi, so it is en error (syntax maybe, dunno) within the repo files.

But either way, even if the script cancels, the DietPi script files were already updated. Actually this is not the best behaviour, as some changed scripts might need the incremental update steps of the [patch_file](https://github.com/Fourdee/DietPi/blob/master/dietpi/patch_file).

I will change it to first run APT update & upgrade and download DietPi scripts afterwards, so in case of APT error, we fully stay on previous version.

But in your case now:
- If version string changed already, the update should have been fully applied, there is no other way.
- Thus only the APT packages did not get updated, which is usually not a problem. I never found a case where the current DietPi version relies on up-to-date APT packages. The versions shipped with our images, should still work with v6.12.
- As long as the repo maintainers solved the current issues, run `G_AGUP && G_AGUG` to bring packages on current stage.
- If you are unsure, and better want to rerun the whole update, then `dietpi-update -1` will do this. It reduces the version string internally by -1 to reapply the current patch.

amk221, 2018-07-26:
Thanks for the detailed response

MichaIng, 2018-07-27:
Issue solved here. I will close.

closed 2018-07-27 by MichaIng