MichaIng/DietPi #1963: Support for BananaPI M1?

johnnyasantoss, 2018-07-26:
#### Details:                                                                                                                                                  
- Date           | Thu 26 Jul 10:16:17 -03 2018                                                                                                                
- Bug report     | N/A                                                                                                                                         
- DietPi version | n/a                                                                                                                                         
- Img creator    | n/a                                                                                                                                         
- Pre-image      | n/a                                                                                                                                         
- SBC device     | BananaPI M1 (index=53)                                                                                                                             
- Kernel version | # 9 SMP PREEMPT Sat May 6 12:20:11 UTC 2017                                                                                                  
- Distro         | jessie (index=3)                                                                                                                            
- Command        | systemctl daemon-reload                                                                                                                     
- Exit code      | 127                                                                                                                                         
- Software title | DietPi-PREP                                                                                                                                 
                                                                                                                                                               
#### Steps to reproduce:                                                                                                                                       
 - Have a fresh bananian install
 - execute the PREP script                                                                                                                                               
                                                                                                                                                               
#### Expected behaviour:                                                                                                                                       
- Upgrade to dietpi                                                                                                          
                                                                                                                                                               
#### Actual behaviour:                                                                                                                                         
- Breaks because of missing systemd (I don't know why, but I can't install `systemd-sysv` in this image)                          
                                                                                                                                                               
#### Additional logs:                                                                                                                                          
```                                                                                                                                                            
Log file contents:                                                                                                                                             
./dietpi-globals: line 1291: systemctl: command not found                                                                                                      
```

johnnyasantoss, 2018-07-26:
[dpkg --get-selections output](https://pastebin.com/mZhQQQE2)

MichaIng, 2018-07-26:
@johnnyasantoss 
Ui, a pure SysV Init image, even lacking it within APT repo. Must be a very old one? Where did you download the Debian image from and which Debian version is it?

In those cases the APT sources need to be updated first to allow installing SystemD.

@Fourdee 
As we had this already one time, I think we should handle systemd-less images somehow:
- Without systemd available, exit script, as the image is too ancient or too much modified from default Debian? With default wheezy (lowest version we allow) even systemd is available on default repo: https://packages.debian.org/wheezy/systemd-sysv
- Otherwise we would need to reset the APT repo first. If (as in this case) no systemd is installed at all (in the other case, there was an ancient version preinstalled at least), this would mean a whole Init system switch. I don't know if we want or can support this within DietPi-PREP, otherwise a meaningful error message would be good, that `systemd-sysv` needs to be installed manually by user and running on the systemd.

johnnyasantoss, 2018-07-26:
@MichaIng I know that bananian is 3yr+ old but It uses Debian (jessie) apt sources as its default. I'm from Brazil and I've tried using every (working) source near me (from https://www.debian.org/mirror/list), still no luck trying to find `systemd-sysv` in my system.

I'll flash another linux distro into my sd card later today, hope I find anything _newer_ than this one.

johnnyasantoss, 2018-07-26:
btw: I thought that DietPi would "fix" my apt repos in order to find `systemd-sysv`.
PS: I'm new to SBCs.

Fourdee, 2018-07-26:
@MichaIng 

> I don't know if we want or can support this within DietPi-PREP, otherwise a meaningful error message would be good, that systemd-sysv needs to be installed manually by user and running on the systemd.

Yep agree 👍 

```systemctl``` is installed via the ```systemd``` main package:
https://packages.debian.org/stretch/amd64/systemd/filelist

I'll make systemd package a pre-req to PREP.

@johnnyasantoss 

Simply installing systemd might resolve this (or at least give us more errors we can look into)
```
apt-get install systemd
```

> btw: I thought that DietPi would "fix" my apt repos in order to find systemd-sysv.

It will do this, however, only after globals are loaded (which currently pre-req's systemd)

johnnyasantoss, 2018-07-26:
I've tried to install `systemd` and `systemd-sysv` none of them were listed on my sources. I'm formatting my sd card again and now I'll install a more updated distro.

MichaIng, 2018-07-26:
@johnnyasantoss 
At least we can see, if the repo then is the only issue. Try:
```
echo 'deb https://deb.debian.org/debian/ jessie main contrib non-free' >> /etc/apt/sources.list
apt-get update
apt-get install systemd
```

Ah hmm I overread that you actually added a repo from https://www.debian.org/mirror/list ? Hmm then systemd should be definitely available. Did you add the repo as above? Does `apt-get update` show any error messages?

johnnyasantoss, 2018-07-26:
@MichaIng No, it didn't show any messages, it was only showing a message that other packages already have a dependency in systemd but it wasn't found.
I can reinstall that distro and try again just for the sake of you guys know what was happening.

johnnyasantoss, 2018-07-26:
```bash
100 root@bananapi ~ # apt-get install systemd systemd-sysv                                                                                                  :(
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package systemd-sysv is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  sysvinit-core

Package systemd is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libsystemd-dev

E: Package 'systemd' has no installation candidate
E: Package 'systemd-sysv' has no installation candidate
```

johnnyasantoss, 2018-07-26:
```bash
100 root@bananapi ~ # apt install libsystemd-dev                                                                                                            :(
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libsystemd-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libsystemd-dev' has no installation candidate
```

johnnyasantoss, 2018-07-26:
I'll now try to use a different Debian flavor(? I don't know what Bananian should be called)

Fourdee, 2018-07-27:
@johnnyasantoss 

Strange, that should not be occurring.

> 80 to remove and 801 not upgraded.    

Does it fully list the packages for removal? There may be a ARMbian based package flagged for removal which is also is flagging ```init systemd-sysv``` for removal.

johnnyasantoss, 2018-07-27:
Yep! after upgrading the package `apt` and running `apt autoremove -y --allow-remove-essentials` the script finished successfully! :tada:
Using base image `Armbian_5.38_Bananapipro_Debian_stretch_next_4.14.14.7z` to DietPi :1st_place_medal:

johnnyasantoss, 2018-07-27:
Can I help someone with this image that I made?
Eg: upload it somewhere?

Fourdee, 2018-07-27:
@johnnyasantoss 

Good stuff, glad to hear its all working now 👍 

> Can I help someone with this image that I made?
Eg: upload it somewhere?

We will not host ARMbian images on the DietPi servers. Apologies. You'll need to host it on your own site/server. 
Regardless, kind of you to offer this image to others 👍 

I'll mark this as closed as the issue is now resolved.

closed 2018-07-27 by Fourdee

johnnyasantoss, 2018-07-27:
Awesome! I'll write about this in my blog. Thanks for helping me out!