MichaIng/DietPi #1966: External harddrive disconnect automatically

roelfoubert, 2018-07-27:
My external harddrive can be disconnected automatically after a various time. In drive_manager it is visible but not accessible by /mnt/...

Fourdee, 2018-07-27:
@roelfoubert 

Its most likely one of the following:
- The drive is failing
- The USB device is not getting enough power

Please can you confirm:
- SBC device (eg: RPi Zero)
- PSU used (eg 5v/1a)
- USB drive used, and, does it have its own PSU?

roelfoubert, 2018-07-28:
I use a raspberry pi 3 with a powersupply of 5V/2.4A
I have 3 drives connected. 2 normal usb drives and one external harddrive. Non of them with own PSU

Fourdee, 2018-07-28:
@roelfoubert 

> I use a raspberry pi 3 with a powersupply of 5V/2.4A
I have 3 drives connected. 2 normal usb drives and one external harddrive. Non of them with own PSU

Might be slightly over the available 1.2A max shared across all USB ports.

To verify power is the issue, when the system has been running for a few hours, check the following command for any voltage issues:
```
dmesg | grep volt
```

roelfoubert, 2018-07-28:
This is the info:
root@RaspberryPi:~# dmesg | grep volt
[   33.271502] Under-voltage detected! (0x00050000)
[   39.511627] Under-voltage detected! (0x00050005)
[   54.071520] Under-voltage detected! (0x00050005)
[ 2186.072408] Under-voltage detected! (0x00050005)
[ 3184.471604] Under-voltage detected! (0x00050005)
[ 3217.751806] Under-voltage detected! (0x00050005)
[ 3261.431646] Under-voltage detected! (0x00050005)
[ 5439.191526] Under-voltage detected! (0x00050005)
[11535.671558] Under-voltage detected! (0x00050005)

Fourdee, 2018-07-28:
@roelfoubert 

Thanks, appears issue is caused by:
- Insufficient PSU, which is not providing stable 5V. Please use the official RPi PSU 5v/2.5A.

You also have filesystem issues (due to the above) on your drives. I'd highly recommend:
- Purchasing an official RPi PSU 5v/2.5A
- Writing a new DietPi image and starting from scratch
- Formatting all the drives prior to use

As this is a hardware issue, with known cause of PSU, i'll mark this as closed.

closed 2018-07-28 by Fourdee