MichaIng/DietPi #1967: Shellinabox | Web interface SSH client

Fourdee, 2018-07-27:
- Install by default on Allo GUI images
- Install by default on all DietPi images?
```
Need to get 115 kB of archives.
After this operation, 506 kB of additional disk space will be used.
```
[image]

[image]

Fourdee, 2018-07-27:
```
/etc/default/shellinabox
SHELLINABOX_ARGS="--no-beep --disable-ssl"
```

MichaIng, 2018-07-28:
@Fourdee 
Nice, if one needs to access from many different clients and/or mobile, where not always SSH client is available. But for security reasons I would always prefer a real SSH client where possible. Also for this reason `--disable-ssl` should be never added? Standalone webserver included which needs SSL certificates? Perhaps we can add self-signed/snakeoil certs first and then handle it within `dietpi-letsencrypt` as well.

For Allo GUI images it could be interesting to add it and implement into the Allo GUI? But on other images I would not add it by default. So far no one asked for it (?) and due to above reasons I also would not use it. I access to my server from local network only.

sparkysbc, 2018-08-21:
without adding  --disable-ssl  users can still access by https://192.168.0.100:4200   ,then need to add permission on advanced option
by adding --disable-ssl  192.168.0.100:4200 or http://192.168.0.100:4200

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+25

MichaIng, 2021-11-07:
This functionality is basically covered by our new DietPi-Dashboard: #448
And installing Shellinabox, when wanted is as easy as: https://packages.debian.org/bullseye/shellinabox
```sh
apt install shellinabox
```
So a `dietpi-software` implementation would be overkill. I'll mark this hence as closed.

closed 2021-11-07 by MichaIng