MichaIng/DietPi #1975: ZNC IRC Bouncer

OakGravy, 2018-07-29:
ZNC IRC Bouncer
https://wiki.znc.in/ZNC
https://wiki.znc.in/Installation

First time using DietPi and I must say its fantastic on my Odroid HC1, though quite surprised ZNC isnt already among the software available in "DietPi-Software" as it is relatively simple to install manually, perhaps this is why it's not in the list? In any case I think it would be a nice addition.

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+11

Ryujinra, 2019-02-10:
I would very much like to see this added as well. Most distributions ship very outdated versions of znc, forcing a build from source every time its updated; if DietPi could ship even the latest Debian package it would be a great help.

Korpip, 2020-05-05:
Would be very welcomed.  I voted.

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not addressed soon. The FeatHub request stays however open to allow votes and if anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng