MichaIng/DietPi #1980: V6.12 update on Arm 71 error

Jester1979ve, 2018-07-31:
Hello dears, below issue about updating dietpi V6.11 to 6.12
waitng your advices to solve the problem, Regards.

Vincenzp
#### Details:
- Date           | Sun 10 Jun 22:18:51 CEST 2018
- Bug report     | N/A
- DietPi version | v6.11
- Img creator    | n/a
- Pre-image      | n/a
- SBC device     | Sparky SBC (armv7l) (index=70)
- Kernel version | #22 SMP PREEMPT Fri Dec 22 10:45:41 IST 2017
- Distro         | stretch (index=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 logitechmediaserver : Depends: libio-socket-ssl-perl but it is not installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

Jester1979ve, 2018-08-02:
Hello,

I tried has you explained above but still NTP issue.
I attached picture with issue shown.
waiting Your feedback.
Regards

[image]

MichaIng, 2018-08-02:
@Jester1979ve 
> Date | Sun 10 Jun 22:18:51 CEST 2018
> ...
> but still NTP issue.

Ah okay network time sync fails in your case, which can be an explanation for a lot of errors, especially when it comes to SSL/HTTPS connections.

Please try the following:
```
G_CONFIG_INJECT 'CONFIG_NTP_MIRROR=' 'CONFIG_NTP_MIRROR=default' /DietPi/dietpi.txt
/DietPi/dietpi/func/dietpi-set_software ntpd-mode 2
/DietPi/dietpi/func/run_ntpd 1
```

If this somehow does not work, there would be something wrong with your internet connection. But you could also try to use your local router as NTP server. In this case open `/etc/systemd/timesyncd.conf` and replace the value (should be 0.debian.pool.ntp.org 1.debian.pool.ntp.org ...) with your router IP.

Jester1979ve, 2018-08-02:
Hello,

I tried first tre lines of commands
after the last one : /DietPi/dietpi/func/run_ntpd 1

It shows: Warning: systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (57/60)
Warning: systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (58/60)
Warning: systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_ntpd | NTPD: Waiting for completion of systemd-timesyncd (59/60)
Warning: systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_ntpd | NTPD: time out waiting for systemd-timesyncd
**[FAILED] NTPD: time sync | An issue has occured**
[ INFO ] DietPi-Run_ntpd | NTPD: Please check the service status for more information:
[ INFO ] DietPi-Run_ntpd | NTPD: systemctl status systemd-timesyncd -l

With command line: /etc/systemd/timesyncd.conf

It says:

root@DietPi:~# /etc/systemd/timesyncd.conf
-bash: /etc/systemd/timesyncd.conf: **Permesso negato** "it means access denied"

MichaIng, 2018-08-02:
@Jester1979ve 
Please copy this to your terminal paste results (it takes 5 seconds waiting time in between):
```
systemctl start systemd-timesyncd
sleep 5
systemctl status systemd-timesyncd
systemctl stop systemd-timesyncd
cat /etc/systemd/timesyncd.conf
```

MichaIng, 2018-10-14:
I mark this issue as closed, since we addressed known time sync issues with past updates.

closed 2018-10-14 by MichaIng