MichaIng/DietPi #1983: Chromium custom.sh is using only half display

mrwhy-orig, 2018-08-02:
Required Information:
     RPi 3
    DietPi version | 6.12

Additional Information (if applicable):

    Software title | Chromium Kiosk
    DietPi 6.12

Steps to reproduce:
Activate Chromium custom.sh in Autostart options

Expected behaviour:
Open Chromium at startup in full screen kiosk mode.
Actual behaviour:
Chromium is opening in windowed mode. Half screen. 

Extra details:

If I log in with root and start the script manually chromium opens full screen. I changed the .config/chromium/Default/Preferences to get chromium display full screen on a FullHD screen. Before i did that, chromium has been displayed on 2/3 of the screen. Is it possible that chromium uses another Preferences file as it gets started in autostart? Which user is starting the process?

Fourdee, 2018-08-03:
@mrwhy-orig 

Hi,

Thanks for the report 👍 

Lets clarify which autostart option is selected. Please paste results:
```
cat /DietPi/dietpi/.dietpi-autostart_index
```

Ideally, when running Chromium under Kiosk, you should use the following from ```dietpi-autostart```
```
'11' ': Chromium - Dedicated use, without desktop'
```

Please clarify if the above is selected.

mrwhy-orig, 2018-08-03:
@Fourdee 

no i'm using the 14 

with the following script:

`#!/bin/bash

url="http://mywebsite.local"
chromium_opts="--kiosk --ash-host-window-bounds='1920,1080' --test-type --start-maximized"
xinit /usr/bin/chromium-browser ${chromium_opts} ${url}
`

As long as I start it manually from shell it works perferctly fine. But if it is called at startup, chromium is displayed only on one half of the screen.

I tried the option 11 as well, but this uses just a part of the display to. It looks like 720p.

Fourdee, 2018-08-04:
@mrwhy-orig 

> Now I found the chromium autostart script and altered the window size there. Now it is working as expected with autostart option 11. Thx!

That was going to be my next suggestion. Great to hear its all resolved 👍 

For reference, the startup file is located and can be modified below:
```
/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
```

I'll mark this as resolved. Please reopen if required.

closed 2018-08-04 by Fourdee