MichaIng/DietPi #1984: custom filter not honored when running backup with dietpi-backup

nnovaes, 2018-08-02:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.12
- Distro version	| stretch
- Kernel version	| Linux DietPi 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
- SBC device		| `Virtual Machine (x86_64)
- Power supply used	| N/A
- SDcard used		| none

#### Additional Information (if applicable):
- Software title	| dietpi-backup
- Was the software title installed freshly or updated/migrated? fresh
- Can this issue be replicated on a fresh installation of DietPi? yes
- `dietpi-bugreport` ID ==> Failed to connect to SFTP server

#### Steps to reproduce:

1. Modify /DietPi/dietpi/.dietpi-backup_inc_exc through Custom filter menu
2. Add line  "- /mnt/dietpi/dietpi_userdata/TV/" to the file
3. Save

#### Expected behaviour:
Directory /mnt/dietpi/dietpi_userdata/TV/ is excluded from backups

#### Actual behaviour:
Directory is included in backup
- ...

#### Extra details:

- ...

MichaIng, 2018-08-02:
@nnovaes 
Many thanks for your report.

Just tested:
```
...
[  OK  ] DietPi-Backup | rsync -aH --delete --delete-excluded --exclude-from=/tmp/.dietpi-backup_filter_inc_exc -v --log-file=/var/log/dietpi-backup.log / /mnt/dietpi-backup/data/
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : cron
[ SUB2 ] DietPi-Process_tool > Apply
[  OK  ] DietPi-Process_tool | Completed
2018-08-02 16:36:17 root@VM-Stretch:~# l /mnt/dietpi-backup/data/mnt/dietpi_userdata/
Music/    Pictures/ Video/
...
2018-08-02 16:36:17 root@VM-Stretch:~# cat /DietPi/dietpi/.dietpi-backup_inc_exc
#DietPi-Backup | Custom include/exclude filters
#
#To INCLUDE (+) a file/folder:
#+ /location/to/my/file
#+ /location/to/my/directory/
#
#To EXCLUDE (-) a file/folder:
- /mnt/dietpi_userdata/downloads
#- /location/to/my/directory/
```
`- /mnt/dietpi_userdata/downloads/` Works here as well 🤔.

DietPi-Backup was modified last on v6.10, thus my v6.13 testing image behaves the same.

Ah, another test shows the problem:
```
2018-08-02 16:43:26 root@VM-Stretch:~# l /mnt/dietpi-backup/data/mnt/dietpi_userdata/
total 16
drwxrwxr-x 2 dietpi dietpi 4096 Jul 20 18:51 downloads
drwxrwxr-x 2 dietpi dietpi 4096 Jul 29 18:53 Music
drwxrwxr-x 2 dietpi dietpi 4096 Jul 20 18:51 Pictures
drwxrwxr-x 2 dietpi dietpi 4096 Jul 20 18:51 Video
2018-08-02 16:43:48 root@VM-Stretch:~# cat /DietPi/dietpi/.dietpi-backup_inc_exc
#DietPi-Backup | Custom include/exclude filters
#
#To INCLUDE (+) a file/folder:
#+ /location/to/my/file
#+ /location/to/my/directory/
#
#To EXCLUDE (-) a file/folder:
- /mnt/dietpi_userdata/
```

It seems, that excludes are ignored, if there are files inside the excluded folders. 🤔 crazy that no one recognized this so far.

nnovaes, 2018-08-02:
root@DietPi:/# cat /DietPi/dietpi/.dietpi-backup_inc_exc
#DietPi-Backup | Custom include/exclude filters
#
#To INCLUDE (+) a file/folder:
#+ /location/to/my/file
#+ /location/to/my/directory/
#
#To EXCLUDE (-) a file/folder:
#- /location/to/my/file
#- /location/to/my/directory/

- /mnt/dietpi/dietpi_userdata/MoviesO/
- /mnt/dietpi/dietpi_userdata/TV/
- /mnt/dietpi/dietpi_userdata/Movies/

*********************
this is not the entire file but just for referenche...

tPi-Backup: rsync -aH --delete --delete-excluded
xclude-from=/tmp/.dietpi-backup_filter_inc_exc -v
og-file=/var/log/dietpi-backup.log / /mnt/dietpi-backup/data/
Exit code: 20
DietPi version: v6.12 | HW_MODEL:20 | HW_ARCH:10 | DISTRO:4
Image creator: DietPi Core Team
Pre-image: Debian 9

cat /var/log/dietpi-backup.log
2018/08/02 16:13:36 [52867] .d..tp..... mnt/
2018/08/02 16:13:36 [52867] cd+++++++++ mnt/dietpi_userdata/
2018/08/02 16:13:36 [52867] cd+++++++++ mnt/dietpi_userdata/Movies/
2018/08/02 16:13:36 [52867] >f+++++++++ mnt/dietpi_userdata/Movies/.DS_Store
2018/08/02 16:13:36 [52867] >f+++++++++ mnt/dietpi_userdata/Movies/._.DS_Store
2018/08/02 16:13:36 [52867] cd+++++++++ mnt/dietpi_userdata/Movies/Almost Famous (2000)/
2018/08/02 16:16:58 [52867] >f+++++++++ mnt/dietpi_userdata/Movies/Almost Famous (2000)/Almost.Famous.2000.EXTENDED.1080p.BluRay.H264.AAC-RARBG.mp4
2018/08/02 16:16:58 [52867] cd+++++++++ mnt/dietpi_userdata/Movies/GoodFellas (1990)/
2018/08/02 16:20:14 [52867] >f+++++++++ mnt/dietpi_userdata/Movies/GoodFellas (1990)/Goodfellas.1990.REMASTERED.1080p.BluRay.H264.AAC-RARBG.mp4

nnovaes, 2018-08-02:
The contents of that file up to the lines i pasted are posted at [pastebin](https://pastebin.com/FSJAe2ja)

MichaIng, 2018-08-02:
@nnovaes 
> /mnt/dietpi/dietpi_userdata/Movies/
> ...
> 2018/08/02 16:13:36 [52867] cd+++++++++ mnt/dietpi_userdata/Movies/

Ah, did you move your data directory to `/mnt/dietpi/dietpi_userdata`?

As DietPi will leave a symlink at `/mnt/dietpi_userdata`, which is explicitly added to inclusion list (as /mnt parent is excluded), I guess it follows this symlink (usually it does not follow symlinks).

Please try to add `/mnt/dietpi_userdata/Movies/` etc to the list instead. `/mnt/dietpi/` will anyway never be copied, as usually within /mnt you have other drives and we just want/need to backup rootfs + dietpi_userdata (everything that is not excluded).

nnovaes, 2018-08-02:
it does actually  work in the way you described and those directories were skipped

closed 2018-08-05 by MichaIng