MichaIng/DietPi #1985: Folding@Home

nnovaes, 2018-08-02:
## Creating a software request:

#### Give us some formal software information:
- Software title | Folding@Home
- Short description | _Folding@home (FAH or F@h) is a distributed computing project for disease research that simulates protein folding, computational drug design, and other types of molecular dynamics. As of today, the project is using the idle resources of personal computers owned by volunteers from all over the world. Thousands of people contribute to the success of this project._
- GitHub URL (if available) |  https://github.com/FoldingAtHome
- External URL (if available) | https://foldingathome.org/
- (Official) install documentation | https://foldingathome.org/support/faq/installation-guides/linux/

#### Are there similar/alternative software titles available with DietPi-Software?
no

#### What makes your requested software better than the above solutions, if available?
N/A

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

lower the entry barrier to get it up and running, set the proper priority to not affect other processes (e.g. when dietpi box is performing other resource heavy tasks), etc

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
no
#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

Sorry, I am totally unexperienced in bash coding.

MichaIng, 2018-08-03:
@nnovaes 
Many thanks for your request. This is indeed an interesting project, if you have a server that is not constantly used, even if it's just at night. I am not sure how much it makes sense to run it on a RPi or similar SBC, but the larger ones and amd64 machines could be a help. Also having a 3D viewer makes it interesting. I will try it out on my VM,

Official packages are (I guess reasonably) just available for amd64: https://foldingathome.org/alternative-downloads/

Hmm, which repo is actually for the base fahclient, just see fahviewer and fahcontrol 🤔: https://github.com/FoldingAtHome
Would be good to check if it's even possible to build this for ARM platforms.

MichaIng, 2018-08-03:
So, to sum up, I like the idea to help in disease research. Currently Debian packages are just available for `x86_64`, and as there is no source code for the client itself, I think there is currently no way to get it running on an ARM SBC.

Configuration is quite easy. Remote access should be added as default, maybe it's possible to allow access for the whole network group (x.x.x.*), will try it. Or allow global access but then use passkey. I hope this is asked for on web UI then.

But for users with server on native PC or VM this is an idea. We could create a DietPi team, I guess tasks are somehow shared within the team and I like the idea to let our servers working together on something useful 😄.

MichaIng, 2018-08-03:
@nnovaes 
Let's wait for Fourdee to review. But since I didn't find out yet, how to create a group, would be awesome if you do it then.

Btw: Web UI does not open any more after a while, instead constantly reloading the page. Just found an open issue about this, found that forced reloading the page (clearing browser cache) resolves it: https://github.com/FoldingAtHome/fah-web-client/issues/6

Fourdee, 2018-08-04:
> So, to sum up, I like the idea to help in disease research
But for users with server on native PC or VM this is an idea. We could create a DietPi team, I guess tasks are somehow shared within the team and I like the idea to let our servers working together on something useful 

I'am all for this 👍 Great idea, and, a for a purpose that benefits the human race.

I'll be able to test this on me Ryzen 5 😃 

@MichaIng 

Is this something you want to pickup (you've done all the research, dont want to take over without asking), or, want me to add it in to DietPi?

> yeah, a DietPi team would be awesome! Let me know if you want me to create it.

Thanks for offering 👍 I've created the team, details below:
[image]

MichaIng, 2018-08-04:
@Fourdee 
Let me make a start. Never added a software title so far, if I remember correctly 😄.

Fourdee, 2018-08-06:
@MichaIng 

> Shall we handle it via dietpi-services? I just had the problem, that using systemctl stop/restart FAHClient somehow didn't stop/restart the parental process, while /etc/init.d/FAHClient stop/restart did, even that ExecStop=/etc/init.d/FAHClient stop is set within the generated systemd unit. Restarting the service leads to up to 1% working unit process is lost. No big issue for my impression.

Ideally, we need to handle this in services and process_tool.

In cases like this, where they still use sysinit services. I generally just make a new systemD service. I can take a look at this if needed, once merged and I can test.

> At least user name and passkey (if requested) and power choice need to be entered interactively.
Do we want to allow to enter these via dietpi.txt for fully automated install?

Unsure at the moment, really depends if this is unique, or, only applies to local system (in which case we could set a default user/pw)?

> Move data directory to /mnt/dietpi_userdata? The size is not too large and we do not support ARM SBCs (SDcards) anyway (37% of current work unit):

When working with .deb installs, we need to be careful how we do this. Ideally we should move, then symlink any userdata to DietPi userdata dir, and, logs to /var/log etc.

> There is no fitting software category. Create a new one: e.g. Scientific projects, or Misc or merge into another category by renaming slightly?

Software projects (new item, similar to hardware projects) might be best?

It could also fit under Cloud, however, everything in this day and age fits into the overly abused and corrupted Cloud definition lol.

MichaIng, 2018-08-06:
> In cases like this, where they still use sysinit services. I generally just make a new systemD service. I can take a look at this if needed, once merged and I can test.

Actually the auto created systemd service finally worked as expected. I think due to install/update etc. back and forth, there was an issue in between. But reboot should fix it, at least it worked then well for me. So no manual systemd unit creation needed.

> Unsure at the moment, really depends if this is unique, or, only applies to local system (in which case we could set a default user/pw)?

According to what I could derive (finished one work unit on the VM, yeah 570 points for DietPi Team 😃), user name is local, until the first work unit god uploaded. I will rested by adding my user name without passkey (requested one) on VM again. I think the error or warning appears then on result upload attempt.
Also if no passkey was requested, I guess everyone could collect for the same user. There are machine counts for your user name etc.

Actually an idea is to create a `DietPi` user and optionally request a passkey for this. This can then be added by default, empty passkey or the one we get. Then users are free to just collect "anonymously" as "DietPi" user for "DietPi" team, or create their own user name and optionally passkey if desired.

Thinking about passkey or not... Who would be sad if someone else collects points for the own user? But everyone could attach his email to the user name and request a passkey. Then one will be not able to access his own user any more. Thus yeah, `DietPi` user needs a passkey then.

Fourdee, 2018-08-06:
@MichaIng 

> Another idea: Computing

Hmm, it has a very wide definition. Technically, any of the Software titles could fit under it?

```Computing Science```? or ```Science stuff```? 😃 
```Computing simulation / computer modeling```?

You decide for this one 👍 

> As this needs a mail address, it's up to you if/which mail to use here. Also I am not 100% sure, if a shared account is wanted/allowed. Maybe ask for this, just to be sure?

Ok, lets merge what you have completed now, I'll run some tests and take a look.

Fourdee, 2018-08-06:
[image]

Silly question, how do I access the web client remotely? 

> "Computational Science" at least fully fits according to wikipedia? 

Sounds good 👍

MichaIng, 2018-08-06:
@Fourdee 
Web access is possible via port `7396`, but only if the 5 seconds were enough for FAHClient to write back the config file after stopping. Will test as well again, if this always works.

Fourdee, 2018-08-07:
@MichaIng 

Once ```Run_Move_Data``` is completed, it should break on next loop with:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/func/dietpi-set_userdata#L156-L161

> Does pgrep checks within loop for FahCore respectively FAHClient parent process. Max 10 seconds each loop, otherwise show error message, but do not exit install for now. In worst case user needs to enable web access by himself.

Perfect, nice one 👍

MichaIng, 2018-08-07:
@Fourdee
It mostly uses just or core on my laptop (Windows). Yesterday on testing on VM, and after raising their virtual cores to all 4 host threads, I found on using 3 cores.
I think, also remembering what I read, that the amount of used cores also depends on the current work unit, but not sure. 

Yeah service shut down takes some time and leads to partly result loss, if not yet saved to disk (each %, on Windows each 15 minutes,  I think). But since services are not restarted all the time, it is acceptable? Or did you face issues with the work unit afterwards?

Fourdee, 2018-08-08:
@MichaIng 

> I think, also remembering what I read, that the amount of used cores also depends on the current work unit, but not sure.

Yep, looks that way 👍 

I'll have to try running 3VM's with 2cores each, see how many points I can achieve today! 😃 

>  Or did you face issues with the work unit afterwards?

I'll need to recheck, however, I believe stopping/starting the service actually allowed for two instances of the application. The previous one (which was trying to finish) and the new one (trying to start).

MichaIng, 2018-08-08:
@Fourdee 
Hmm, okay (lets stay with discussion here). So the pgrep pattern needs to be reviewed. Actually I worry a bid, if those test installs, which always start on a WU, but never finish it, somehow delay the research process. Would be nice if we could somehow "free" the WU, so others can instantly start with it.

But disabling autostart for testing is no option, is exactly that full start is what we need to test/track 🤔.

Fourdee, 2018-08-08:
@MichaIng 

Mmmm, could we pre-create the config.xml, before deb install? As a last resort, might do the job?
Ah i see the IP detection issues.

Testing

[image]

MichaIng, 2018-08-08:
@Fourdee 
Hmm, the problem is that the settings are/need to match the system, regarding CPU/core count, GPU etc. I think it will cause more problem, if we somehow try to create a most-compatible one. The client needs to check and pre-create configuration.

Fourdee, 2018-08-08:
@MichaIng 

```
root@DietPi:~# FAHClient --help | grep config
Usage: FAHClient [[--config] <filename>] [OPTIONS]...
  --configure
      Create a basic configuration file, then exit.
```

MichaIng, 2018-08-08:
@Fourdee 
Ahh yes good point. Maybe disable autostart with debconf-set-selection, hope that really nothing starts up, then `FAHClient --configure`, hope that it really delays bash follow up until fully finished.
Then during install configuration step enable autostart.

Fourdee, 2018-08-08:
I think we can do own service for this:

🈯️ 

```
rm /etc/fahclient/config.xml

FAHClient --allow='0/0' --web-allow='0/0' --user=DietPi --team=234437 --gpu=true --log-rotate=false --log=/var/log/fahclient.log --run-as=fahclient --power=light

#Change name in web interface
#restart
#loads name from config.xml

```

MichaIng, 2018-08-08:
@Fourdee 
Hmm, actually I like it more, if we do not create own service and have arguments inside that might confuse users. But on the other hand, if it's working reliably.

In this case:
- Disable autostart via debconf-set-selections
- Install deb (should not start then)
- Install our service file, instead of edit/create config file.
- Leave auto configuration of conflicting settings to FAHClient, started by dietpi-services.
?

Fourdee, 2018-08-08:
> Disable autostart via debconf-set-selections

Not possible due to https://github.com/FoldingAtHome/fah-issues/issues/1193

The autostart must be set to ```true```

MichaIng, 2018-08-08:
@Fourdee 
Okay, so the only issue then is, that after deb install the service starts automatically, as well as on boot.

As long as it is stopped before config step, it is no big issue, what is it possible to then disable the service  (dietpi_controlled), have it not started on boot (before dietpi-login) and not face the startup error?

Fourdee, 2018-08-08:
@MichaIng 

> Okay, so the only issue then is, that after deb install the service starts automatically, as well as on boot.

I worked around this with:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L3903

Then
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L7917-L7922

So in practice, whether or not the auto service start creates the config.xml has no relevance, as we clear it  afterwards, then allow the service with its command line options to set the default items. config.xml is then generated when slot number is set (total CPU count), and/or, user makes changes in web GUI.

MichaIng, 2018-08-08:
> /mnt/dietpi_userdata/fahclient/config.xml

Is this automatically used with our service started, or can we adjust the service to directly create config.xml within userdata?
Then we would not need to to chown/clear `/etc/fahclient/config.xml`, just `rm -R /etc/fahclient` instead?
So do the following all in configuration step?
```
mkdir -p $G_FP_DIETPI_USERDATA/fahclient
chown -R fahclient:dietpi $G_FP_DIETPI_USERDATA/fahclient
> /var/log/fahclient.log
chown fahclient:dietpi /var/log/fahclient.log
rm -R /etc/fahclient
```

Fourdee, 2018-08-09:
> Last question is what happens, if user manually installs an update via new deb package? Does /etc/init.d/FAHClient get recreated, as well as old data and config dir then, as the init.d service assumes clean install (no config.xml etc)?

No idea, we'd need to test it locally with a dpkg reinstall.

> Is this automatically used with our service started (due to service file WorkingDir), or can we adjust the service to directly create config.xml within userdata?

Yep, the working dir and data dir combo ensures ```/mnt/dietpi_userdata/fahclient/config.xml``` is used.

> Then we would not need to chown/clear /etc/fahclient/config.xml, just rm -R /etc/fahclient instead?
So do the following all in configuration step?

Yep, i'll check 👍

Fourdee, 2018-08-09:
@MichaIng 

Interesting, you can manually add more slots in the config.xml to run more jobs:
[image]
```
root@DietPi:~# cat /mnt/dietpi_userdata/fahclient/config.xml
<config>
  <!-- Slot Control -->
  <power v='FULL'/>

  <!-- User Information -->
  <user v='Fourdee'/>

  <!-- Folding Slots -->
  <slot id='0' type='CPU'/>
  <slot id='1' type='CPU'/>
  <slot id='2' type='CPU'/>
</config>
```

Would be great if we could add default slots to:
- CPU cores total / 2?

I'll check CLI options

MichaIng, 2018-08-09:
> CPU cores total / 2

Hmm, on the other hand the last two work units I had took all four cores and gave many points. I guess the more cores, the more points you get within same time (1 WU 4 cores < 2 WU 2 cores).

Fourdee, 2018-08-09:
@MichaIng 

🈯️ http://dietpi.com/software (under misc to keep 2 lines of cats)
🈯️ https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=130#p13704

Fourdee, 2018-08-09:
@MichaIng 

> Hmm, on the other hand the last two work units I had took all four cores and gave many points. I guess the more cores, the more points you get within same time (1 WU 4 cores < 2 WU 2 cores).

Yep, its "pot luck" on the jobs vs cores. I managed to get my Ryzen running 3 jobs with 2 cores each.
[image]

Fourdee, 2018-08-09:
@MichaIng 

Testing passed my end. I believe we can mark this as completed?

MichaIng, 2018-08-09:
@Fourdee 
I will do another test on my 3 VMs, then mark as closed, if passed.

Fourdee, 2018-08-10:
@MichaIng 

> Testing passed

Excellent, great work on this 👍 

I'll mark this as completed.

Fourdee, 2018-08-10:
@MichaIng 

> So points become less, when you need longer. But does not explain why I got assigned way less points than what the final calculation (shown within the log right before uploading) was. But however, does not matter much 😉.

At least in this case the phrase "Points means prizes" does not apply 🤣

MichaIng, 2018-08-10:
@Fourdee 
Last open question:
> If we use "DietPi" without passkey, theoretically someone could request a passkey for this, which would break our default config. Either we need to create a passkey our side or we need to use a user name, where no passkey can be requested for. Found "anonymous" e.g. to be default on Windows client, maybe for this one it's the case.

I just requests via fake mail passkeys for "anonymous" and "Anonymous" and it showed "mail sent..", thus at least at the first stage there seem to be no check for obviously non-individual user names to block passkey request for them.

bb30994, 2018-08-11:
> 04:09:49:WARNING:WU00:FS00:AS lowered CPUs from 6 to 2

At the beginning of FAH's log, you'll find a summary of the hardware it detects.  It won't use threads that it can't detect.

I've never run on a SBC, so I don't know what that detection summary is going to show.

bb30994, 2018-08-11:
> Last open question:

    If we use "DietPi" without passkey, theoretically someone could request a passkey for this, which would break our default config. Either we need to create a passkey our side or we need to use a user name, where no passkey can be requested for. Found "anonymous" e.g. to be default on Windows client, maybe for this one it's the case.

> I just requests via fake mail passkeys for "anonymous" and "Anonymous" and it showed "mail sent..", thus at least at the first stage there seem to be no check for obviously non-individual user names to block passkey request for them.
Right.  See my last item above.

MichaIng, 2018-08-11:
@bb30994 
Nice, many thanks for your explanations 👍.

> Note that a Passkey is REQUIRED plus 10 completed WUs before you earn bonus points

Ahh okay, this explains why I so far just got the "Base Credit" for each job. Some more links/hints within the clients/UI could help to avoid confusion about his, but anyway collecting points is a minor issue here 😉.

> The same is true if you divide your CPU resources across multiple VMs. The best policy is to put all your resources toward a single job.

Jep, agree now. Next job here, again all 4 cores in use. Wouldn't make sense to split it up. I just wonder, as the first two work units just took 1 core, even that the second (on VM) respectively the other 3 (host machine) were on idle. Maybe the "Folding Power" setting also has an influence here? Never understood what this settings actually does, as at least I can't see some influence, when switching it. At least on the current WU, always 100% on all cores/threads.
@Fourdee 
Maybe you can retry if you get single WUs using 6 core with power setting to "full".

> In other words, don't share passkeys. That was never indended.

As said, I would not add this as sort of a benefit of long term solution. I just would like to assure (for the long term), that when our users install folding@home, the whole install + startup process fully finishes without any user interaction need and without any error message/popup of the kind "wrong passkey" / "please enter the passkey for your user name", as for full automated install we need to preconfigure one.

As theoretically someone could currently request a passkey for our preconfigured user name, I think requesting one ourself and preconfigure it with DietPi install is the safest method.
But we can then encourage users on docs etc. to choose an own user name.

MichaIng, 2018-08-11:
Something like this would be great:
```
<!-- Folding Slots -->
<slot id='0' type='CPU'>
  <cpus v='$G_HW_CPU_COUNT'/>
</slot>
```
Maybe as well possible via FAHClient argument?

Fourdee, 2018-08-12:
Reference:

I tried setting 
```
<cpus v='3'/>
```

```
15:02:36:WARNING:WU00:FS00:Changed SMP threads from 1 to 3 this can cause some work units to fail
15:02:36:WARNING:WU00:FS00:AS lowered CPUs from 3 to 1
#CPU 3 reported in web ui
```

Still using 1 thread/core.

```
root@DietPi:~# cat /proc/cpuinfo
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 76
model name      : Intel(R) Atom(TM) x5-Z8350  CPU @ 1.44GHz
stepping        : 4
microcode       : 0x406
cpu MHz         : 1921.418
cache size      : 1024 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 4
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 11
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb pti tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs            : cpu_meltdown spectre_v1 spectre_v2
bogomips        : 2880.00
clflush size    : 64
cache_alignment : 64
address sizes   : 36 bits physical, 48 bits virtual
power management:
```

Fourdee, 2018-08-12:
Ok, as this is now completed, i'll mark as closed. Please reopen if required.

MichaIng, 2018-08-12:
@Fourdee 
What you think about creating a passkey for our default user "DietPi"? In case should be done with dietpi.com mail.

Fourdee, 2018-08-12:
@MichaIng 

Yep, can do:
```
Hello DietPi,

   Your passkey is: 06c869246e88c00cb05cc4d1758a97f9
  Your username is: DietPi

Please use these to activate your client.
```

via daniel.knight@dietpi.com

Fourdee, 2018-08-12:
🈯️ Tested install, working fine with key. Marking as closed.

closed 2018-08-12 by Fourdee

nnovaes, 2018-08-16:
Thanks all for your efforts! Hopefully the community here will engage on this and dedicate a few clocks to this project. My Dietpi 6.13 is up and running and proudly folding on Dietpi's team behalf.