MichaIng/DietPi #1986: Image | Firefly RK3399

carlosedp, 2018-08-02:
I have build an image for Firefly RK3399 SBC based on the existing FriendlyARM NanoPC-T4 from [here](https://github.com/Fourdee/DietPi/issues/1829).

The image still has some NanoPC-T4 markings on it's config so it might require updates. The Kernel is the 4.4.138 built from Rockchip tree and contains most useful modules for Docker, Kubernetes, networking as IPVS and IPTables. The modules are in a .tar.gz file that need to be copied to the rootfs 
 `/lib/modules` after install. Newer DietPi rootfs already contains this in right place.

[image]

#### Device information:
- Device name:  Firefly RK3399
- Official product URL: http://en.t-firefly.com/product/rk3399.html
- Image download URL: https://drive.google.com/file/d/1m7OUr-Kpva6wJA6qkV3Ys_np53dsGJDz/view?usp=sharing

Benchmarks:

[image]

## Writing DietPi to the onboard EMMC:

Requires:

* Firefly RK3399
* USB-C Cable (eg: https://www.amazon.co.uk/Syncwire-USB-3-0-Type-Cable/dp/B01C8BFHW4)
* Ethernet connected (or WiFi can be setup with serial console during 1st run setup)
* A Windows system (to use the provided easy write tool with USB-C, Linux options require command line).
-- Linux users, please follow this guide: http://wiki.friendlyarm.com/wiki/index.php/NanoPC-T4#Flash_Image_to_eMMC_with_Linux_Upgrade_Tool

## Writing the images to the EMMC

* Download image file and unpack
* Power off the board
* Plug in the USB-C cable and power cable (do not power on). Plug in the other end of the USB-C cable into your Windows system.
* Launch the program AndroidTool.exe

Now we need to enter into MASKROM mode:

* Follow the steps from [http://en.t-firefly.com/doc/product/info/id/479.html](http://en.t-firefly.com/doc/product/info/id/479.html).
* When you see a MASKROM message appear in the AndroidTool.exe window, release the buttons.
* Select Run. This will write the images to the EMMC.
When the transfer is completed, the system will automatically reboot and begin the 1st run setup.

The image works with Firefly M.2 to SATA [adapter card](http://shop.t-firefly.com/goods.php?id=52). The drivers were included into the Kernel. Check benchmark image for the "Custom" line.

@Fourdee , can you check this out and update the config files/rootfs to contain the kernel modules and Firefly naming all around? Also, testers needed.

Fourdee, 2018-08-03:
@carlosedp 

Great work, thanks for creating and sharing this 👍 

> @Fourdee , can you check this out and update the config files/rootfs to contain the kernel modules and Firefly naming all around?

Yep, leave it with me.

carlosedp, 2018-08-03:
Gonna update my initial comment with this link since it contains the modules and correct naming.

Fourdee, 2018-08-16:
@carlosedp 

Ah yes, I repacked the ```.img``` for the image we uploaded: https://dietpi.com/downloads/testing/DietPi_FireflyRK3399-ARMv8-Stretch.7z

A quick way to test if its been packed, try opening the ```.img``` in 7zip. ```.iso``` will open, packed wont 😃

MichaIng, 2018-10-19:
We have anther open image request for ROC-RK3328-CC: https://dietpi.com/phpbb/viewtopic.php?f=12&t=2694

They at least share the same CPU with Rock64 at least, not sure if images can be used cross devices?

carlosedp, 2018-10-19:
Usually not because of the DTBs for each board. This seems very similar in specs to Rock64 from Pine64. The Rockchip tree doesn't contain it but I believe Firefly probably do. Need to check this once I get some time.

MichaIng, 2020-01-27:
@carlosedp 
I created a new Buster image, based on Armbian with Linux mainline kernel 5.4: https://dietpi.com/downloads/testing/DietPi_FireflyRK3399-ARMv8-Buster.7z
It is an SDcard image, no GPT => eMMC installer image like the one you created from NanoPC T4, since I am not familiar with the Android Tools method. Also it should be possible to flash it directly to eMMC via adapter or from running system on SDcard, remounting rootfs as rw and using `dd`.

However if some is experienced with the Android Tools (or equivalent) method to create an installer image (boot from e.g. USB flash, install directly to internal eMMC), then I am open for it and would love to integrate it into our image creation script: https://github.com/MichaIng/DietPi/blob/dev/.meta/dietpi-imager

MichaIng, 2020-06-18:
I mark this as closed. Interest in this board is VERY small, however Buster-based image is up, marked as beta, any observed issues can be handled via dedicated issue.

closed 2020-06-18 by MichaIng

belveder79, 2022-04-06:
@Fourdee I would be very interested in testing your image and hear if you are still actively using the firefly board...

MichaIng, 2022-04-06:
At least for common server usage, Android is indeed a questionable OS 😉. Let me know if you run into any issues. One AFAIK was that network connections can be dropped in the middle, when they are managed by connman or NetworkManager, so best is to run the script with screen and keyboard attached (instead of SSH), so you are able to re-establish the network connection from the related error prompt, if necessary.

belveder79, 2023-07-23:
@carlosedp do you still have those flashable images and some instruction how to flash it to emmc without an adaptor? 

I am close to use a nail and a hammer and use this board as a decorative element on the wall for my office...

MichaIng, 2023-07-27:
Jep, using the Armbian Bookworm image, then running our installer and choosing "Generic RK3399" should work: https://dietpi.com/docs/hardware/#make-your-own-distribution

belveder79, 2023-07-31:
@MichaIng @carlosedp yes, this indeed worked flawlessly, thx a lot...