MichaIng/DietPi #1989: feedbin

msongz, 2018-08-03:
#### Give us some formal software information:
- Software title | feedbin
- Short description | a rss reader with beautiful UI and UX !
- GitHub URL (if available) | https://github.com/feedbin/feedbin
- External URL (if available) | https://feedbin.com/
- (Official) install documentation | https://github.com/feedbin/feedbin#installation

MichaIng, 2018-08-03:
@msongz 
Thanks for your request.

I actually thought about adding [discourse](https://www.discourse.org/) to our software list, but the installations from my point of view and knowledge is a nightmare, as well with the need for ruby and many other dependencies to build.

`feedbin` also needs [ruby](https://www.ruby-lang.org/en/) and according to the instructions `a Rack compatible application server`. Not sure what this means, or let's say to implement this would need someone with knowledge/experience about this kind of environment and ruby and how to setup this on the architectures/devices we support.

Next dependency we then need to add is PostgreSQL. I thought for longer time to add this database, as it is also supported by ownCloud/Nextcloud and some other software titles we offer. I think this would be the first step, add postgres as a real alternative SQL database, which can be chosen instead of MariaDB for user who are used to it or setups that benefit from it.

So it is not something that will be implemented fast and I guess we need some help for it.

Fourdee, 2018-08-04:
@MichaIng 

> I actually thought about adding discourse to our software list, as well with the need for ruby and many other dependencies to build.

Yep, I looked at this in the past and came to the same conclusion. Didn't seem worth implementing.

Seems they are aware of the issues with this, even limiting support to Docker: https://github.com/discourse/discourse/blob/master/docs/INSTALL.md#how-do-i-install-discourse

> So it is not something that will be implemented fast and I guess we need some help for it.

Yep agree. Unless we have a large number of users requesting 'feedbin', will be some time before we can pick this up.