MichaIng/DietPi #1990: YMPD installation broken on stretch armv6

magicfoxt, 2018-08-03:
#### Required Information:
- DietPi Version | v6.3
- SBC Device     | RPi B (armv6l) (index=1)
- Distro         | stretch (index=4)
- Command        | Connection test: http://dietpi.com/downloads/binaries/all/mpd_0.20.11-1_armv6.deb
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:
YMPD installation

#### Actual behaviour:
mpd_0.20.11-1_armv6.deb file was not found in repo (but mpd_0.20.18-1_armv6.deb was here).

#### Steps to reproduce:
Installing MPD music player daemon with DietPi-Software

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2018-08-03 16:16:06--  http://dietpi.com/downloads/binaries/all/mpd_0.20.11-1_armv6.deb
Resolving dietpi.com (dietpi.com)... 185.101.93.93
Connecting to dietpi.com (dietpi.com)|185.101.93.93|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://dietpi.com/downloads/binaries/all/mpd_0.20.11-1_armv6.deb [following]
Spider mode enabled. Check if remote file exists.
--2018-08-03 16:16:07--  https://dietpi.com/downloads/binaries/all/mpd_0.20.11-1_armv6.deb
Connecting to dietpi.com (dietpi.com)|185.101.93.93|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```

MichaIng, 2018-08-03:
@magicfoxt 
Thanks for your report. Jep the old versions are indeed missing.

@Fourdee 
Did you recently remove the old binaries from the server? Although a `dietpi-update` fixes this, for we should keep the old ones to prevent such issues.

https://github.com/Fourdee/DietPi/blob/0a6e43cac45ab4e7639b4c0be878c75f87451350/dietpi/dietpi-software#L3727
- mpd_0.20.11-1_ for pre v6.4
https://github.com/Fourdee/DietPi/blob/6e9c18b2ee3b89f3a5dc1092149866dfb3f18f38/dietpi/dietpi-software#L3724
- mpd_0.19.21_arm64.deb for pre v6.9 (?)

Could be also solved via symlinks, so the newest version will be installed. We should really safe the binaries without version string, thus we can update them without needing to adjust download link and/or keep old binaries.

To fix now, I suggest you update DietPi to the current version.
If for some reason you want to stay on v6.3, you could do the following:
`sed -i 's/mpd_0.20.11-1/mpd_0.20.18-1/g' /DietPi/dietpi/dietpi-software`

We will resolve/recreate the old binaries on the server.

magicfoxt, 2018-08-03:
Thanks for you reply and actions.

Fourdee, 2018-08-04:
@MichaIng 

> Did you recently remove the old binaries from the server? Although a dietpi-update fixes this, for we should keep the old ones to prevent such issues.

Probably due to server move and me trying to clean up the new one. Although updating DietPi resolves this and your symlink fix, however, my mistake.

@magicfoxt 

If you would be so kind as to confirm this is now working? We can mark this as resolved.

magicfoxt, 2018-08-04:
Yes, I confirm it's working for me and I did a dietpi-update to be up to date ;)
Many thanks for your support !

MichaIng, 2018-08-04:
@magicfoxt 
Thanks for feedback, I will mark this as closed then 🙂.

closed 2018-08-04 by MichaIng